Koenkk/zigbee2mqtt #15975: [New device support]: S520513W

Cherryblue, 2023-01-06:
### Link

https://www.se.com/ww/en/product/S520513W/connected-dimmer-wiser-odace-rotary-push-typewhite/?node=94999087365-dimmers&parent-subcategory-id=5620&range=60324-odace&filter=business-5-residential-and-small-business&selected-node-id=94999093066

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0x94deb8fffe628a41","nwkAddr":51104,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"NHROTARY/DIMMER/1","epList":[3,21,242],"endpoints":{"3":{"profId":260,"epId":3,"devId":257,"inClusterList":[0,3,4,5,6,8,769,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"dateCode":""}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":23}},"lightingBallastCfg":{"attributes":{"ballastStatus":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dae797","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dae797","endpointID":1},{"cluster":769,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dae797","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}},"21":{"profId":260,"epId":21,"devId":260,"inClusterList":[0,3,2821,65303],"outClusterList":[3,4,5,6,8,258],"clusters":{"genBasic":{"attributes":{"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"","swBuildId":"002.011.015 R","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-220419804},"lastSeen":1672991490067,"defaultSendRequestWhen":"immediate"}

### Comments

This device is currently recognized as another Schneider Electrics Dimmer : [WDE002334](https://www.se.com/ww/en/product/WDE002334/wiser-exxact-dimmer-rotary-universal-led-white/)). This other dimmer reference is old and not produced anymore.

Once peered, no command worked. The only message I have is "UNSUPPORTED ATTRIBUTE" every time I try something.
My understanding is that this dimmer doesn't work as the other dimmer which z2m recognized.

Cherryblue, 2023-02-06:
Well, I'm still hoping for this device to be taken into account into a future version of z2m.

closed 2023-03-17 by github-actions[bot]

pcloviro, 2023-05-28:
Well, I'm too hoping for this device to be taken into account into a future version of z2m. I Bought 2 of them, and they are paired, but  known as..unknown in zigbee2mqtt

pcloviro, 2023-05-29:
finally, I got it to work. First, delete the device, restart pairing (3 press on button), wait 10 second, letting zigbee2mqtt to find information. identified as NHROTARY/DIMMER/1 model WDE002334. And it works
[image]

mrgrlscz, 2024-01-24:
Works well for me too!

romainalca, 2024-09-28:
@pcloviro @mrgrlscz  what are your versions of zigbee2mqtt ? I am interested to buy this device but I want to be sure that I will be totaly okay ! 🙂
Other question : what kind of router do You have and your firmware version please ? Thanks

mrgrlscz, 2024-09-28:
Z2M v1.39.1-1 running with a Sonoff type E controller firmware v7.4.2