Koenkk/zigbee2mqtt #15979: TERNCY-DC01 now identified as Sunricher SR-ZG9010A

DubhAd, 2023-01-06:
### What happened?

I updated Zigbee2MQTT 1.28.1 to 1.29.1, after this my TERNCY-DC01 sensors are mis-identified as Sunricher SR-ZG9010A and no longer function. No other devices were impacted.

> "manufacturerID":4648,"model":"SR-ZG9010A"

Removing and re-pairing doesn't change this.

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

20220726

### Adapter

Sonoff CC2652

DubhAd, 2023-01-06:
Looks like this was caused by https://github.com/Koenkk/zigbee-herdsman-converters/pull/5184

DubhAd, 2023-01-06:
I can confirm that reverting that change fixes this and the sensors are both correctly identified and work again.

DubhAd, 2023-01-06:
Here's an entry from `database.db` for one of them

```json
{
  "id": 83,
  "type": "EndDevice",
  "ieeeAddr": "0x000d6f00184cbf7f",
  "nwkAddr": 28291,
  "manufId": 4648,
  "manufName": "Xiaoyan",
  "powerSource": "Battery",
  "modelId": "TERNCY-DC01",
  "epList": [
    1
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 12,
      "inClusterList": [
        0,
        1,
        3,
        15,
        32,
        1026,
        64716
      ],
      "outClusterList": [
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "TERNCY-DC01",
            "manufacturerName": "Xiaoyan",
            "powerSource": 3,
            "zclVersion": 2,
            "hwVersion": 1
          }
        },
        "genOta": {
          "attributes": {
            "currentFileVersion": 154
          }
        },
        "msTemperatureMeasurement": {
          "attributes": {
            "measuredValue": 228
          }
        },
        "genPowerCfg": {
          "attributes": {
            "batteryPercentageRemaining": 56
          }
        },
        "genBinaryInput": {
          "attributes": {
            "presentValue": 0
          }
        },
        "genPollCtrl": {
          "attributes": {
            "checkinInterval": 3600
          }
        }
      },
      "binds": [
        {
          "cluster": 32,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0024c2ae7e",
          "endpointID": 1
        }
      ],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "hwVersion": 1,
  "zclVersion": 2,
  "interviewCompleted": true,
  "meta": {},
  "lastSeen": 1673018689148,
  "defaultSendRequestWhen": "active"
}
```

asgothian, 2023-01-06:
thank you. Since you have the TERNCY device, can you run a test for me ? I do not currently have access to the sun richer device, but will do the same test once I have:
- patch the device definition for the SR-ZG9010A back in
- modify the fingerprint from  `[{modelID: 'TERNCY-DC01', manufacturer: 'Sunricher'}]` to ` [{modelID: 'TERNCY-DC01', manufacturerName: 'Sunricher'}],`
- run zigbee2mqtt again - if everything behaves as planned, your devices should remain TERNCY-DC01.

Thank you for your help with this.

DubhAd, 2023-01-06:
Thank you. I'll give that a shot tomorrow - too late today to mess about with things without getting yelled at 😁

DubhAd, 2023-01-07:
Happy to say that worked, made that change, restarted Z2M, they're still identified correctly and work as before.

Thank you

asgothian, 2023-01-07:
Thank you for the test. I will double-check that the change will correctly detect the Sunricher Device and then put in a PR to fix it.

Koenkk, 2023-01-08:
Since https://github.com/Koenkk/zigbee-herdsman-converters/pull/5282 is merged, I assume this can be closed now. Thanks!

closed 2023-01-08 by Koenkk