Koenkk/zigbee2mqtt #15987: zigbee switches sometimes have to be woken up

syl20cjs, 2023-01-06:
### What happened?

hello,

sometimes I have to "wake up" my zigbee switches, especially the following one : 

- xiaomi opple 6 buttons : WXCJKG13LM
- xiaomi opple 4 buttons : WXCJKG12LM
- xiaomi 2 buttons switch : WXKG07LM

It happens often that I push it once, it does nothing, and after a few seconds if I push again, it does what its supposed to do.

I observered that it seems to be less frequent after a restart of my home assistant/MQTT instance. 

I also have other switches (

### What did you expect to happen?

I expect switches to work directly at first push =)

### How to reproduce it (minimal and precise)

hard to reproduce, it happen some times, not necessarily after a long period.

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

20210708

### Adapter

sonoff zigbee dongle plus

KenanSikiric, 2023-01-06:
I noticed the same behaviour lately with my IKEA switches (TRÅDFRI and STYRBAR), still being on 1.29.0. Not sure if that version introduced the bug but the dates seem to reasonably align with when I started experiencing it.

syl20cjs, 2023-01-06:
yeah you are right the problem appeared a few versions before, don't remember exactly when

MattWestb, 2023-01-07:
Its one more issue with one IKEA controller that have being sniffed and the device is sending the first command OK in the Zigbee network but the system is **not** "using" the command. The second command sent is working OK in Zigbee network and Z2M **is** using the command. I dont remember the topic of the issue but its around one week ago.

cova-fe, 2023-01-07:
In my installation I can see the same issue but only with STYRBAR E2001/E2002 (fw 1.0.024). On the same installation, TRADFRI E1524/E1810 (fw 2.3.080) and E1812 (fw 2.3.080) behaves correctly.
Maybe this could be related to firmware version, unfortunately I was not able to update the E2001 device even though it reports "new fimware available", So I can't yet verify the situation with a newer FW.

MattWestb, 2023-01-07:
All first and second gen IKEA controller with latest firmware can only binding to device and all firmware before it was only possible bonding to groups.
Styrbar (3rd gen) is possible updating to the latest version in ZHA if patching for firmware downgrading then the firmware inside the OTA package is one newer version but the meta tagged version of the OTA file is wrongly made. 
The draw back is that you is loosing group binding that is one braking thing is working with real Zigbee lights.
And ZHA have no problems with lost command sent from the device but can having repented commands from croup broadcast from some radio libs.

cova-fe, 2023-01-07:
> All first and second gen IKEA controller with latest firmware can only binding to device and all firmware before it was only possible bonding to groups. Styrbar (3rd gen) is possible updating to the [...]

I'm not really sure to get what you suggestion is, tbh. Could you rephrase it?

PulsarFX, 2023-01-09:
I've noticed such things with an aqara switch: https://www.zigbee2mqtt.io/devices/WXKG11LM.html#xiaomi-wxkg11lm
still on z2m 1.28.4

KenanSikiric, 2023-02-04:
Might make sense to add that this issue exists on a "command level" and not on a device level, i.e. you have to "wake" every single button (and even button hold commands) before they work individually rather than just "waking" the entire device with any button and having them all respond immediately. This basically means that if i want to turn on two different lights with one device, i still end up pressing buttons 4 times, 2 for each button/light (assuming the issue shows up).

PulsarFX, 2023-03-09:
Still hsppens on my side

CReimer, 2023-03-14:
With the ikea switches it’s the same when they are connected to ikea bulbs directly. This made me give up on zigbee completely for now

maxime1992, 2023-04-14:
Up

BartRuSec, 2023-04-19:
I have the same problem with TS0044 on latest Z2M and upgraded firmware from #13478. Every 2 or 3 days my switch is disconnecting and I have pair it again.

MattWestb, 2023-04-19:
@BartRuSec Is you running current Z2M then it was one fix for battery draining added for some time ago ?
Ypu need resetting the device after have updated Z2M then its need losing some not good config that  was made before or its continuing daring batteries.

BartRuSec, 2023-04-19:
@MattWestb I did it and it didn't help. I do not think that it is a direct problem with batteries but with switching between coordinator and router and than device enters to pairing mode (even z2m shows as paired). I will try to downgrade z2m to version 1.30.2-1  and observe the impact.

KenanSikiric, 2023-04-20:
@BartRuSec this issue is not related to disconnection/unpairing issues. It is best described as a sort of sleep state like OP does, where you have to click the same button twice for it to be triggered.

BartRuSec, 2023-05-03:
@KenanSikiric what I have observed that it is responsive but it behaves like in pairing mode. I have tested it on z2m ver 1.30.2, 1.30.3 and now I am on 1.30.4 and will check if it is occurs or not. I have Sonoff Dongle P with FW from https://github.com/Koenkk/zigbee2mqtt/issues/13478 topic

closed 2023-06-10 by github-actions[bot]

gofort, 2023-11-18:
Up, the problem still exists.
Here are some details.
In general, the root of the problem is in the state.
Here are the states of 2 switches WXKG11LM:

1.
```
{
    "battery": 100,
    "device_temperature": 29,
    "linkquality": 196,
    "power_outage_count": 32,
    "voltage": 3045
}
```
2. 
```
{
    "battery": 100,
    "device_temperature": 29,
    "linkquality": 200,
    "power_outage_count": 15,
    "voltage": 3055,
    "action": ""
}
```

The second switch is "awaken", because it has `action` equal to an empty string. The first switch is not awakened as `action` field is not available.
Because of that, Home Assistant Trigger doesn't work from the first click. 
This problem is very easy to reproduce: just restart docker container with zigbee2mqtt. This will reset the state and `action` will disappear from the state.

gofort, 2023-11-19:
+ https://www.reddit.com/r/homeassistant/comments/v1nlyr/aqara_mini_switch_with_zigbee2mqtt_automations/
@Koenkk can you please reopen this issue?

DuckyCrayfish, 2024-02-28:
I am also experiencing this.

BertilJ, 2024-03-04:
I'm having the same issue with ALL my older Aqara/Xiaomi-buttons and some older Ikea.

KenanSikiric, 2025-04-03:
Very late reply, but I finally fixed the issue on my end. Previously, I used the switches in automations with a device trigger. Changing this to use an MQTT trigger got rid of the "wake up" behaviour and now they all work as they should.

gofort, 2025-04-03:
I still hope it will be solved one day :lol:

TitjyDev, 2025-05-11:
> Very late reply, but I finally fixed the issue on my end. Previously, I used the switches in automations with a device trigger. Changing this to use an MQTT trigger got rid of the "wake up" behaviour and now they all work as they should.

For a newbie, can explain how MQTT trigger works? Can you share your yaml please?