Koenkk/zigbee2mqtt #15988: RM3500ZB Sinopé Water Heater temperature not reporting and current not in the good range

Queferr, 2023-01-06:
### What happened?

RM3500ZB Sinopé Water Heater temperature not reporting and current not in the good range.

Also tested in the Edge version and same results.
Temperature (sensor on the device) is always "Null".
Current(A) is 1000 times too high than normal.
All the other functions are good.

### Zigbee2MQTT version

1.29.1-1

### Adapter firmware version

0x26580700

### Adapter

ConBee2

alray31, 2023-01-07:
Should receive mine by the end of next week, will try to help as soon as I can install it on my water heater.
 
Are you sure the current isn't reported in mA, that would explain why the value is 1000x more than what you expect?
For the temperature, what is the result if you poll ''measuredValue'' from the Temperature cluster in the dev console?

alray31, 2023-01-07:
Can you also poll rmsCurrent, acCurrentDivisor and acCurrentMultiplier from the haElectricalMeasurement cluster and report the values?

alray31, 2023-01-07:
Here is the code for the RM3500ZB from sinope.js

`
zigbeeModel: ['RM3500ZB'],
model: 'RM3500ZB',
vendor: 'Sinopé',
description: 'Calypso smart water heater controller',
fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ias_water_leak_alarm_1, fz.temperature],
toZigbee: [tz.on_off],
exposes: [e.switch(), e.power(), e.current(), e.voltage(), e.energy(), e.water_leak(), e.temperature()],
configure: async (device, coordinatorEndpoint) => {
       const endpoint = device.getEndpoint(1);
       const binds = ['genOnOff', 'haElectricalMeasurement', 'seMetering', 'Temperature'];
       await reporting.bind(endpoint, coordinatorEndpoint, binds);
       await reporting.onOff(endpoint);
       await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
       await reporting.activePower(endpoint);
       await reporting.rmsCurrent(endpoint);
       await reporting.rmsVoltage(endpoint);
       await reporting.readMeteringMultiplierDivisor(endpoint);
       await reporting.currentSummDelivered(endpoint);
       await reporting.temperature(endpoint, {min: 60, max: 3600, change: 1});`

What I suspect is that maybe the readEletricalMeasurementMultiplierDivisors function is not returning any divisor value for rmsCurrent  so the raw value in mA might be showing instead of being divided by 1000 to expose Ampere instead of milli ampere. Hence why I'm curious to know what are your values when you poll rmsCurrent, acCurrentDivisor and acCurrentMultiplier.

You should try to use an external converter to troubleshoot, try to change the configure function to manually divide rmsCurrent by 1000 and see if the exposed current is now correct after rebooting z2m. Somehting like this:

`
zigbeeModel: ['RM3500ZB'],
model: 'RM3500ZB',
vendor: 'Sinopé',
description: 'Calypso smart water heater controller',
fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ias_water_leak_alarm_1, fz.temperature],
toZigbee: [tz.on_off],
exposes: [e.switch(), e.power(), e.current(), e.voltage(), e.energy(), e.water_leak(), e.temperature()],
configure: async (device, coordinatorEndpoint) => {
      const endpoint = device.getEndpoint(1);
      const binds = ['genOnOff', 'haElectricalMeasurement', 'seMetering', 'Temperature'];
      await reporting.bind(endpoint, coordinatorEndpoint, binds);
      await reporting.onOff(endpoint);
      await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
      await reporting.activePower(endpoint);
      const current = await reporting.rmsCurrent(endpoint);
      device.metrics.current = current / 1000;  // Divide RMS current by 1000 to convert from milliamperes to amperes
      await reporting.rmsVoltage(endpoint);
      await reporting.readMeteringMultiplierDivisor(endpoint);
      await reporting.currentSummDelivered(endpoint);
      await reporting.temperature(endpoint, {min: 60, max: 3600, change: 1});
}
`

Queferr, 2023-01-09:
Hi, thank you for your quick help!
Yes I think it's in mA.

From the DEV Console:
Read result of 'haElectricalMeasurement': {"rmsCurrent":3240}
Read result of 'haElectricalMeasurement': {"acCurrentDivisor":1000,"acCurrentMultiplier":1}

After reading the value in DEV Console, the state changed to the correct divisor: 3.24A

For the temperature:
Read result of 'msTemperatureMeasurement': {"measuredValue":1660}

After the reading the value adjusted in the device to 16.60deg but the value is not updating after that.

And I not mentioned it before but the "energy" field is "Null" too.
Thanks

alray31, 2023-01-13:
Sorry for the late answer,

I got my Calypso wednesday and installed it this morning. Got the same problems as you and as I though when I read your last reply, this is only a reporting issue. I'm pretty sure it can be fixed directly from the sinope.js converter but here what you can do now to fix it on your side.

The temparature max report interval is set to 3600 so it will only report at least once an hour, You can reduce the interval to what you'd like.

Go in the ''report'' tab and add a line like this. In this example, the max report interval is set to 60 (1 minute) instead of 3600. It might be a bit overkill to ask a temp report each minute, you may adjust to whatever you'd like but you'll definitevly see the value updated each minute.
[image]

For the energy consumption in kWh not being reported,
Go to dev console and poll the attributes ''multiplier'' and 'diviser'' from the seMetering cluster (they are about midway in the attributes list of seMetering.
 
This will set the multiplier & divisor values for the currentSummDelivered attribute of seMetering (''energy'' is looking for this to expose the total energy in kWh).

Then, like for the temperature attribute, go back in the report tab and add an entry for seMetering currentSummDelivered and set the min/max report interval to what you'd like
[image]

[image]

[image]

[image]

[image]

[image]

Queferr, 2023-01-17:
Thank you very much for your answer. All seems to be ok now. I hope someone will modify the sinope.js.

alray31, 2023-01-22:
@MattL0 Well the converter in sinope.js works, its just that Power and Current will not correctly show unless you manually poll all these attributes at least once:

- haElectricalMeasurement acCurrentDivisor
- haElectricalMeasurement acCurrentMultiplier
- seMetering Multiplier
- seMetering Divisor

And the report interval for the temperature is a bit high imo but that can easily be adjusted in the reports tab.

I'm currently investigating another issue with this device. There is a safety feature on this thing that will restore power to the water heater automatically when the temperature decrease to 45 Celsius. This is apparently something that can be deactivated since Sinope are explaining in their installation guide that this can be deactivated from their priopriatary software (Neviweb). 

This feature is hazardous when using the Calypso with the Sedna water valve. I just had a problem beacause of that and came back home in a hurry after I got a notification of a water leak at home. I left home for 2 days, so I shut the Sedna water valve to prevent leak and turned off Calypso so I dont waste electricity heating the water when I'm not home. Because the water valve is shut, your hot water system is now a closed system (your water heater and your house piping). When the temperature reach 45, Calypso turns on itself and the water heat goes up from 45 to 60 causing a proportional inscrease in pressure (Pascal's law: In a constant volume, a temperature increase will cause a pressure increase). Because the pressure can't be absorbed by the volume of water from the city (sedna valve closed), the pressure will eventually reach the 150 psi required to open the releif valve of the water heater. My releif valve discharge pipe redirect water inside a pan under my water heater where there is leak detector....  So for me, it was mor fear than harm, only about 4 cups of hot water leaked inside the pan.

But for someone with a defective pressure releif valve (i.e clogged with sediments) this could lead to the water heater cracking open under this pressure and so causing major water damage from the 60 gallons of water that will comes out....

So yeah, I need to find what attribute to change in order to deactivate this ''safety'' feature.

[image]

Disconnecting the temperature probe each time I leave is not really convenient for a smart home perspective

alray31, 2023-01-22:
So after a bit research, I found [here](https://github.com/claudegel/sinope-zha) that this attribute is called drConfigWaterTempMin, its a manufacturer specific attribute from Sinopé:
```
Cluster: 0xff01
Attribute: 0x0076
Data type: t.uint8_t
Function: drConfigWaterTempMin
Values: 45 or 0
```
[image]

This attribute is not yet in manuSpecificSinope...
I guess we first have to add
`drConfigWaterTempMin: {ID: 118, type: DataType.uint8},`
to src/zcl/definition/cluster.ts  ?

And then find a way to either set this attribute to 45 (On) or 0 (Off) with a switch?

Would that make sense?

alray31, 2023-02-01:
The reporting issues should now be fixed in zigbee2mqtt version 1.30.0-1 thanks to gembak
I have also added some manufacturer specific attributes for sinopé to src/zcl/definition/cluster.ts in a hope to soon be able to test the drConfigWaterTempMin attribute to turn on/off the water temp safety feature.

closed 2023-03-12 by github-actions[bot]