Koenkk/zigbee2mqtt #15989: [New device support]: Neo ZigBee Smart Plug (PLUG-001SPB2)

jsvensson, 2023-01-06:
### Link

(unable to find)

### Database entry

{"id":8,"type":"Router","ieeeAddr":"0xa4c138e19755c118","nwkAddr":882,"manufId":4417,"manufName":"_TZ3000_gjnozsaz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_gjnozsaz","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1673017693528}

### Comments

This seems to be another whitelabel version of a TuYa device. Manufacturer is called "Neo Coolcam" but doesn't seem to have any web presence.

- Model ID: `TS011F`
- Manufacturer name: `_TZ3000_gjnozsaz`
- Manufacturer ID: `4417`

I see entries with similar manufacturer names under [the TS011F_plug_3 definition](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js#L2530), with the manufacturer name following the same format.

I managed to create an external converter (below) that was enough to get it into Z2M and toggle power on/off, but unfortunately that's the extent of what I could figure out so far.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS011F'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'PLUG-001SPB2', // Vendor model number, look on the device for a model number
    vendor: 'Neo', // Vendor of the device (only used for documentation and startup logging)
    description: 'Power Plug', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off], // We will add this later
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.switch()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

jsvensson, 2023-01-06:
After some more googling I'm also seeing this sold as "Tuya Smart Zigbee 3.0 Power Plug" with identical appearance.

Koenkk, 2023-01-07:
I've added this as a whitelabel of the `TS011F_plug_3`, so it will appear on the supported devices list with the next release.

jsvensson, 2023-01-07:
Thanks a lot! Waiting for the next release so I can play around with the power monitoring on these things.

closed 2023-01-07 by jsvensson