Koenkk/zigbee2mqtt #15992: Lidl HG06492C/HG07834C/HG06104A power_on_behavior error

PiotrZych, 2023-01-06:
### What happened?

Lidl HG06492C/HG07834C/HG06104A devices  doesn't allow to set "power_on_behavior".

### What did you expect to happen?

power_on_behavior update on device

### How to reproduce it (minimal and precise)

In zigbee2mqtt web gui try to set power_on_behavior.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

`Debug 2023-01-06 20:28:54Received MQTT message on 'zigbee2mqtt/0x84ba20fffe5d7fe6/set' with data '{"power_on_behavior":"off"}'
Debug 2023-01-06 20:28:54Publishing 'set' 'power_on_behavior' to '0x84ba20fffe5d7fe6'
Error 2023-01-06 20:28:54Publish 'set' 'power_on_behavior' to '0x84ba20fffe5d7fe6' failed: 'Error: Write 0x84ba20fffe5d7fe6/1 genOnOff({"startUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2023-01-06 20:28:54Error: Write 0x84ba20fffe5d7fe6/1 genOnOff({"startUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:105:13) at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:246:36)`

jperquin, 2023-01-19:
I am experiencing exactly the same problem. Publishing directly via MQTTExplorer, ie. zigbee2mqtt/FRIENDLY_NAME/set with json payload {"power_on_behavior": on} does not work either.

Alfagek, 2023-01-22:
Same problem here with the HG06492C but also with the HG06492B.

Koenkk, 2023-01-23:
None of the Lidl lights support this, it will be disabled for all in the next z2m release.

closed 2023-01-23 by Koenkk

Alfagek, 2023-01-23:
Thanks a lot @Koenkk 👍🏻