Koenkk/zigbee2mqtt #15997: Can't trigger new Hue Effects on groups

deviantintegral, 2023-01-06:
### What happened?

While existing effects like "Breathe" and "OK" can be triggered on groups, the new Candlelight and Fireplace effects don't work.

Candle in particular has an intentional flicker. I would think that being able to send the command to a group would mean that all bulbs are in sync. Otherwise, the out-of-sync aspect is very noticeable in two-light fixtures.

### What did you expect to happen?

I expected the new commands in the Effect dropdown to work.

### How to reproduce it (minimal and precise)

1. Create a group with at least one Hue light supporting a new effect.
2. In Home Assistant, select the Candle effect for the light group.
3. See nothing happens.
4. Confirm sending `set` manually via MQTT doesn't work either.

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

20221102

### Adapter

zzh!

kris-anderson, 2023-01-11:
While this would be nice and I ran into the same issue, I was able to update my automations in Home Assistant to refer to the entities themselves instead of the group.

deviantintegral, 2023-02-11:
Not stale!

lundyfpv, 2023-02-16:
This does still seem to be an issue.

Koenkk, 2023-02-16:
Implemented this, can you check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

lundyfpv, 2023-02-17:
> Implemented this, can you check if it works?
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

I can confirm that Candle, Fireplace and Colorloop still do not work on groups with 1.30.1-dev commit: e934b0d. They do still work at the individual level. THe older effects like breath and okay still work on groups however.

Koenkk, 2023-02-18:
@lundyfpv can you provide the herdsman debug log when setting a hue effect to a group?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

lundyfpv, 2023-02-18:
@Koenkk I'm using HAOS and the Zigbee2MQTT Edge addon. Where are the herdsman debug logs for HAOS? The link you sent seems to indicate they are logged in the addon Log tab. This is what I have from that log file...

`2023-02-18 08:55:55Publishing 'set' 'color' to 'group_bulbs_living_room'
2023-02-18 08:55:55Requested color_temp 575 is higher than maximum supported 500, using maximum!
2023-02-18 08:55:56Received MQTT message on 'zigbee2mqtt/group_bulbs_living_room/set' with data '{"state":"ON","color":{"x":0.701,"y":0.299}}'
2023-02-18 08:55:56Skipping state because of Home Assistant
2023-02-18 08:55:56Publishing 'set' 'color' to 'group_bulbs_living_room'
2023-02-18 08:55:59Received MQTT message on 'zigbee2mqtt/group_bulbs_living_room/set' with data '{"state":"ON","effect":"blink"}'
2023-02-18 08:55:59Publishing 'set' 'state' to 'group_bulbs_living_room'
2023-02-18 08:55:59Publishing 'set' 'effect' to 'group_bulbs_living_room'
2023-02-18 08:56:02Received MQTT message on 'zigbee2mqtt/group_bulbs_living_room/set' with data '{"state":"ON","effect":"breathe"}'
2023-02-18 08:56:02Publishing 'set' 'state' to 'group_bulbs_living_room'
2023-02-18 08:56:03Publishing 'set' 'effect' to 'group_bulbs_living_room'
2023-02-18 08:56:06Received MQTT message on 'zigbee2mqtt/group_bulbs_living_room/set' with data '{"state":"ON","effect":"stop_effect"}'
2023-02-18 08:56:06Publishing 'set' 'state' to 'group_bulbs_living_room'
2023-02-18 08:56:06Publishing 'set' 'effect' to 'group_bulbs_living_room'
2023-02-18 08:56:09Received MQTT message on 'zigbee2mqtt/group_bulbs_living_room/set' with data '{"state":"ON","effect":"candle"}'
2023-02-18 08:56:09Publishing 'set' 'state' to 'group_bulbs_living_room'
2023-02-18 08:56:09Publishing 'set' 'effect' to 'group_bulbs_living_room'
2023-02-18 08:56:11Received MQTT message on 'zigbee2mqtt/group_bulbs_living_room/set' with data '{"state":"ON","effect":"fireplace"}'
2023-02-18 08:56:11Publishing 'set' 'state' to 'group_bulbs_living_room'
2023-02-18 08:56:11Publishing 'set' 'effect' to 'group_bulbs_living_room'
2023-02-18 08:56:13Received MQTT message on 'zigbee2mqtt/group_bulbs_living_room/set' with data '{"state":"ON","effect":"colorloop"}'
2023-02-18 08:56:13Publishing 'set' 'state' to 'group_bulbs_living_room'
2023-02-18 08:56:13Publishing 'set' 'effect' to 'group_bulbs_living_room'
`

lundyfpv, 2023-02-18:
Here are the logs from Breathe and Okay that do seem to work fine. 

`Zigbee2MQTT:debug 2023-02-18 09:32:15: Received MQTT message on 'zigbee2mqtt/group_bulbs_living_room/set' with data '{"state":"ON","effect":"breathe"}'
Zigbee2MQTT:debug 2023-02-18 09:32:15: Publishing 'set' 'state' to 'group_bulbs_living_room'
2023-02-18T15:32:15.277Z zigbee-herdsman:controller:group Command 1 genOnOff.on({})
2023-02-18T15:32:15.278Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":195,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,150,1]}}
2023-02-18T15:32:15.278Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,2,1,1,0,0,0,0,0,0,0,255,0,0,1,6,0,195,0,30,3,0,17,150,1,145]
2023-02-18T15:32:15.295Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2023-02-18T15:32:15.295Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2023-02-18T15:32:15.295Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2023-02-18T15:32:15.296Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
2023-02-18T15:32:15.296Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-02-18T15:32:15.375Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,195,5]
2023-02-18T15:32:15.375Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,195,5]
2023-02-18T15:32:15.375Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,195] - 5
2023-02-18T15:32:15.376Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":195}
2023-02-18T15:32:15.376Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2023-02-18 09:32:15: Publishing 'set' 'effect' to 'group_bulbs_living_room'
2023-02-18T15:32:15.577Z zigbee-herdsman:controller:group Command 1 genIdentify.triggerEffect({"effectid":1,"effectvariant":0})
2023-02-18T15:32:15.578Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":3,"transid":196,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[17,151,64,1,0]}}
2023-02-18T15:32:15.578Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,25,36,2,1,1,0,0,0,0,0,0,0,255,0,0,1,3,0,196,0,30,5,0,17,151,64,1,0,218]
2023-02-18T15:32:15.599Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2023-02-18T15:32:15.599Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2023-02-18T15:32:15.600Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2023-02-18T15:32:15.600Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
2023-02-18T15:32:15.601Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-02-18T15:32:15.668Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,196,2]
2023-02-18T15:32:15.668Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,196,2]
2023-02-18T15:32:15.669Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,196] - 2
2023-02-18T15:32:15.669Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":196}
2023-02-18T15:32:15.672Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2023-02-18 09:32:15: MQTT publish: topic 'zigbee2mqtt/bulb_living_room_corner_lamp', payload '{"brightness":213,"color":{"h":240,"hue":240,"s":98,"saturation":98,"x":0.144,"y":0.053},"color_mode":"xy","color_temp":500,"linkquality":149,"power_on_behavior":null,"state":"ON","update":{"installed_version":16784906,"latest_version":16784918,"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2023-02-18 09:32:15: MQTT publish: topic 'zigbee2mqtt/group_bulbs_living_room', payload '{"brightness":213,"color":{"h":240,"hue":240,"s":98,"saturation":98,"x":0.144,"y":0.053},"color_mode":"xy","color_temp":500,"state":"ON"}'
Zigbee2MQTT:info  2023-02-18 09:32:15: MQTT publish: topic 'zigbee2mqtt/bulb_bar_lamp_color', payload '{"brightness":213,"color":{"h":240,"hue":240,"s":98,"saturation":98,"x":0.144,"y":0.053},"color_mode":"xy","color_temp":500,"linkquality":112,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16784906,"latest_version":16784918,"state":"available"},"update_available":true}'
Zigbee2MQTT:debug 2023-02-18 09:32:18: Received MQTT message on 'zigbee2mqtt/group_bulbs_living_room/set' with data '{"state":"ON","effect":"okay"}'
Zigbee2MQTT:debug 2023-02-18 09:32:18: Publishing 'set' 'state' to 'group_bulbs_living_room'
2023-02-18T15:32:18.789Z zigbee-herdsman:controller:group Command 1 genOnOff.on({})
2023-02-18T15:32:18.790Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":197,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,152,1]}}
2023-02-18T15:32:18.791Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,2,1,1,0,0,0,0,0,0,0,255,0,0,1,6,0,197,0,30,3,0,17,152,1,153]
2023-02-18T15:32:18.815Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
[... cut ...]
2023-02-18T15:32:23.531Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":175,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":33,"attrData":18}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":34956,"endpoint":1,"linkquality":43,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'`

Koenkk, 2023-02-19:
@lundyfpv in your log I see that the Hue effect command is send to group 1 correctly:

```
Zigbee2MQTT:debug 2023-02-18 09:27:16: Publishing 'set' 'effect' to 'group_bulbs_living_room'
2023-02-18T15:27:16.657Z zigbee-herdsman:controller:group Command 1 manuSpecificPhilips2.multiColor({"data":{"type":"Buffer","data":[33,0,1,1]}})
2023-02-18T15:27:16.658Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":64515,"transid":175,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[17,136,0,33,0,1,1]}}
```

My first guess would be that the Hue bulbs don't support these commands through groups. To confirm this, someone needs to check wether this is supported through the Hue hub.

gdgib, 2023-03-05:
I'm not sure that's supported in the v2 API (https://developers.meethue.com/develop/hue-api-v2/api-reference/#resource_grouped_light__id__put), but it definitely was in the v1 API (https://developers.meethue.com/develop/hue-api/groupds-api/#set-gr-state).

HOWEVER, I just tested this using the v1 API, and it absolutely doesn't work for the colorloop effect.  I can set the effect only on lights themselves.  Attempting to set the light group effect to colorloop changes nothing.   This suggests it may be a firmware limitation.

The older, and now gone, "Hue Labs" colorloop effect did allow groups to be set to colorloop all at once without visible popcorn.  But I couldn't say if it was implemented on the bulb or the hub.  The ability to control this was uneven and poorly implemented and is long gone now.  I have memories of this being only available for either all lights in the system through the app and needing to use alexa integration to control it at a room/zone level.

I also believe the colorloop effect, though still implemented in hardware, is being deprecated somewhat in the API because it's not listed in the v2 API docs I glanced through just now.  That may mean that they have changed the way colorloop is implemented in a fundamental way during firmware updates.

Overall, I can't tell if:

1. The hardware/firmware supports setting effects on groups, but the v1 API (which is being deprecated) doesn't work correctly any more.
2. The hardware/firmware does not support setting effects on groups.

I suspect it's actually case 2, since the v2 API doesn't support it, and the v1 API for it doesn't work.

In other words, I'm fairly sure this is a limitation of hue, and not a Z2M bug.  But I absolutely cannot prove that.

gdgib, 2023-03-05:
Please note that this is similar to #16929, but that issue is about using scenes, and the current hue hub,app, & API all support effects through scenes as one would expect.  So assuming we can get #16929 fixed, perhaps using scenes would be a valid alternative for the use case here, assuming I'm correct and the problem here really is a hue firmware limitation.

gdgib, 2023-04-05:
Still working on this.  Just slowly.

closed 2023-05-13 by github-actions[bot]

gdgib, 2024-01-25:
Can confirm this is still broken.  I haven't forgotten, but obviously I haven't gotten time either.

valhall453, 2024-09-13:
Just got some hue bulbs and am also encountering this still.  Can activated candle/fireplace individually on each bulb just fine but cannot apply to a group of bulbs.

ristoavila, 2024-11-01:
I also do see the same problem, would be nice to be able set effects

trenchcoatjedi, 2025-03-01:
EXACT same issue with this light strip.

Hue white and color ambiance LightStrip outdoor 2m  https://www.zigbee2mqtt.io/devices/9290022890.html#philips-9290022890