Koenkk/zigbee2mqtt #15998: [Feature request]: Tuya TS0601 temperature can only have integers

berrywhite96, 2023-01-06:
### Is your feature request related to a problem? Please describe

Currently the Tuya TS0601 `current_heating_setpoint` can only have integers, but 0.5 steps are possible.

When I set the temperature e.g. 21.5 with HA, Z2M sets it to a minus value.

### Describe the solution you'd like

Change the step value for `current_heating_setpoint` to 0.5.

### Describe alternatives you've considered

None

### Additional context

None

dabm-git, 2023-01-09:
I have the same problem, specifically I have a few _TZE200_aoclfnxz, TS0601, model: BHT-002-GCLZB

It only accepts integers, both in calibration and target temperature, apparently from the panel, it manually accepts steps of 0.5 degrees, but it seems that the model is limited in terms of Zigbee firmware, since it only accepts integers in this way.

Therefore, although the panel physically accepts 0.5 degree jumps, the zigbee internal logic really only accepts 1 degree jumps.

This is somewhat unfortunate for these models, and the best thing to do is to get rid of them, unless there is some way to update the firmware.

berrywhite96, 2023-01-09:
> I have the same problem, specifically I have a few _TZE200_aoclfnxz, TS0601, model: BHT-002-GCLZB
>
> It only accepts integers, both in calibration and target temperature, apparently from the panel, it [...]

Oh wow that's sad to hear, because they are well made and cheap. Never thought they just forgot to implement the 0.5 steps.

Ticket can be closed if there is really no way to do it.

closed 2023-01-10 by Koenkk

guny74, 2023-01-31:
Really strange, behavour of the firmware. But if you use the schedule/program it works with x.5 degree....
You can set at the program 20.5. if the schedule time comes on the device the new setpoint is 20.5, but over the zigbee 20 degree as "current_heating_setpoint"  is delivered.... so the current setoint have to be calculated by the the current time and the schedules...

berrywhite96, 2023-01-31:
> Really strange, behavour of the firmware. But if you use the schedule/program it works with x.5 degree.... You can set at the program 20.5. if the schedule time comes on the device the new setpoint [...]

Can confirm this behaviour, this makes it even more annoying to not set manually a stepped temperature. Maybe going to write a helper entity for it, still don't have a replacement for it.

guny74, 2023-02-01:
I tried to use deconz, there the "current_heating_setpoint" is delivered correctly from device to the software, but the other way also only integrer...but at deconz I could not set the schedule correctly... so I hope someone will work here on it and find a better solution