Koenkk/zigbee2mqtt #15999: [New device support]: Bseed thermostat TS0601 TZE200_5toc8efa

VoltickVL, 2023-01-07:
### Link

https://aliexpress.ru/item/1005004352027105.html?spm=a2g2w.orderdetail.0.0.2d174aa6ejwWSN&sku_id=12000029623915182

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0x003c84fffed3326f","nwkAddr":43783,"manufId":4098,"manufName":"_TZE200_5toc8efa","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000)L+\u0012\u0001)L+\u0012","65506":31,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE200_5toc8efa","powerSource":1,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1673060688980,"defaultSendRequestWhen":"immediate"}

### Comments

I have problem with Bseed thermostat TS0601 TZE200_5toc8efa electric floor heating at Zigbee2Mqtt Edge. 
But Bseed thermostat TS0601 TZE200_aoclfnxz work fine.

Please add support for TZE200_5toc8efa to Zigbee2Mqtt Edge...

Koenkk, 2023-01-08:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-08 by Koenkk

VoltickVL, 2023-01-08:
I have Zigbee2MQTT Edge Addon, but my device still not unavailable. Pls help (:

VoltickVL, 2023-01-08:
I have Zigbee2MQTT Edge Addon, but my device still not unavailable. Pls help how i can add my device...

VoltickVL, 2023-01-09:
I have this errors... And when i set current_heating_setpoint to 30 *C, my device set only 0.3 *C and i have this warning:

> Zigbee2MQTT:warn  2023-01-09 16:06:06: zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,235]}}
> Zigbee2MQTT:warn  2023-01-09 16:07:06: zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,255]}}
> Zigbee2MQTT:warn  2023-01-09 16:08:07: zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,245]}}

elemarek, 2023-02-21:
Does this thermostat suffer from the same defect as Moes brand thermostats such as congestion with an enormous number of packets?

VGorokhov, 2024-05-30:
**LINK** 
[https://aliexpress.ru/item/1005005646484680.html?spm=a2g2w.detail.rcmdprod.3.3f9d626cTLlvNc&mixer_rcmd_bucket_id=aerabtestalgoRecommendAbV26_controlRu3&pdp_trigger_item_id=0_1005003250877648&ru_algo_pv_id=8cb101-40a3c7-1759f0-919701-1717056000&scenario=pcDetailBottomMoreThisSeller&sku_id=12000033965583156&traffic_source=recommendation&type_rcmd=core](BSEED ZigBee Smart Socket)

**Database entry**

{
    "linkquality": 176,
    "child_lock": "UNLOCK",
    "current": null,
    "energy": null,
    "indicator_mode": "off/on",
    "power": null,
    "power_outage_memory": "restore",
    "state": "OFF",
    "update": {
        "installed_version": 64,
        "latest_version": 192,
        "state": "available"
    },
    "update_available": null,
    "voltage": null
}

**Comments**
I have problem with BSEED ZigBee Smart Socket [TS011F_plug_3]

**Power**
No response

**Current**
No response

**Voltage**
No response

**Energy**
No response

nickeeey, 2024-11-01:
I really want to integrate this Thermostat. Has anyone got it working?