Koenkk/zigbee2mqtt #16000: Since 1.28 problems with network (sonoff dongle E)

kizio1000, 2023-01-07:
### What happened?

Since 1.28 network devices started to go randomly offline. After update to 1.29 and 1.29.1 z2m starts randomly to loose connection and go offline after couple hours of normal work w/o any reason. Frequently problems with starting z2m too (usually first start fails but restarting addon leads to correct start).

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220719140040

### Debug log

debug 2023-01-07 00:06:46: Loaded state from file /config/zigbee2mqtt/state.json
info  2023-01-07 00:06:46: Logging to console and directory: '/config/zigbee2mqtt/log/2023-01-07.00-06-46' filename: log.txt
debug 2023-01-07 00:06:46: Removing old log directory '/config/zigbee2mqtt/log/2023-01-04.00-08-26'
info  2023-01-07 00:06:46: Starting Zigbee2MQTT version 1.29.1 (commit #unknown)
info  2023-01-07 00:06:46: Starting zigbee-herdsman (0.14.83)
debug 2023-01-07 00:06:46: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220719140040-if00"}}'
debug 2023-01-07 00:11:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:16:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:21:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:26:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:31:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:36:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:41:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:46:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:51:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:56:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:01:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:06:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:11:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:16:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:21:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:26:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:31:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:36:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:41:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:46:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:51:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:56:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:01:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:06:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:11:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:16:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:21:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:26:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:31:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:36:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:41:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:46:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:51:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:56:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:01:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:06:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:11:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:16:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:21:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:26:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:31:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:36:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:41:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:46:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:51:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:56:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:01:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:06:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:11:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:16:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:21:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:26:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:31:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:36:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:41:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:46:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:51:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:56:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 05:01:50: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 05:06:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 05:11:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 05:16:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 05:21:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 05:26:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 05:31:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 05:36:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 05:41:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 05:46:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 05:51:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 05:56:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 06:01:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 06:06:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 06:11:52: Saving state to file /config/zigbee2mqtt/state.json
error 2023-01-07 06:12:06: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
    at Frontend.stop (/app/lib/extension/frontend.ts:83:39)
    at Controller.callExtensions (/app/lib/controller.ts:316:40)
    at Controller.stop (/app/lib/controller.ts:193:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9))

xm4rcell0x, 2023-01-07:
Hi! i have your exact same problem since december. Now i'm with 1.29 stable and problem is still here. It's frustrating because every couple of days in the morning i have to restart manually the container. 
can you confirm you have the same logs as me ? https://drive.google.com/drive/folders/1jdS9ZyjI79l3Y9ZEzMUek9V1IQRpiI3w?usp=share_link

trackhacs, 2023-01-07:
same here - 1.24 was working great with Dongle - E.

since upgrade 1.28 -> 1.29.x   unusable. 

needed to switch dongle (conbee2).

xm4rcell0x, 2023-01-07:
Thanks man, really appreciated! 
I really hope devs can fix this problem because it's a pain

JacekKac, 2023-01-07:
i am on 1.28 and sonoff E dongle also on 3 hosts. No problems

trackhacs, 2023-01-07:
u will find a lot complains here about E with >= 1.28.

seems like here a some specific factors causing it, cause works ok for some.

cloudbr34k84, 2023-01-07:
I'm sonoff P and have the same issues. 

I'm finding devices have 1-2min delays between activating

I'm finding I'm having to restart Z2M to just turn off lights etc.

I can't figure it out, but definitely since 1.29.

xm4rcell0x, 2023-01-07:
> I'm sonoff P and have the same issues. 
> 
> I'm finding devices have 1-2min delays between activating
> 
> I'm finding I'm having to restart Z2M to just turn off lights etc.
> 
> I can't figure it out, but definitely since 1.29.

Your frontend is reachable? 
As you can see from our logs frontend stops to working, so all the devices are unresponsive unless a manual restart

xm4rcell0x, 2023-01-07:
> same here - 1.24 was working great with Dongle - E.
> 
> 
> 
> since upgrade 1.28 -> 1.29.x   unusable. 
> 
> 
> 
> needed to switch dongle (conbee2).

Sry for stupid question, but how to downgrade to 1.24? 
I'm on Docker on Unraid and I pull the image with "latest" tag

trackhacs, 2023-01-07:
had that couple of days myself.

you just pick a container on the link below, than change the yml file, than just docker compose up. should automaticly download and switch version.

[(https://hub.docker.com/r/koenkk/zigbee2mqtt/tags)]

afterwards you could delete the the db and json files - if you want a clean start. 
if you leave configuration.yaml intact - z2q and HASS will remember the names, but you still need to rejoin all devices.

jjansen85, 2023-01-07:
> I'm sonoff P and have the same issues
> I'm finding devices have 1-2min delays between activating
> I'm finding I'm having to restart Z2M to just turn off lights etc.
> I can't figure it out, but definitely since 1.29.

@cloudbr34k84 your issues might be related to this https://github.com/Koenkk/zigbee2mqtt/issues/15856

trackhacs, 2023-01-07:
yes _ here seems to be a couple of parallel issues on this.

cloudbr34k84, 2023-01-07:
> Your frontend is reachable?
> As you can see from our logs frontend stops to working, so all the devices are unresponsive unless a manual restart

My frontend is reachable, but might as well not be

I just woke to find majority of devices unavailable...

kiekerjan18, 2023-01-07:
> same here - 1.24 was working great with Dongle - E.
> 
> since upgrade 1.28 -> 1.29.x unusable.
> 
> needed to switch dongle (conbee2).

can you switch dongle without losing all devices and repair them?

BjornHgh, 2023-01-08:
+1
Sunday morning, all zigbee devices down, lights in the living room were on though offline, again. YAZR, yet another Z2M restart.. Yesterday same story, had to reboot my entire (Synology) VM several times, Sonoff dongle seemed to be unresponsive: unplug, reseat, reboot... Coffee first, then out hunting for at least 10 sensors thoughout the house playing hide & seek.

cloudbr34k84, 2023-01-08:
What coordinator are you using??

I had the same this morning and just now with 20 devices out of 120 lol

cloudbr34k84, 2023-01-08:
> +1
> Sunday morning, all zigbee devices down, lights in the living room were on though offline, again. YAZR, yet another Z2M restart.. Yesterday same story, had to reboot my entire (Synology) VM [...]

I'm pretty close to starting ZigBee all over again... Lol

BjornHgh, 2023-01-08:
> I'm pretty close to starting ZigBee all over again... Lol

Suggestions for more stable core components are always welcome. (The new HA official dongle perhaps?)

xm4rcell0x, 2023-01-08:
Skyconnect has the same chipset as the dongle-e
The problem is in Z2M itself so we can do nothing.
We can only provide logs on logs and hoping and praying that koenkk see us 😅

Koenkk, 2023-01-08:
@kirovilya can you check if we can improve the EZSP integration based on the feedback here?

trackhacs, 2023-01-08:
just go to 

> > same here - 1.24 was working great with Dongle - E.
> > since upgrade 1.28 -> 1.29.x unusable.
> > needed to switch dongle (conbee2).
> 
> can you switch dongle without losing all devices and repair them?

only way i found so far is to keep the configuration.yml (where all the identifier and friendly name is stored), delete all the other db and json stuff. 
you still need to repair all the devices - but works instantly and nothing change with HASS.

kiekerjan18, 2023-01-08:
> 

> just go to
>
>
> only way i found so far is to keep the configuration.yml (where all the identifier and friendly name is stored), delete all the other db and json stuff. you still need to repair all the [...]

I have also a conbee switch, but repairing of all devices worries me a bit. Some are hidden behind fixed light and walls. Thanks for the info...

xm4rcell0x, 2023-01-08:
We are all with 6.10.3fw based on EZSP 8? 
Just to know ... maybe the 7.x based on EZSP 9 can fix some problems?

trackhacs, 2023-01-08:
as far as i read into it - 7.x is unofficial and might brake path to downgrade again tough.

t112013, 2023-01-10:
i think same here, out of nothing, not possible to turn lights off. the problem is that devices are not marked as unabaliable, restart of the ad-on helps. 
It seems like the coordinatorjust dissapears ? 
How can i check if the coordinator is online ?  how can an automation help me restarting the ad-on ? 
thanks for great help guys

Zigbee2MQTT:info  2023-01-10 21:34:02: MQTT publish: topic 'zigbee2mqtt/Kjokken SL 04', payload '{"brightness":254,"color_mode":"color_temp","color_temp":260,"hue_power_on_behavior":"recover","linkquality":172,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16784906,"latest_version":16784906,"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2023-01-10 21:34:20: Succesfully changed options
Zigbee2MQTT:info  2023-01-10 21:34:20: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"2jr6g-1"}'
Zigbee2MQTT:debug 2023-01-10 21:34:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-01-10 21:39:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-01-10 21:44:39: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-01-10 21:44:42: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"error"}},"transaction":"oqa30-1"}'
Zigbee2MQTT:info  2023-01-10 21:44:55: Succesfully changed options
Zigbee2MQTT:info  2023-01-10 21:44:55: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"oqa30-3"}'
Zigbee2MQTT:info  2023-01-10 22:07:11: Starting network scan (includeRoutes 'false')
Zigbee2MQTT:error 2023-01-10 22:07:17: Failed to execute LQI for 'Coordinator'
Zigbee2MQTT:error 2023-01-10 22:07:23: Failed to execute LQI for 'Inngang SL 01'
Zigbee2MQTT:error 2023-01-10 22:07:29: Failed to execute LQI for 'Inngang SL 02'
Zigbee2MQTT:error 2023-01-10 22:07:35: Failed to execute LQI for 'Inngang SL 03'
Zigbee2MQTT:error 2023-01-10 22:07:41: Failed to execute LQI for 'Inngang SL 04'

BjornHgh, 2023-01-11:
Last weekend I re-paired all zigbee devices to Z2M. A lot of them seemed to prefer the coordinator by default. 
Allowed pairing only to the nearest router (dropdown menu) for each device, seems to be running stable since. Fingers crossed.

kizio1000, 2023-01-12:
For like two days it worked w/o disconnecting on edge version, today the same problem, this in logs:

`debug 2023-01-12 00:01:50: Loaded state from file /config/zigbee2mqtt/state.json
info  2023-01-12 00:01:50: Logging to console and directory: '/config/zigbee2mqtt/log/2023-01-12.00-01-50' filename: log.txt
debug 2023-01-12 00:01:51: Removing old log directory '/config/zigbee2mqtt/log/2023-01-08.10-41-54'
info  2023-01-12 00:01:51: Starting Zigbee2MQTT version 1.29.1-dev (commit #e92d3f4)
info  2023-01-12 00:01:51: Starting zigbee-herdsman (0.14.84)
debug 2023-01-12 00:01:51: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220719140040-if00"}}'
debug 2023-01-12 00:06:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 00:11:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 00:16:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 00:21:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 00:26:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 00:31:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 00:36:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 00:41:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 00:46:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 00:51:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 00:56:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 01:01:52: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 01:06:52: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 01:11:52: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 01:16:52: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 01:21:52: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 01:26:52: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 01:31:52: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 01:36:52: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 01:41:52: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 01:46:52: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 01:51:52: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 01:56:52: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 02:01:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 02:06:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 02:11:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 02:16:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 02:21:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 02:26:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 02:31:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 02:36:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 02:41:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 02:46:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 02:51:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 02:56:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 03:01:53: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 03:06:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 03:11:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 03:16:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 03:21:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 03:26:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 03:31:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 03:36:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 03:41:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 03:46:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 03:51:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 03:56:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 04:01:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 04:06:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 04:11:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 04:16:54: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 04:21:55: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 04:26:55: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 04:31:55: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 04:36:55: Saving state to file /config/zigbee2mqtt/state.json
[... cut ...]
debug 2023-01-12 05:41:56: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-12 05:46:56: Saving state to file /config/zigbee2mqtt/state.json
error 2023-01-12 05:48:16: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
    at Frontend.stop (/app/lib/extension/frontend.ts:83:39)
    at Controller.callExtensions (/app/lib/controller.ts:316:40)
    at Controller.stop (/app/lib/controller.ts:193:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9))
`

xm4rcell0x, 2023-01-14:
> For like two days it worked w/o disconnecting on edge version, today the same problem, this in logs:
>
>
>
>
>
> `debug 2023-01-12 00:01:50: Loaded state from file /config/zigbee2mqtt/state.json
>
> info [...]

Also for me
From Wednesday to Friday it worked without any problem . Today morning Z2M has crashed ... really curious to know what changed in these 3 days

frog-rabbit, 2023-01-15:
same here. worked a couple of days and now suddenly some devices are not responding anymore. no network route (205)

xm4rcell0x, 2023-01-16:
New day, new error. 
for the first time i found the z2m container stopped. 
that's the log:
`error 2023-01-16 06:00:12: Not connected to MQTT server!
debug 2023-01-16 06:00:16: Received Zigbee message from 'Presa Camera di Marcello', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
error 2023-01-16 06:00:16: Not connected to MQTT server!
error 2023-01-16 06:00:16: Cannot send message: topic: 'zigbee2mqtt/Presa Camera di Marcello', payload: '{"child_lock":"UNLOCK","current":0,"energy":0.15,"indicator_mode":"off/on","last_seen":"2023-01-16T06:00:16+01:00","linkquality":88,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","latest_version":192,"state":"available"},"update_available":null,"voltage":232}
debug 2023-01-16 06:00:18: Received Zigbee message from 'Frigorifero', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
error 2023-01-16 06:00:18: Not connected to MQTT server!
error 2023-01-16 06:00:18: Cannot send message: topic: 'zigbee2mqtt/Frigorifero', payload: '{"child_lock":"LOCK","current":0.07,"energy":34.08,"indicator_mode":"off/on","last_seen":"2023-01-16T06:00:18+01:00","linkquality":100,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":"unknown","latest_version":192,"state":"available"},"update_available":null,"voltage":233}
error 2023-01-16 06:00:22: Not connected to MQTT server!
warn  2023-01-16 06:00:26: Failed to ping 'Lampadario Taverna' (attempt 1/1, Read 0x94deb8fffec13016/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 3958 - 1 - 57 - 0 - 1 after 10000ms))
debug 2023-01-16 06:00:29: Active device 'Lampadario Taverna' was last seen '481.70' minutes ago.
error 2023-01-16 06:00:32: Not connected to MQTT server!
debug 2023-01-16 06:00:32: Received Zigbee message from 'Segnapassi Scala', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":70}' from endpoint 1 with groupID 0
error 2023-01-16 06:00:32: Not connected to MQTT server!
error 2023-01-16 06:00:32: Cannot send message: topic: 'zigbee2mqtt/Segnapassi Scala', payload: '{"last_seen":"2023-01-16T06:00:32+01:00","linkquality":148,"power_on_behavior":"previous","state":"OFF","switch_type":"toggle"}
debug 2023-01-16 06:00:35: Received Zigbee message from 'Finestra Piccola', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
error 2023-01-16 06:00:35: Not connected to MQTT server!
error 2023-01-16 06:00:35: Cannot send message: topic: 'zigbee2mqtt/Finestra Piccola', payload: '{"calibration_left":"OFF","calibration_right":"OFF","calibration_time_left":12.4,"calibration_time_right":10,"last_seen":"2023-01-16T06:00:35+01:00","linkquality":100,"motor_reversal_left":"OFF","motor_reversal_right":"OFF","moving_left":"STOP","moving_right":"STOP","position_left":60,"position_right":0,"state_left":"OPEN","state_right":"CLOSE"}
error 2023-01-16 06:00:35: Not connected to MQTT server!
error 2023-01-16 06:00:35: Cannot send message: topic: 'zigbee2mqtt/Finestra Piccola/left', payload: '{"calibration":"OFF","calibration_time":12.4,"motor_reversal":"OFF","moving":"STOP","position":60,"state":"OPEN"}
error 2023-01-16 06:00:35: Not connected to MQTT server!
error 2023-01-16 06:00:35: Cannot send message: topic: 'zigbee2mqtt/Finestra Piccola/right', payload: '{"calibration":"OFF","calibration_time":10,"motor_reversal":"OFF","moving":"STOP","position":0,"state":"CLOSE"}
debug 2023-01-16 06:00:35: Succesfully pinged 'Finestra Piccola' (attempt 1/2)
debug 2023-01-16 06:00:39: Received Zigbee message from 'Intrattenimento Salone', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
error 2023-01-16 06:00:39: Not connected to MQTT server!
error 2023-01-16 06:00:39: Cannot send message: topic: 'zigbee2mqtt/Intrattenimento Salone', payload: '{"child_lock":"UNLOCK","current":0.09,"energy":24.07,"indicator_mode":"off/on","last_seen":"2023-01-16T06:00:39+01:00","linkquality":148,"power":8,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":233}
debug 2023-01-16 06:00:41: Received Zigbee message from 'Finestrone', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
error 2023-01-16 06:00:41: Not connected to MQTT server!
error 2023-01-16 06:00:41: Cannot send message: topic: 'zigbee2mqtt/Finestrone', payload: '{"backlight_mode":"high","calibration":"OFF","calibration_time":13.4,"last_seen":"2023-01-16T06:00:41+01:00","linkquality":136,"motor_reversal":"OFF","moving":"STOP","position":0,"state":"CLOSE"}
error 2023-01-16 06:00:41: Not connected to MQTT server!
error 2023-01-16 06:00:41: Cannot send message: topic: 'zigbee2mqtt/Finestrone/mode', payload: '{"backlight":"high"}
debug 2023-01-16 06:00:41: Succesfully pinged 'Finestrone' (attempt 1/2)
error 2023-01-16 06:00:42: Not connected to MQTT server!
error 2023-01-16 06:00:52: Not connected to MQTT server!
[... cut ...]
error 2023-01-16 06:21:10: Failed to start zigbee
error 2023-01-16 06:21:10: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-01-16 06:21:10: Exiting...
error 2023-01-16 06:21:10: Error: Failure send getNetworkParameters:{"type":"Buffer","data":[27,0,1,40,0]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:516:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
`

closed 2023-02-24 by github-actions[bot]

Atn-D, 2023-02-24:
Was there a fix?

xm4rcell0x, 2023-02-24:
Nope
Every morning I have to restart manually the container!

kirovilya, 2023-02-24:
try to use dev / edge version of z2m