Koenkk/zigbee2mqtt #16003: Zigbee2mqtt fails to start on ZB Dongle-E

phodina, 2023-01-07:
### What happened?

I'm trying to run the zigbee2mqtt (v1.28.0) for the first time. I bought the SONOFF Zigbee 3.0 USB Dongle Plus-E.

So I start the zigbee2mqtt and it gives me errors - see the log file.

I've checked with issues already posted and added the `adapter: ezsp` as suggested here https://github.com/Koenkk/zigbee2mqtt/issues/15175.

### What did you expect to happen?

My assumption is that the zigbee2mqtt will run and talk to the adapter.

### How to reproduce it (minimal and precise)

Just connect the dongle and start the zigbee2mqtt with this configuration:
```
$ cat ~/.z2m/configuration.yaml
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mqtt
serial:
  port: /dev/ttyACM0
frontend:
  port: 8080
advanced:
  network_key:
    - 109
    - 222
    - 193
    - 179
    - 112
    - 122
    - 140
    - 5
    - 38
    - 7
    - 109
    - 248
    - 221
    - 248
    - 6
    - 71
adapter: ezsp
```

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

6.10.3-41

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

### Debug log

```
debug 2023-01-07 10:39:06: Can't load state from file /home/pethod/.z2m/state.json (doesn't exist)
info  2023-01-07 10:39:06: Logging to console and directory: '/home/pethod/.z2m/log/2023-01-07.10-39-06' filename: log.txt
info  2023-01-07 10:39:06: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
info  2023-01-07 10:39:06: Starting zigbee-herdsman (0.14.62)
debug 2023-01-07 10:39:06: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/home/pethod/.z2m/coordinator_backup.json","databaseBackupPath":"/home/pethod/.z2m/database.db.backup","databasePath":"/home/pethod/.z2m/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
error 2023-01-07 10:39:26: Error while starting zigbee-herdsman
error 2023-01-07 10:39:26: Failed to start zigbee
error 2023-01-07 10:39:26: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-01-07 10:39:26: Exiting...
error 2023-01-07 10:39:26: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/nix/store/dw1byj0i2hals5rinva1bphqzc74sdlp-zigbee2mqtt-1.28.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/nix/store/dw1byj0i2hals5rinva1bphqzc74sdlp-zigbee2mqtt-1.28.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/nix/store/dw1byj0i2hals5rinva1bphqzc74sdlp-zigbee2mqtt-1.28.0/lib/node_modules/zigbee2mqtt/lib/zigbee.ts:58:27)
    at Controller.start (/nix/store/dw1byj0i2hals5rinva1bphqzc74sdlp-zigbee2mqtt-1.28.0/lib/node_modules/zigbee2mqtt/lib/controller.ts:101:27)
    at start (/nix/store/dw1byj0i2hals5rinva1bphqzc74sdlp-zigbee2mqtt-1.28.0/lib/node_modules/zigbee2mqtt/index.js:109:5)
```

trackhacs, 2023-01-07:
how you running  z2m ?  system/docker ?

are you sure its ttyACM0 ?

does the user running z2m has access rights to access ttyACM0 ?

phodina, 2023-01-07:
Hi @trackhacs,

I can see it the device being enumerated and also it has the access to the dev node:
```
# dmesg
[265230.077302] usb 1-3: new full-speed USB device number 81 using xhci_hcd
[265230.236817] usb 1-3: New USB device found, idVendor=1a86, idProduct=55d4, bcdDevice= 4.42
[265230.236829] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[265230.236833] usb 1-3: Product: SONOFF Zigbee 3.0 USB Dongle Plus V2
[265230.236837] usb 1-3: Manufacturer: ITEAD
[265230.236840] usb 1-3: SerialNumber: 20220812114603
[265230.239100] cdc_acm 1-3:1.0: ttyACM0: USB ACM device

$ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Jan  7 10:11 usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812114603-if00 -> ../../ttyACM0
$ test -w /dev/ttyACM0 && echo success || echo failure
success

$ Elelabs_EzspFwUtility.py probe -p /dev/ttyACM0
2023/01/07 10:17:16 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2023/01/07 10:17:16 Elelabs_EzspFwUtility:   Firmware: 6.10.3-41
2023/01/07 10:17:16 Elelabs_EzspFwUtility:   EZSP v8
```

I run z2m from NixOS:
```
$ nix-shell -p zigbee2mqtt
$ zigbee2mqtt
```

xm4rcell0x, 2023-01-07:
Where do you add "adapter"? 
It seems to be under advanced, but you have to put it under serial

phodina, 2023-01-07:
Thanks, that seems to be the issue. Now it appears to start

```
$ zigbee2mqtt
Zigbee2MQTT:debug 2023-01-07 11:09:36: Can't load state from file /home/pethod/.z2m/state.json (doesn't exist)
Zigbee2MQTT:info  2023-01-07 11:09:36: Logging to console and directory: '/home/pethod/.z2m/log/2023-01-07.11-09-36' filename: log.txt
Zigbee2MQTT:debug 2023-01-07 11:09:36: Removing old log directory '/home/pethod/.z2m/log/2023-01-07.09-21-59'
Zigbee2MQTT:info  2023-01-07 11:09:36: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
Zigbee2MQTT:info  2023-01-07 11:09:36: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:debug 2023-01-07 11:09:36: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/home/pethod/.z2m/coordinator_backup.json","databaseBackupPath":"/home/pethod/.z2m/database.db.backup","databasePath":"/home/pethod/.z2m/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/ttyACM0"}}'
Assertion failed: Command (setValue) returned unexpected state: [object Object]
Assertion failed: Command (setValue) returned unexpected state: [object Object]
Zigbee2MQTT:info  2023-01-07 11:09:39: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-01-07 11:09:39: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
Zigbee2MQTT:debug 2023-01-07 11:09:39: Zigbee network parameters: {"channel":11,"extendedPanID":221,"panID":6754}
Zigbee2MQTT:info  2023-01-07 11:09:39: Currently 0 devices are joined:
Zigbee2MQTT:warn  2023-01-07 11:09:39: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2023-01-07 11:09:39: Allowing new devices to join.
Zigbee2MQTT:warn  2023-01-07 11:09:39: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2023-01-07 11:09:39: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2023-01-07 11:09:39: Connecting to MQTT server at mqtt://mqtt
Zigbee2MQTT:debug 2023-01-07 11:09:39: Using MQTT anonymous login
Zigbee2MQTT:error 2023-01-07 11:09:39: MQTT error: getaddrinfo ENOTFOUND mqtt
Zigbee2MQTT:error 2023-01-07 11:09:39: MQTT failed to connect, exiting...
Zigbee2MQTT:info  2023-01-07 11:09:39: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2023-01-07 11:09:39: Stopped zigbee-herdsman
```

phodina, 2023-01-07:
I've also fixed the IP address of the mqtt server and enabled anonymous login in the configuration. Now it runs:
```
zigbee2mqtt
Zigbee2MQTT:debug 2023-01-07 11:12:24: Can't load state from file /home/pethod/.z2m/state.json (doesn't exist)
Zigbee2MQTT:info  2023-01-07 11:12:24: Logging to console and directory: '/home/pethod/.z2m/log/2023-01-07.11-12-24' filename: log.txt
Zigbee2MQTT:debug 2023-01-07 11:12:24: Removing old log directory '/home/pethod/.z2m/log/2023-01-07.09-22-57'
Zigbee2MQTT:info  2023-01-07 11:12:24: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
Zigbee2MQTT:info  2023-01-07 11:12:24: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:debug 2023-01-07 11:12:24: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/home/pethod/.z2m/coordinator_backup.json","databaseBackupPath":"/home/pethod/.z2m/database.db.backup","databasePath":"/home/pethod/.z2m/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/ttyACM0"}}'
Assertion failed: Command (setValue) returned unexpected state: [object Object]
Assertion failed: Command (setValue) returned unexpected state: [object Object]
Zigbee2MQTT:info  2023-01-07 11:12:26: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-01-07 11:12:26: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
Zigbee2MQTT:debug 2023-01-07 11:12:26: Zigbee network parameters: {"channel":11,"extendedPanID":221,"panID":6754}
Zigbee2MQTT:info  2023-01-07 11:12:26: Currently 0 devices are joined:
Zigbee2MQTT:warn  2023-01-07 11:12:26: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2023-01-07 11:12:26: Allowing new devices to join.
Zigbee2MQTT:warn  2023-01-07 11:12:26: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2023-01-07 11:12:26: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2023-01-07 11:12:26: Connecting to MQTT server at mqtt://127.0.0.1
Zigbee2MQTT:debug 2023-01-07 11:12:26: Using MQTT anonymous login
Zigbee2MQTT:info  2023-01-07 11:12:26: Connected to MQTT server
Zigbee2MQTT:info  2023-01-07 11:12:26: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2023-01-07 11:12:26: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2023-01-07 11:12:26: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"unknown","coordinator":{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"},"log_level":"debug","network":{"channel":11,"extendedPanID":221,"panID":6754},"permit_join":true,"version":"1.28.0"}'
```
```
$ mosquitto -c mosquitto.conf
1673086297: mosquitto version 1.6.12 starting
1673086297: Config loaded from mosquitto.conf.
1673086297: Opening ipv4 listen socket on port 1883.
1673086297: Opening ipv6 listen socket on port 1883.
1673086297: mosquitto version 1.6.12 running
1673086346: New connection from 127.0.0.1 on port 1883.
1673086346: New client connected from 127.0.0.1 as mqttjs_76db1497 (p2, c1, k60).
```

phodina, 2023-01-07:
And I'm able to see the Web interface on `127.0.0.1:8080`.

Thanks @trackhacs for spotting the error in the config file!

Just one small issue I noticed and that's connectivity if the dongle is attached to the device directly in the port. It doesn't start and also interferes with the WIFi.

Is there any other solution beside placing it further on longer USB cable 1~2 meters?

trackhacs, 2023-01-07:
wasnt me though @xm4rcell0x 

;)

phodina, 2023-01-07:
Thank you both :-)

closed 2023-01-07 by phodina

xm4rcell0x, 2023-01-07:
@phodina let us know how it works in the next couple of days
A lot of us with Dongle-E have problem with this hardware.

trackhacs, 2023-01-07:
I just updated conbee2 to 0x26780700,  now I also have *lost* devices. 
after very good impression on 1.24 that feels really unstable now.