Koenkk/zigbee2mqtt #16005: Interview failed with SNZB-04

phodina, 2023-01-07:
### What happened?

Hello,
I'm trying to configure the Sonoff Door Sensor SNZB-04. I can see the device in the list of devices.

However, it shows it's not correctly configured. When click on reconfigure it also fails.

I've also tried:
    Force remove device through z2m frontend
    stop z2m
    start z2m
    try to pair device

I don't have other sensors atm but there are people who claim to have it working https://github.com/Koenkk/zigbee2mqtt/issues/13600#issuecomment-1287756456

### What did you expect to happen?

The device is configured correctly and recognized as https://www.zigbee2mqtt.io/devices/SNZB-04.html.

### How to reproduce it (minimal and precise)

Enable paring for z2m and press the switch on the sensor for 5s to se 3 LED flashes.
Then wait and see the configuration fail message in the dashboard.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

6.10.3-41

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

### Debug log

```
$ zigbee2mqtt
Zigbee2MQTT:debug 2023-01-07 12:12:42: Loaded state from file /home/pethod/.z2m/state.json
Zigbee2MQTT:info  2023-01-07 12:12:42: Logging to console and directory: '/home/pethod/.z2m/log/2023-01-07.12-12-42' filename: log.txt
Zigbee2MQTT:debug 2023-01-07 12:12:42: Removing old log directory '/home/pethod/.z2m/log/2023-01-07.10-03-11'
Zigbee2MQTT:info  2023-01-07 12:12:42: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
Zigbee2MQTT:info  2023-01-07 12:12:42: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:debug 2023-01-07 12:12:42: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/home/pethod/.z2m/coordinator_backup.json","databaseBackupPath":"/home/pethod/.z2m/database.db.backup","databasePath":"/home/pethod/.z2m/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/ttyACM0"}}'
Assertion failed: Command (setValue) returned unexpected state: [object Object]
Assertion failed: Command (setValue) returned unexpected state: [object Object]
Zigbee2MQTT:info  2023-01-07 12:12:45: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-01-07 12:12:45: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
Zigbee2MQTT:debug 2023-01-07 12:12:45: Zigbee network parameters: {"channel":11,"extendedPanID":221,"panID":6754}
Zigbee2MQTT:info  2023-01-07 12:12:45: Currently 1 devices are joined:
Zigbee2MQTT:info  2023-01-07 12:12:45: 0x00124b0024ce7007 (0x00124b0024ce7007): SNZB-04 - SONOFF Contact sensor (EndDevice)
Zigbee2MQTT:warn  2023-01-07 12:12:45: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2023-01-07 12:12:45: Allowing new devices to join.
Zigbee2MQTT:warn  2023-01-07 12:12:45: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2023-01-07 12:12:45: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2023-01-07 12:12:45: Connecting to MQTT server at mqtt://127.0.0.1
Zigbee2MQTT:debug 2023-01-07 12:12:45: Using MQTT anonymous login
Zigbee2MQTT:info  2023-01-07 12:12:45: Connected to MQTT server
Zigbee2MQTT:info  2023-01-07 12:12:45: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2023-01-07 12:12:45: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2023-01-07 12:12:45: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"unknown","coordinator":{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"},"log_level":"debug","network":{"channel":11,"extendedPanID":221,"panID":6754},"permit_join":true,"version":"1.28.0"}'
Zigbee2MQTT:debug 2023-01-07 12:12:52: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"0x00124b0024ce7007","transaction":"1tv74-9"}'
Zigbee2MQTT:info  2023-01-07 12:12:52: Configuring '0x00124b0024ce7007'
Zigbee2MQTT:error 2023-01-07 12:12:53: Failed to configure '0x00124b0024ce7007', attempt 1 (Error: Bind 0x00124b0024ce7007/1 genPowerCfg from '0xe0798dfffe718dfc/1' failed (ZdoRequest error)
    at Driver.zdoRequest (/nix/store/dw1byj0i2hals5rinva1bphqzc74sdlp-zigbee2mqtt-1.28.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:551:19)
    at Object.func (/nix/store/dw1byj0i2hals5rinva1bphqzc74sdlp-zigbee2mqtt-1.28.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:518:13)
    at Queue.executeNext (/nix/store/dw1byj0i2hals5rinva1bphqzc74sdlp-zigbee2mqtt-1.28.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32))
Zigbee2MQTT:info  2023-01-07 12:12:53: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x00124b0024ce7007"},"error":"Failed to configure (Bind 0x00124b0024ce7007/1 genPowerCfg from '0xe0798dfffe718dfc/1' failed (ZdoRequest error))","status":"error","transaction":"1tv74-9"}'
[... 106 lines cut ...]
1673090982: Received PINGREQ from mqttjs_176cfa0d
1673090982: Sending PINGRESP to mqttjs_176cfa0d
1673091042: Received PINGREQ from mqttjs_176cfa0d
1673091042: Sending PINGRESP to mqttjs_176cfa0d
1673091102: Received PINGREQ from mqttjs_176cfa0d
1673091102: Sending PINGRESP to mqttjs_176cfa0d
1673091162: Received PUBLISH from mqttjs_176cfa0d (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (79 bytes))
1673091162: Sending PUBLISH to mqttjs_176cfa0d (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (79 bytes))
1673091222: Received PINGREQ from mqttjs_176cfa0d
1673091222: Sending PINGRESP to mqttjs_176cfa0d
1673091282: Received PINGREQ from mqttjs_176cfa0d
1673091282: Sending PINGRESP to mqttjs_176cfa0d
1673091342: Received PINGREQ from mqttjs_176cfa0d
1673091342: Sending PINGRESP to mqttjs_176cfa0d
1673091402: Received PINGREQ from mqttjs_176cfa0d
1673091402: Sending PINGRESP to mqttjs_176cfa0d
1673091462: Received PUBLISH from mqttjs_176cfa0d (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (79 bytes))
1673091462: Sending PUBLISH to mqttjs_176cfa0d (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (79 bytes))
1673091522: Received PINGREQ from mqttjs_176cfa0d
1673091522: Sending PINGRESP to mqttjs_176cfa0d
1673091582: Received PINGREQ from mqttjs_176cfa0d
1673091582: Sending PINGRESP to mqttjs_176cfa0d
1673091642: Received PINGREQ from mqttjs_176cfa0d
1673091642: Sending PINGRESP to mqttjs_176cfa0d
```

kizio1000, 2023-01-07:
I have two of them and couple of other SNZB-xx devices. They all suck (disappear from network, go off-line, etc) Usually it helps to remove battery before repairing. I was reading changelog of z2m v.1.29  and there was info that SNZB devices were fixed, so far I did not noticed any improvement in this aspect after updating to 1.29.1 which has by the way serous problems working with my sonoff E, so I went back to 1.28.4 which sucks at working with SNZB devices...

phodina, 2023-01-07:
Okay, thanks for the info. I'll try to remove the battery and repair the device. 

Do you have some other brand you'd recommend that's more reliable then?

phodina, 2023-01-07:
Also I found mention of the issues in this discussion https://github.com/Koenkk/zigbee2mqtt/discussions/10989

phodina, 2023-01-07:
Got there working for now by inserting some cardboard to push properly the coin cell battery

closed 2023-01-08 by phodina

kub3let, 2023-01-28:
I just migrated from tasmota2mqtt to zigbee2mqtt.

This is the first sensors that I connected and for some reason no matter what I do it says interview failed.

The battery has good connection, I bend the pins.

Everything is working, it reports open/close instantly and battery is displayed as well, so wondering what interview failed means and whether I should care about it ?