Koenkk/zigbee2mqtt #16008: [New device support]:

Seolur, 2023-01-07:
### Link

https://www.tervix.de/index.php/our-products/41-smart-home-zigbee/126-cras-sed-3

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0x2c1165fffe51456d","nwkAddr":58993,"manufId":4098,"manufName":"_TZE200_8thwkzxl","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_8thwkzxl","powerSource":3,"zclVersion":3,"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1673099364634,"defaultSendRequestWhen":"immediate"}

### Comments

Hi!
Zigbee Model
TS0601
Zigbee Manufacturer
_TZE200_8thwkzxl
Support status
Not supported

Zigbee2mqtt
Current version: 1.29.1-1

Home Assistant 2023.1.1
Supervisor 2022.12.1
Operating System 9.4
Frontend: 20230104.0 - latest

How this device can be supported? This is Tuya Thermostat, the same as other termostats, like this: 
https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html

Koenkk, 2023-01-08:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-08 by Koenkk

Seolur, 2023-01-20:
Thank you!
Is there are any plans to add this to Zigbee2MQTT?
Zigbee2MQTT Edge constantly crashes, it works not more than 5-10 minutes, then become crashed. Zigbee2MQTT with the same coordinator works good.