Koenkk/zigbee2mqtt #16010: TuYa IH012-RT01 Battery level isn't exposed as described in the device page https://www.zigbee2mqtt.io/devices/IH012-RT01.html

acaonweb, 2023-01-07:
### What happened?

[This Device](https://www.zigbee2mqtt.io/devices/IH012-RT01.html) is added and works but battery level and voltage aren't exposed,
[image]

Zigbee2MQTT version
[1.29.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.29.0) commit: [dddce571](https://github.com/Koenkk/zigbee2mqtt/commit/dddce571)
Coordinator type
zStack3x0
Coordinator revision

Coordinator IEEE Address
0x00124b0021b95272
Frontend version
0.6.121

### What did you expect to happen?

I expect that those datas are exposed

### How to reproduce it (minimal and precise)

pair it 

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20220219

### Adapter

Zzh! CC2652R Multiprotocol RF Stick

### Debug log

Debug 2023-01-07 16:15:24Retrieving state of 'Body_Sensor_Bagno_zb' after reconnect
Debug 2023-01-07 16:15:31Received Zigbee message from 'Body_Sensor_Bagno_zb', type 'attributeReport', cluster 'ssIasZone', data '{"61441":1,"currentZoneSensitivityLevel":1}' from endpoint 1 with groupID 0
Info 2023-01-07 16:15:31MQTT publish: topic 'zigbee2mqtt/Body_Sensor_Bagno_zb', payload '{"battery_low":false,"keep_time":60,"linkquality":0,"occupancy":true,"sensitivity":"medium","tamper":false}'
Debug 2023-01-07 16:15:32Received Zigbee message from 'Body_Sensor_Bagno_zb', type 'attributeReport', cluster 'ssIasZone', data '{"61441":1,"currentZoneSensitivityLevel":1}' from endpoint 1 with groupID 0
Info 2023-01-07 16:15:32MQTT publish: topic 'zigbee2mqtt/Body_Sensor_Bagno_zb', payload '{"battery_low":false,"keep_time":60,"linkquality":0,"occupancy":true,"sensitivity":"medium","tamper":false}'
Debug 2023-01-07 16:15:32Received Zigbee message from 'Body_Sensor_Bagno_zb', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
Info 2023-01-07 16:15:32MQTT publish: topic 'zigbee2mqtt/Body_Sensor_Bagno_zb', payload '{"battery_low":false,"keep_time":60,"linkquality":0,"occupancy":true,"sensitivity":"medium","tamper":false}'
Debug 2023-01-07 16:15:33Received Zigbee message from 'Body_Sensor_Bagno_zb', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
Info 2023-01-07 16:15:33MQTT publish: topic 'zigbee2mqtt/Body_Sensor_Bagno_zb', payload '{"battery_low":false,"keep_time":60,"linkquality":0,"occupancy":true,"sensitivity":"medium","tamper":false}'
Debug 2023-01-07 16:15:33Received Zigbee message from 'Body_Sensor_Bagno_zb', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0

Koenkk, 2023-01-08:
Not all versions of this device support the battery %, some only do battery_state (low/middle/high)

acaonweb, 2023-01-08:
ok, there is a way to get the state? thanx so much

Koenkk, 2023-01-09:
Can you provide the data/database.db entry of this device?

acaonweb, 2023-01-09:
I'm not so expert... how do that?

acaonweb, 2023-01-09:
{"id":29,"type":"EndDevice","ieeeAddr":"0xa4c13890a3865400","nwkAddr":28598,"manufId":4417,"manufName":"_TZ3040_6ygjfyll","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,4,1280,0],"outClusterList":[4,6,4096,25,10],"clusters":{"ssIasZone":{"attributes":{"61441":1,"currentZoneSensitivityLevel":1,"iasCieAddr":"0x00124b0021b95272","zoneState":1,"zoneStatus":1}},"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TZ3040_6ygjfyll","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1673297402040,"defaultSendRequestWhen":"immediate"}

acaonweb, 2023-01-09:
Is it what you asked for?
I'm not so expert

Koenkk, 2023-01-10:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0202', ['_TZ3000_mcxw5ehu', '_TZ3000_6ygjfyll', '_TZ3040_6ygjfyll']),
    model: 'IH012-RT01',
    vendor: 'TuYa',
    description: 'Motion sensor',
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.ignore_basic_report, fz.ZM35HQ_attr, fz.battery],
    toZigbee: [tz.ZM35HQ_attr],
    exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery(), e.battery_voltage(),
        exposes.enum('sensitivity', ea.ALL, ['low', 'medium', 'high']).withDescription('PIR sensor sensitivity'),
        exposes.enum('keep_time', ea.ALL, [30, 60, 120]).withDescription('PIR keep time in seconds'),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m
- open frontend -> device -> press yellow reconfigure button and make sure this succeeds. If not make sure to wakeup the device right before clicking the reconfigure button
- see if battery % now appears within 24 hours.

acaonweb, 2023-01-10:
Ok I'll do in next days because I'm abroad for work

acaonweb, 2023-01-10:
Ok now I cam see battery level in percentuage, nothing about voltage (still N/A)

Koenkk, 2023-01-12:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/16010#issuecomment-1377488099, please try again.

acaonweb, 2023-01-12:
Sorry, what I have to do?
Updating zigbee2mqtt?

acaonweb, 2023-01-13:
> Updated [#16010 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/16010#issuecomment-1377488099), please try again.

It's works!!!! 
Great!!!!

Koenkk, 2023-01-14:
Great, integrated the changes. You can keep using the external converter until the next release.

closed 2023-01-14 by Koenkk