Koenkk/zigbee2mqtt #16011: TS110E_2gang stopped reacting to "state": "toggle"

ikarisan, 2023-01-07:
### What happened?

I am using a "TS110E_2gang" dimmer (_TZ3210_3mpwqzuu) and sudenly it stopped reacting to the {"state": "toggle"} command. 
In the response the states change from

{
"state": "OFF",
"state_l1": "OFF",
"state_l2": "OFF",
}

to 

{
"state": "OFF",
"state_l1": "ON",
"state_l2": "OFF",
}

but nothing happens. 
When requesting the current state (zigbee2mqtt/Dimmer1/get) the state is still OFF. But the zigbee2mqtt GUI shows "ON".
If I use "state": "on" and "state": "off" instead the lights turn on and off. 

### What did you expect to happen?

I expected that the state of the light changes as it does when sending {"state": "on"/"off"}

### How to reproduce it (minimal and precise)

Pair a TuYa TS110E_2gang and send

{
  "state_l1": "toggle"
}

to zigbee2mqtt/friendlyname/set

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

20201010

### Adapter

CC2538+CC2592, zStack30x

### Debug log

This is the corresponding debug log:

```
debug 2023-01-07 17:32:12Received MQTT message on 'zigbee2mqtt/Dimmer1/set' with data '{ "state_l1": "toggle" }'
debug 2023-01-07 17:32:12Publishing 'set' 'state' to 'Dimmer1'
info 2023-01-07 17:32:13MQTT publish: topic 'zigbee2mqtt/Dimmer1', payload '{"brightness":253,"brightness_l1":254,"brightness_l2":255,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Dimmer1","hardwareVersion":1,"ieeeAddr":"0xa4c13801e3dcd9b5","manufacturerID":4417,"manufacturerName":"_TZ3210_3mpwqzuu","model":"TS110E_2gang","networkAddress":37924,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-01-07T17:32:13+01:00","linkquality":26,"max_brightness":255,"max_brightness_l1":255,"max_brightness_l2":255,"min_brightness":1,"min_brightness_l1":1,"min_brightness_l2":1,"power_on_behavior":"off","power_on_behavior_l1":"off","power_outage_memory_l1":"restore","power_outage_memory_l2":"restore","state":"OFF","state_l1":"OFF","state_l2":"OFF","switch_type":"toggle","switch_type_l1":"momentary","switch_type_l2":"momentary"}'
info 2023-01-07 17:32:13MQTT publish: topic 'zigbee2mqtt/Dimmer1', payload '{"brightness":253,"brightness_l1":254,"brightness_l2":255,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Dimmer1","hardwareVersion":1,"ieeeAddr":"0xa4c13801e3dcd9b5","manufacturerID":4417,"manufacturerName":"_TZ3210_3mpwqzuu","model":"TS110E_2gang","networkAddress":37924,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-01-07T17:32:13+01:00","linkquality":26,"max_brightness":255,"max_brightness_l1":255,"max_brightness_l2":255,"min_brightness":1,"min_brightness_l1":1,"min_brightness_l2":1,"power_on_behavior":"off","power_on_behavior_l1":"off","power_outage_memory_l1":"restore","power_outage_memory_l2":"restore","state":"OFF","state_l1":"ON","state_l2":"OFF","switch_type":"toggle","switch_type_l1":"momentary","switch_type_l2":"momentary"}'
```

ikarisan, 2023-01-08:
Extract of my database.db file

[device.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/10368708/device.json.txt)

ikarisan, 2023-01-08:
I think I found the reason for not reacting to the toggle copmmand.
zigbee2mqtt sends an unknown on/off command to the device.

When switching the light on via {"status": "on"} or {"state": "off"} it is translated to

Cluster: On/Off (0x0006)
Profile: Home Automation (0x0104)
[..]
Command: On (0x01)

or

Cluster: On/Off (0x0006)
Profile: Home Automation (0x0104)
[..]
Command: On (0x00)

But if I use "toggle" the folowing is send to the dimmer two times

Source Endpoint: 4
Destination Endpoint: 1
Cluster: On/Off (0x0006)
Profile: Home Automation (0x0104)
Counter: 187
[..]
Sequence Number: 80
Command: Unknown (0xfd)

Followed by:

Source Endpoint: 1
Destination Endpoint: 1
Cluster: On/Off (0x0006)
Profile: Home Automation (0x0104)
Counter: 220
[..]
Sequence Number: 177
Command: Toggle (0x02)

Followed by (again, 4 times):

Source Endpoint: 4
Destination Endpoint: 1
Cluster: On/Off (0x0006)
Profile: Home Automation (0x0104)
Counter: 188
[..]
Sequence Number: 80
Command: Unknown (0xfd)

ikarisan, 2023-01-08:
On request, I can provide a pcapng file.

ikarisan, 2023-01-11:
After sending

{"switch_type": "toggle"}
{"switch_type_l1": "toggle"}
{"switch_type_l2": "toggle"}
{"state": "off"}
{"state_l1": "off"}
{"state_l2": "off"}
{"state": "on", brightness: 254}
{"state_l1": "on", brightness: 254}
{"state_l2": "on", brightness: 254}

the 

{"switch_type_l1": "toggle"}

command works again.

ikarisan, 2023-01-11:
No, sorry. It only switches off the light.
Now, with every {"state": "toggle"} the dimmer turns on and off. But with every togle the "brightness" status is decreased by 1.
254...253...252...251...

joggs, 2023-02-11:
I also have problems with this device.
I cannot change states .

Zigbee Model: TS110E
Zigbee Manufacturer: _TZ3210_wdexaypg

Error: Publish 'set' 'state' to 'Hemmabio ' failed: 'Error: Command 0xa4c13840794a9452/1 genLevelCtrl.moveToLevelWithOnOff({"level":129,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 36645 - 1 - 167 - 8 - 11 after 10000ms)'

ikarisan, 2023-02-19:
For the moment I use a work around, implemented in a Node2Red flow. With every received message from the device I save the current state to a persistance file. Then at every "toggle" command from a button I read the state from file and generate the needed ON/OFF message for the dimmer. But I don't like this and the causative bug should be fixed.

closed 2023-04-01 by github-actions[bot]