Koenkk/zigbee2mqtt #16015: Temperature calibration unstable on xiaomi SP-EUC01

KimCarlsen, 2023-01-07:
### What happened?

After i adjusted temperature_calibration on my smart plug the readings are bumping up and down.
[image]
[image]

The spikes fits with the original value, as i adjusted the calibration with -4

HA version 2022.12.8
SP-EUC01 firmware date: 12-06-2021

### What did you expect to happen?

consistant readings

### How to reproduce it (minimal and precise)

set device_temperature_calibration to -4 (same with -5)

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20220219

### Adapter

Sonoff CC2652P

Koenkk, 2023-01-08:
Can you provide the debug log when the temperature jumps up?

KimCarlsen, 2023-01-08:
[z2m.txt](https://github.com/Koenkk/zigbee2mqtt/files/10368624/z2m.txt)
Affected device "Murphy"
15:12:04 temperature 11
15:12:30 temperature 6

The peaks comes every 30 min, down to the second

Koenkk, 2023-01-09:
Can you provide the debug log of the z2m start? (start till 1 min after start)

KimCarlsen, 2023-01-09:
[Z2MStart.txt](https://github.com/Koenkk/zigbee2mqtt/files/10375939/Z2MStart.txt)

Koenkk, 2023-01-09:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/553d5f0b200536c5d478a7f734499381

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

KimCarlsen, 2023-01-10:
Better. now the spikes are only one degree off

Koenkk, 2023-01-11:
I updated https://gist.github.com/Koenkk/553d5f0b200536c5d478a7f734499381, can you provide the debug logging when running with that external converter and getting the off by one degree?

KimCarlsen, 2023-01-13:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10413644/log.txt)

one degree spike at 17:43:13
Back to normal at 17:49:58

Koenkk, 2023-01-14:
Can you provide the logging again? I updated https://gist.github.com/Koenkk/553d5f0b200536c5d478a7f734499381

KimCarlsen, 2023-01-15:
Seems it has been fixed no. no spikes for 12 hours. thanks

KimCarlsen, 2023-02-15:
@Koenkk will this be part of the release?

Koenkk, 2023-02-15:
I don't understand how https://gist.github.com/Koenkk/553d5f0b200536c5d478a7f734499381 can fix this, this is the same code as in 1.30.1. How are you running z2m?

KimCarlsen, 2023-02-15:
Just tested with 1.30.1 (was using 1.30.0), but still it fails if i remove the 
" external_converters:
  - ext_converter.js"
It is running on windows server 2019 as a windows service.
Is there any file i can compare?

Koenkk, 2023-02-16:
After updating, did you execute `npm ci` in the zigbee2mqtt directory? Also, what is the content of the `package.json` file?

KimCarlsen, 2023-02-17:
okay. reinstalled everything and now it seems to work, so something must have been missing updates. thanks

Koenkk, 2023-02-18:
Great!

closed 2023-02-18 by Koenkk