Koenkk/zigbee2mqtt #16021: Legrand - 412015: Failed to configure 'powerMeter'

Ivano62, 2023-01-08:
### What happened?

From 1.29 release I get the error "Failed to configure 'powerMeter'"

### What did you expect to happen?

The device work correctly

### Zigbee2MQTT version

1.29.1-dev (commit: e773511)

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Debug log

Zigbee2MQTT:error 2023-01-08 09:38:15: Failed to configure 'powerMeter', attempt 1 (Error: ConfigureReporting 0x0004740000a11c40/1 haElectricalMeasurement([{"attribute":"apparentPower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22)
    at Object.apparentPower (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:207:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/legrand.js:314:13)
    at Configure.configure (/app/lib/extension/configure.ts:117:13)
    at Immediate.<anonymous> (/app/lib/extension/configure.ts:69:17))

Koenkk, 2023-01-08:
Already fixed by @sjorge  in https://github.com/Koenkk/zigbee-herdsman-converters/pull/5286

closed 2023-01-08 by Koenkk

Ivano62, 2023-01-08:
It isn't fixed, this is the after upgrade to 1.29.1-dev commit: 30df0e2<https://github.com/Koenkk/zigbee2mqtt/commit/30df0e2>

Using '/app/data' as data directory
Zigbee2MQTT:error 2023-01-08 09:56:45: Failed to configure 'powerMeter', attempt 1 (Error: ConfigureReporting 0x0004740000a11c40/1 haElectricalMeasurement([{"attribute":"apparentPower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22)
    at Object.apparentPower (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:207:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/legrand.js:314:13)
    at Configure.configure (/app/lib/extension/configure.ts:117:13)
    at Immediate.<anonymous> (/app/lib/extension/configure.ts:69:17))

Zigbee2MQTT:error 2023-01-08 09:57:49: Failed to configure 'powerMeter', attempt 2 (Error: Bind 0x0004740000a11c40/1 genIdentify from '0x00212effff0620d6/1' failed (waiting for response TIMEOUT)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:768:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/legrand.js:311:13)
    at Configure.configure (/app/lib/extension/configure.ts:117:13))

Zigbee2MQTT:error 2023-01-08 09:57:51: Failed to configure 'powerMeter', attempt 3 (Error: ConfigureReporting 0x0004740000a11c40/1 haElectricalMeasurement([{"attribute":"apparentPower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22)
    at Object.apparentPower (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:207:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/legrand.js:314:13)
    at Configure.configure (/app/lib/extension/configure.ts:117:13))

Closed #16021<https://github.com/Koenkk/zigbee2mqtt/issues/16021> as completed.

Koenkk, 2023-01-08:
`30df0e2` doesn't have this fix yet. Try again in 30 mins from now.