Koenkk/zigbee2mqtt #16026: [New device support]: QBKG28LM - Aqara H1 Pro switch (no neutral, two keys)

pyatins, 2023-01-08:
### Link

https://www.aqara.com/cn/d27_overview

### Database entry

{"id":9,"type":"EndDevice","ieeeAddr":"0x04cf8cdf3c8ddadb","nwkAddr":25310,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.l2acn1","epList":[1,2,41,42,51,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"dateCode":"05-17-2020","zclVersion":3}},"aqaraOpple":{"attributes":{"4":1,"238":3617,"240":0,"247":{"type":"Buffer","data":[100,16,1,101,16,0,3,40,26,5,33,65,118,154,32,0,10,33,198,244,11,32,0,13,35,33,14,0,0,14,35,0,0,0,0]},"512":0,"513":1}},"genOnOff":{"attributes":{"245":40033792,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"245":40033792,"onOff":0}},"aqaraOpple":{"attributes":{"512":0}}},"binds":[],"configuredReportings":[],"meta":{}},"41":{"epId":41,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"42":{"epId":42,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"51":{"epId":51,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":33,"stackVersion":2,"hwVersion":1,"dateCode":"05-17-2020","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2114939527},"lastSeen":1673118449122,"defaultSendRequestWhen":"immediate"}

### Comments

Manage to add Aqara H1 Pro two-key switch without neutral (model: QBKG28LM).

Photo of the switch with a white background:
[image]

Screenshot of the UI when using an external converter:
[image]

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const utils = require('zigbee-herdsman-converters/lib/utils');
const herdsman = require('zigbee-herdsman');

const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};

const cfz = {
    xiaomi_multistate_action: {
        cluster: 'genMultistateInput',
        type: ['attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            if (utils.hasAlreadyProcessedMessage(msg, model)) return;
            let actionLookup = {0: 'hold', 1: 'single', 2: 'double', 3: 'triple', 255: 'release'};
            if (model.model === 'WXKG12LM') {
                actionLookup = {...actionLookup, 16: 'hold', 17: 'release', 18: 'shake'};
            }

            let buttonLookup = null;
            if (['WXKG02LM_rev2', 'WXKG07LM', 'WXKG15LM', 'WXKG17LM'].includes(model.model)) {
                buttonLookup = {1: 'left', 2: 'right', 3: 'both'};
            }
            if (['QBKG12LM', 'QBKG24LM'].includes(model.model)) buttonLookup = {5: 'left', 6: 'right', 7: 'both'};
            if (['QBKG39LM', 'QBKG41LM', 'WS-EUK02', 'WS-EUK04', 'QBKG20LM', 'QBKG31LM', 'QBKG28LM'].includes(model.model)) {
                buttonLookup = {41: 'left', 42: 'right', 51: 'both'};
            }
            if (['QBKG25LM', 'QBKG26LM', 'QBKG34LM'].includes(model.model)) {
                buttonLookup = {
                    41: 'left', 42: 'center', 43: 'right',
                    51: 'left_center', 52: 'left_right', 53: 'center_right',
                    61: 'all',
                };
            }

            const action = actionLookup[msg.data['presentValue']];
            if (buttonLookup) {
                const button = buttonLookup[msg.endpoint.ID];
                if (button) {
                    return {action: `${action}_${button}`};
                }
            } else {
                return {action};
            }
[... 69 lines cut ...]
    vendor: 'Xiaomi',
    description: 'Aqara smart wall switch H1 Pro (no neutral, double rocker)',
    fromZigbee: [fz.on_off, cfz.xiaomi_multistate_action, fz.aqara_opple],
    toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode_opple, ctz.xiaomi_switch_power_outage_memory, tz.aqara_switch_mode_switch, tz.xiaomi_flip_indicator_light],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },
    exposes: [
        e.switch().withEndpoint('left'), e.switch().withEndpoint('right'), e.device_temperature(),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled']).withDescription('Decoupled mode for left button').withEndpoint('left'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled']).withDescription('Decoupled mode for right button').withEndpoint('right'),
        exposes.enum('mode_switch', ea.ALL, ['anti_flicker_mode', 'quick_mode']).withDescription('Anti flicker mode can be used to solve blinking issues of some lights.' + 'Quick mode makes the device respond faster.'),
        e.action(['single_left', 'double_left', 'single_right', 'double_right', 'single_both', 'double_both']),
        e.power_outage_memory(), e.flip_indicator_light()
    ],
    onEvent: preventReset,
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
    },
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

pyatins, 2023-02-15:
up

Koenkk, 2023-02-15:
Could you make a pr to add support for this device? (click the edit icon on https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/xiaomi.js)

pyatins, 2023-02-17:
I did it, but I made 3 pr (because I don't know how to change 3 files at once in one request):

- zigbee-herdsman-converters/devices/xiaomi.js
- zigbee-herdsman-converters/converters/fromZigbee.js
- zigbee-herdsman-converters/converters/toZigbee.js

Koenkk, 2023-02-18:
Merged, thanks!

closed 2023-02-18 by Koenkk