Koenkk/zigbee2mqtt #16035: Sonoff ZBMINI often turns off right after turning on by button in HA or button in Z2MQTT interface

jendrush, 2023-01-08:
### What happened?

Today I swapped the Sonoff Mini with Wifi for a Sonoff ZBMINI on Zigbee. I have the unit in the stairwell, so there are 2 stair switches and one cross switch. With the Mini it worked fine, but with the ZBMINI I have a problem. Namely, very often when I turn on the light with the button on the HA interface, or Z2MQTT it turns off in the same fraction of a second. Looking at the Zigbee2MQQ logs the following messages then occur:

- When it is off and I turn it on, and it turns itself off:

Info 2023-01-08 20:24:08MQTT publish: topic 'zigbee2mqtt/light mezzanine terrace', payload '{"last_seen": "2023-01-08T20:24:07+01:00", "linkquality":87, "state": "OFF"}'
Info 2023-01-08 20:24:08MQTT publish: topic 'zigbee2mqtt/mid-floor terrace light', payload '{"last_seen": "2023-01-08T20:24:07+01:00", "linkquality":87, "state": "ON"}'
Info 2023-01-08 20:24:09MQTT publish: topic 'zigbee2mqtt/ Light mezzanine terrace', payload '{"last_seen": "2023-01-08T20:24:07+01:00", "linkquality":91, "state": "OFF"}'

- when it is off and I turn it on and it stays on correctly:

'Info 2023-01-08 20:31:44MQTT publish: topic 'zigbee2mqtt/light mezzanine terrace', payload '{"last_seen": "2023-01-08T20:31:44+01:00", "linkquality":87, "state": "OFF"}'
Info 2023-01-08 20:31:44MQTT publish: topic 'zigbee2mqtt/mid-floor terrace light', payload '{"last_seen": "2023-01-08T20:31:44+01:00", "linkquality":87, "state": "ON"}'
Info 2023-01-08 20:31:45MQTT publish: topic 'zigbee2mqtt/ Light mezzanine terrace', payload '{"last_seen": "2023-01-08T20:31:44+01:00", "linkquality":87, "state": "ON"}'

- when it is on and I turn it off:

Info 2023-01-08 20:32:26MQTT publish: topic 'zigbee2mqtt/mid-floor terrace light', payload '{"last_seen": "2023-01-08T20:32:26+01:00", "linkquality":94, "state": "ON"}'
Info 2023-01-08 20:32:26MQTT publish: topic 'zigbee2mqtt/mid-floor terrace light', payload '{"last_seen": "2023-01-08T20:32:26+01:00", "linkquality":94, "state": "OFF"}'
Info 2023-01-08 20:32:27MQTT publish: topic 'zigbee2mqtt/ Light mezzanine terrace', payload '{"last_seen": "2023-01-08T20:32:26+01:00", "linkquality":94, "state": "OFF"}'

- when I turn on with a physical button and it turns on correctly:

2023-01-08 20:33:10MQTT publish: topic 'zigbee2mqtt/light mezzanine terrace', payload '{"last_seen": "2023-01-08T20:33:08+01:00", "linkquality":87, "state": "ON"}'

- when I turn off with a physical button and it shuts down correctly:

'Info 2023-01-08 20:33:21MQTT publish: topic 'zigbee2mqtt/light mezzanine terrace', payload '{"last_seen": "2023-01-08T20:33:19+01:00", "linkquality":91, "state": "OFF"}'

Sometimes it will also happen that when the light is off and the physical button is pressed, the light turns on for a moment and immediately goes off and there is only a message:

Info 2023-01-08 20:37:05MQTT publish: topic 'zigbee2mqtt/light mezzanine terrace', payload '{"last_seen": "2023-01-08T20:37:06+01:00", "linkquality":91, "state": "OFF"}'

This is the only switch/light I have this situation with, and I have over 50 Zigbee devices in all.

Translated with www.DeepL.com/Translator (free version)

### What did you expect to happen?

I expected that when I turn on the switch the light should turn off, and when I turn it off it should turn off. 

### How to reproduce it (minimal and precise)

Don't know how to reproduce. 

### Zigbee2MQTT version

1.29.1 

### Adapter firmware version

20211217

### Adapter

Sonoff ZBDongle-P ZigBee Dongle Plus CC2652P

closed 2023-02-16 by github-actions[bot]

georgyangelov, 2023-12-09:
@jendrush I have the same problem, did you manage to find a fix?

jendrush, 2023-12-09:
> @jendrush I have the same problem, did you manage to find a fix?

I changed the device. In the place where Zbmini did not work properly I installed QS-Zigbee-C01. I also recently bought a ZBMINIL2 and this one I can recommend, it works with momentary buttons, and with two-state buttons. It does not need a neutral wire and is even smaller.