Koenkk/zigbee2mqtt #16038: error after updating to last version

Erikwout, 2023-01-08:
### What happened?

I updated to the latest version(1.29.1-1) and received the following error in the log ;
app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

### What did you expect to happen?

work as normal with previous version (1.28.4-1)

### How to reproduce it (minimal and precise)

update

### Zigbee2MQTT version

1.29.1-1

### Adapter firmware version

Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20211217,"transportrev":2},"type":"zStack3x0"}'

### Adapter

Zigbee ethernet adapter

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10369620/log.txt)

cloudbr34k84, 2023-01-09:
@Erikwout are you still getting issues?? What sort of issues are you experiencing?

Erikwout, 2023-01-09:
Hi Brad
I have rolled back to previous version and now working great, nee version did not start at all
See error I pasted in the text

Reg
Erik

cloudbr34k84, 2023-01-09:
How do you roll back?

Erikwout, 2023-01-09:
I did a restore of the backup of the integration within home assistant, the one it makes  during the upgrade process (if you enable it😃)

cloudbr34k84, 2023-01-09:
Mmmm bugger

cloudbr34k84, 2023-01-09:
> is there anything i can do/help ?

I checked my backups and i had saved the latest back up. Just seeing how
that goes/

 thanks for the offer, its been terrible since the latest Z2m update..

Koenkk, 2023-01-09:
This error is caused by an incompatible external converter. Removing it from your `configuration.yaml` should fix this issue.

Erikwout, 2023-01-09:
How can I determine the incompatible converter?
As I  have not changed config but just updated the software itself
After roll back of software all working fine without any errors

Thanks
Erik

Koenkk, 2023-01-10:
By removing the external converters from your `configuration.yaml` one-by-one. External converters are only meant for temporarily usage as they can break over release. If you post the converter here I can integrate it for out-of-the-box support.

Erikwout, 2023-01-11:
Hi Koen
i will try this a bit later, but please find 1 zip with files that I created to get my 2 HUE lamps working
external_converters: 
  - 929003115801.js
  - 929003116201.js
[My converters.zip](https://github.com/Koenkk/zigbee2mqtt/files/10391295/My.converters.zip)

Koenkk, 2023-01-12:
@Erikwout added both!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Erikwout, 2023-01-15:
Hi Koen,
Just installed dev branch and working fine, thanks for the help
Reg
Erik

Koenkk, 2023-01-16:
Great!

closed 2023-01-16 by Koenkk