Koenkk/zigbee2mqtt #16043: '(0xc7: NWK_TABLE_FULL) - what does this mean?

cloudbr34k84, 2023-01-09:
### What happened?

Error 2023-01-09 15:39:08Publish 'set' 'state' to 'Powder Room Light Switches' failed: 'Error: Command 0xa4c1389a34414fb0/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":39275,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'

### Zigbee2MQTT version

1.29.0 commit: unknown

### Adapter firmware version

20220219

### Adapter

Sonoff Coordinator P

sjorge, 2023-01-09:
It means the network table on the coordinator is full. This might be fixed in the upcoming firmware release, which I think is happening soon.

cloudbr34k84, 2023-01-10:
> It means the network table on the coordinator is full. This might be fixed in the upcoming firmware release, which I think is happening soon.

this has never been an issue before and i once had 140 zigbee devices now 122??

closed 2023-01-14 by cloudbr34k84

Orochica, 2023-02-19:
I get the same error
`NWK_TABLE_FULL`

DenDeXTeR51, 2023-04-28:
How many devices you have?

cloudbr34k84, 2023-04-29:
> How many devices you have?

130 devices