Koenkk/zigbee2mqtt #16044: [New device support]:

VoltickVL, 2023-01-09:
### Link

https://aliexpress.ru/item/1005004352027105.html?spm=a2g2w.orderdetail.0.0.2d174aa6ejwWSN&sku_id=12000029623915182

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0x003c84fffef71d75","nwkAddr":60068,"manufId":4098,"manufName":"_TZE200_5toc8efa","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0017�N+\u0013!�N+\u0012\"�N+\u0012","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_5toc8efa","stackVersion":0,"dateCode":"","appVersion":72}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1673253884958,"defaultSendRequestWhen":"immediate"}

### Comments

I use dev channel (Zigbee2Mqtt Edge) and add new external converter for TZE200_5toc8efa (moes.js).

I have problem with Bseed thermostat TS0601 TZE200_5toc8efa electric floor heating at Zigbee2Mqtt Edge.
But Bseed thermostat TS0601 TZE200_aoclfnxz work fine.

When i set heat to 30 *C, my device select 3*C. When i set 20*C, my device select 2 *C. When i set 25*C, my device select 2.5*C. 

Please repair it or tell me what i doing wrong.

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const zosung = require('zigbee-herdsman-converters/lib/zosung');
const fzZosung = zosung.fzZosung;
const tzZosung = zosung.tzZosung;
const ez = zosung.presetsZosung;

const exposesLocal = {
    hour: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('h').withValueMin(0).withValueMax(23),
    minute: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('m').withValueMin(0).withValueMax(59),
    program_temperature: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('°C')
        .withValueMin(5).withValueMax(35).withValueStep(0.5),
};

module.exports = [
    {
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_cymsnfvf'},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_2xlvlnez'}],
        model: 'ZP-LZ-FR2U',
        vendor: 'Moes',
        description: 'Zigbee 3.0 dual USB wireless socket plug',
        extend: tuya.extend.switch({powerOutageMemory: true, indicatorMode: true, childLock: true, endpoints: ['l1', 'l2']}),
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2};
        },
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(device.getEndpoint(1));
            await reporting.onOff(device.getEndpoint(2));
        },
    },
    {
        fingerprint: [{modelID: 'TS0121', manufacturerName: '_TYZB01_iuepbmpv'}, {modelID: 'TS011F', manufacturerName: '_TZ3000_zmy1waw6'},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_bkfe0bab'}],
        model: 'MS-104Z',
        description: 'Smart light switch module (1 gang)',
        vendor: 'Moes',
        extend: tuya.extend.switch({powerOnBehavior: true}),
[... 339 lines cut ...]
        ],
        toZigbee: [tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code],
        exposes: [ez.learn_ir_code(), ez.learned_ir_code(), ez.ir_code_to_send(), e.battery(), e.battery_voltage()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genPowerCfg', ['batteryVoltage', 'batteryPercentageRemaining']);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.batteryVoltage(endpoint);
        },
    },
    {
        fingerprint: [{modelID: 'TS0011', manufacturerName: '_TZ3000_hhiodade'}],
        model: 'ZS-EUB_1gang',
        vendor: 'Moes',
        description: 'Wall light switch (1 gang)',
        extend: tuya.extend.switch({backlightModeLowMediumHigh: true, switchType: true}),
        configure: async (device, coordinatorEndpoint, logger) => {
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            device.powerSource = 'Mains (single phase)';
            device.save();
        },
    },
];
```

VoltickVL, 2023-01-09:
Log:

> 
Warning 2023-01-09 18:45:11zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,250]}}
Warning 2023-01-09 18:46:11zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,245]}}
Warning 2023-01-09 18:47:11zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,250]}}
Warning 2023-01-09 18:48:11zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,250]}}
Warning 2023-01-09 18:49:11zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,245]}}
Warning 2023-01-09 18:50:10zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,250]}}
Warning 2023-01-09 18:51:10zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,250]}}
>

VoltickVL, 2023-01-09:
LOG:
```shell
Warning 2023-01-09 21:31:11 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:31:50 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:31:53 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:32:11 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:32:46 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:32:49 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:33:17 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:34:18 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,19]}}
```

VoltickVL, 2023-01-10:
I update external converter and now when i set temp to 25 *C at device set same temp, but in z2m i see value 250.
I am not a programmer, please tell me how to edit the code so that the temperature of 25 degrees Celsius and not 250 degrees Celsius is displayed in z2m... Pls help me(

VoltickVL, 2023-01-15:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5334

VoltickVL, 2023-01-20:
Добрый день мне пришлось внести изменения в свой external_converter чтобы всё работало. Автор не принимает мои правки в файлы From ZigBee.js и ToZigbee.js видимо потому что там есть небольшие ошибки в коде так как я не программист ява. На мои просьбы довести код "до ума" автор не реагирует. Поэтому я скопировал части кода из файлов From ZigBee.js и ToZigbee.js в свой external_converter и исправил в тех местах где мне казалось, что была ошибка. И теперь всё работает как надо ! Если в теме будут грамотные JS-кодеры внесите пожалуйста правки основываясь на моём коде в From ZigBee.js и ToZigbee.js.  

====
Good afternoon, I had to make changes to my external_converter to make everything work. The author does not accept my edits to files From ZigBee.js and ToZigbee.js apparently because there are small errors in the code since I am not a Java programmer. The author does not respond to my requests to bring the code "to mind". So I copied parts of the code from the files From ZigBee.js and ToZigbee.js in its external_converter and fixed in those places where it seemed to me that there was a mistake. And now everything works as it should ! If there are competent JS coders in the topic, please make edits based on my code in From ZigBee.js and ToZigbee.js .

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposesLocal = {
    hour: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('h').withValueMin(0).withValueMax(23),
    minute: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('m').withValueMin(0).withValueMax(59),
    program_temperature: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('°C')
        .withValueMin(5).withValueMax(35).withValueStep(0.5),
};
const moes_thermostat = {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'moes_thermostat');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
            let temperature;
            switch (dp) {
            case tuya.dataPoints.moesSchedule:
                return {
                    program: {
                        weekdays_p1_hour: value[0],
                        weekdays_p1_minute: value[1],
                        weekdays_p1_temperature: value[2] / 2,
                        weekdays_p2_hour: value[3],
                        weekdays_p2_minute: value[4],
                        weekdays_p2_temperature: value[5] / 2,
                        weekdays_p3_hour: value[6],
                        weekdays_p3_minute: value[7],
                        weekdays_p3_temperature: value[8] / 2,
                        weekdays_p4_hour: value[9],
                        weekdays_p4_minute: value[10],
                        weekdays_p4_temperature: value[11] / 2,
                        saturday_p1_hour: value[12],
                        saturday_p1_minute: value[13],
                        saturday_p1_temperature: value[14] / 2,
                        saturday_p2_hour: value[15],
                        saturday_p2_minute: value[16],
                        saturday_p2_temperature: value[17] / 2,
                        saturday_p3_hour: value[18],
                        saturday_p3_minute: value[19],
                        saturday_p3_temperature: value[20] / 2,
                        saturday_p4_hour: value[21],
                        saturday_p4_minute: value[22],
                        saturday_p4_temperature: value[23] / 2,
                        sunday_p1_hour: value[24],
                        sunday_p1_minute: value[25],
                        sunday_p1_temperature: value[26] / 2,
                        sunday_p2_hour: value[27],
                        sunday_p2_minute: value[28],
                        sunday_p2_temperature: value[29] / 2,
                        sunday_p3_hour: value[30],
                        sunday_p3_minute: value[31],
                        sunday_p3_temperature: value[32] / 2,
                        sunday_p4_hour: value[33],
                        sunday_p4_minute: value[34],
                        sunday_p4_temperature: value[35] / 2,
                    },
                };
[... 110 lines cut ...]
                .withFeature(exposesLocal.program_temperature('sunday_p1_temperature'))
                .withFeature(exposesLocal.hour('sunday_p2_hour'))
                .withFeature(exposesLocal.minute('sunday_p2_minute'))
                .withFeature(exposesLocal.program_temperature('sunday_p2_temperature'))
                .withFeature(exposesLocal.hour('sunday_p3_hour'))
                .withFeature(exposesLocal.minute('sunday_p3_minute'))
                .withFeature(exposesLocal.program_temperature('sunday_p3_temperature'))
                .withFeature(exposesLocal.hour('sunday_p4_hour'))
                .withFeature(exposesLocal.minute('sunday_p4_minute'))
                .withFeature(exposesLocal.program_temperature('sunday_p4_temperature')),
        ],
[... cut ...]
            [103, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [104, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [105, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [106, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [107, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDP],
        ],
    },
};
module.exports = definition;
```

VoltickVL, 2023-01-20:
Единственное мне не удалось решить проблему с ошибкой Datapoints, но и с ней всё работает как нужно. Если кто разберется дайте знать!

===
The only thing I could not solve the problem with the Datapoints error, but everything works as it should. If anyone understands, let me know!

```shell
Warning 2023-01-09 21:31:11 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:31:50 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:31:53 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:32:11 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:32:46 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:32:49 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:33:17 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:34:18 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,19]}}
Warning 2023-01-10 00:04:21 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,4]}}
Warning 2023-01-10 00:05:21 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,9]}}
```

VladimirTuzovGitHub, 2023-01-20:
Приветствую! Я немного внес изменения в Ваш конвертор, поправил остальные данные на основе Вашего шаблона которые тоже выводились в десятых, это: deadzone_temperature, max_temperature_limit, min_temperature_limit. Но как выяснилось есть проблема с deadzone_temperature она не передается с устройства а просто дублирует min_temperature_limit, видимо конвертор не верно читает datapoints и видимо ошибка Unrecognized DP #102 with data и есть datapoints 102 который относится к deadzone_temperature и его нужно определить в конверторе.
[Bseed_zigbee_ thermostat.txt](https://github.com/Koenkk/zigbee2mqtt/files/10465211/Bseed_zigbee_.thermostat.txt)

VoltickVL, 2023-01-20:
> Приветствую! Я немного внес изменения в Ваш конвертор, поправил остальные данные на основе Вашего шаблона которые тоже выводились в десятых, это: deadzone_temperature, max_temperature_limit, [...]

Отлично! Осталось ещё не много я думаю, мы на верном пути. Взял ваш код, даже картинка Bseed, а не Moes. Отлично !Спасибо !

VoltickVL, 2023-01-30:
> Приветствую! Я немного внес изменения в Ваш конвертор, поправил остальные данные на основе Вашего шаблона которые тоже выводились в десятых, это: deadzone_temperature, max_temperature_limit, [...]

Удалось победить ошибку 102?

elemarek, 2023-02-21:
Does this thermostat suffer from the same defect as Moes brand thermostats such as congestion with an enormous number of packets?

closed 2023-04-01 by github-actions[bot]

kamilnuriev, 2023-06-03:
в 1.31.0 значение "local_temperature" и "max_temperature" в 10 раз выше, прошу исправить!

```
{
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 18,
    "deadzone_temperature": 5,
    "heat": "OFF",
    "linkquality": 120,
    "local_temperature": 285,
    "local_temperature_calibration": 0,
    "max_temperature": 350,
    "max_temperature_limit": 35,
    "min_temperature_limit": 5,
    "preset": "hold",
    "preset_mode": "hold",
    "program": {
        "saturday_p1_hour": 200,
        "saturday_p1_minute": 17,
        "saturday_p1_temperature": 15,
        "saturday_p2_hour": 0,
        "saturday_p2_minute": 200,
        "saturday_p2_temperature": null,
        "saturday_p3_temperature": null,
        "saturday_p4_temperature": null,
        "sunday_p1_temperature": null,
        "sunday_p2_temperature": null,
        "sunday_p3_temperature": null,
        "sunday_p4_temperature": null,
        "weekdays_p1_hour": 7,
        "weekdays_p1_minute": 6,
        "weekdays_p1_temperature": 0,
        "weekdays_p2_hour": 0,
        "weekdays_p2_minute": 200,
        "weekdays_p2_temperature": 5.5,
        "weekdays_p3_hour": 30,
        "weekdays_p3_minute": 0,
        "weekdays_p3_temperature": 100,
        "weekdays_p4_hour": 13,
        "weekdays_p4_minute": 30,
        "weekdays_p4_temperature": 0
    },
    "running_state": "idle",
    "sensor": "OU",
    "system_mode": "off"
}
```

VoltickVL, 2023-08-30:
in 1.31.0 "local_temperature" and "max_temperature" are 10 times higher, please fix

Same problem...  I create new issue https://github.com/Koenkk/zigbee2mqtt/issues/18791

VoltickVL, 2023-09-03:
Plz tell me, who fix this problem(

VoltickVL, 2023-09-03:
> Приветствую! Я немного внес изменения в Ваш конвертор, поправил остальные данные на основе Вашего шаблона которые тоже выводились в десятых, это: deadzone_temperature, max_temperature_limit, [...]

Подскажите у вас получилось решить проблему? Внешний конвертер не работает( Последний апдейт z2m не помог

0anton, 2023-10-26:
> Does this thermostat suffer from the same defect as Moes brand thermostats such as congestion with an enormous number of packets?

Yes, it does.

alekslegkih, 2024-11-29:
Кто ни будь может объяснить как сейчас подключить конвертор?
Файл расширением js и ts и длинное название и короткое, все равно термостат Moes BHT-002-GCLZB