Koenkk/zigbee2mqtt #16049: [Feature request]: change `systemd` init script behaviour to fail if not connected with mqtt server

IgnacioHR, 2023-01-09:
### Is your feature request related to a problem? Please describe

The problem I'm facing is as follows:

1. There are plenty of users reporting that when starting zigbee2mqtt the frontend does not start, I faced the same today.
2. I noticed that the reason behind this issue is not a problem in the frontend but another problem connecting to `mqtt` server and the front-end not started because it is started after the connection with mqtt server is made. I don't know why connection with `mqtt` did not success but this is NOT the focus of this request. I'll fix that on my own later if I can!
3. It looks like the connection with `mqtt` server loops forever internally, it never fails with timeout?
4. When running `systemctl start zigbee2mqtt` it returns immediately without waiting for the services to be operative at all so it is really difficult to detect a failed start.

Additional information:

Here, the service shows as `active (running)` but this is not true.
```
root@z2mqtt:/opt/zigbee2mqtt/data/log/2023-01-09.09-40-54# systemctl status zigbee2mqtt
* zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-01-09 10:11:21 UTC; 8s ago
   Main PID: 46392 (npm start)
      Tasks: 23 (limit: 38063)
     Memory: 82.6M
        CPU: 1.813s
     CGroup: /system.slice/zigbee2mqtt.service
             |-46392 npm start
             |-46403 sh -c -- node index.js
             `-46404 node index.js

Jan 09 10:11:21 z2mqtt systemd[1]: Started zigbee2mqtt.
Jan 09 10:11:21 z2mqtt npm[46392]: > zigbee2mqtt@1.29.1 start
Jan 09 10:11:21 z2mqtt npm[46392]: > node index.js
Jan 09 10:11:23 z2mqtt npm[46404]: Zigbee2MQTT:info  2023-01-09 10:11:23: Logging to console and directory: '/opt/zigbee2mqtt/data/l>
Jan 09 10:11:23 z2mqtt npm[46404]: Zigbee2MQTT:info  2023-01-09 10:11:23: Starting Zigbee2MQTT version 1.29.1 (commit #7d67ffc)
Jan 09 10:11:23 z2mqtt npm[46404]: Zigbee2MQTT:info  2023-01-09 10:11:23: Starting zigbee-herdsman (0.14.83)
```

`netstat` indicates the frontend has not started
```
root@z2mqtt:/opt/zigbee2mqtt# netstat -anp | grep 8080
root@z2mqtt:/opt/zigbee2mqtt#
```

log indicates:
```
root@z2mqtt:/opt/zigbee2mqtt/data/log/2023-01-09.10-11-23# cat log.txt
info  2023-01-09 10:11:23: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-01-09.10-11-23' filename: log.txt
info  2023-01-09 10:11:23: Starting Zigbee2MQTT version 1.29.1 (commit #7d67ffc)
info  2023-01-09 10:11:23: Starting zigbee-herdsman (0.14.83)
```

When issuing `systemctl stop zigbee2mqtt` the log says:
```
root@z2mqtt:/opt/zigbee2mqtt/data/log/2023-01-09.10-11-23# cat log.txt
info  2023-01-09 10:11:23: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-01-09.10-11-23' filename: log.txt
info  2023-01-09 10:11:23: Starting Zigbee2MQTT version 1.29.1 (commit #7d67ffc)
info  2023-01-09 10:11:23: Starting zigbee-herdsman (0.14.83)
error 2023-01-09 10:14:45: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
    at Frontend.stop (/opt/zigbee2mqtt/lib/extension/frontend.ts:83:39)
    at Controller.callExtensions (/opt/zigbee2mqtt/lib/controller.ts:316:40)
    at Controller.stop (/opt/zigbee2mqtt/lib/controller.ts:193:9)
    at stop (/opt/zigbee2mqtt/index.js:111:5)
    at process.handleQuit (/opt/zigbee2mqtt/index.js:117:9))
```

So, the frontend has not started and no connection was made to `mqtt` server and the init script indicates the service has started and is OK which is not right.

### Describe the solution you'd like

Review the init script behaviour so issuing systemctl status can tell the user if the service is really started or if it is still "starting" 

### Describe alternatives you've considered

An alternative is to change how the service starts so, if the frontend is active, it should be always started prior to the connection to the `mqtt` server. The frontend web page should include ways to tell the user if the connection to `mqtt` server was successful or not.

### Additional context

I'm in the process of migrating my `zigbee` network at from from deCONZ to zigbee2mqtt because the number of supported devices is higher. I've to confess i'm having a hard time trying to put all things together and made a robust infrastructure. 

Hopefully, this can be turned into an opportunity for some improvements!

Cheers
Ignacio

IgnacioHR, 2023-02-09:
ping (just to keep this open for longer)

junosuarez, 2023-03-13:
I think this would be a great enhancement. The Docker solution (adding a HEALTHCHECK command) would be different than the systemd service, but they could share a common definition of what "healthy" means - from looking at startup logs, it seems to be "can connect to coordinator" and "can connect to mqtt"

closed 2023-04-22 by github-actions[bot]