Koenkk/zigbee2mqtt #16054: unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ytibqbra'

saulleg, 2023-01-09:
### Link

https://es.aliexpress.com/item/1005001626291641.html

### Database entry

Device '0xa4c1385220b21e6a' with Zigbee model 'TS0601' and manufacturer name '_TZE200_ytibqbra' is NOT supported,

### Comments

Zigbee2MQTT:info  2023-01-09 13:23:13: Successfully interviewed '0xa4c1385220b21e6a', device has successfully been paired
Zigbee2MQTT:warn  2023-01-09 13:23:13: Device '0xa4c1385220b21e6a' with Zigbee model 'TS0601' and manufacturer name '_TZE200_ytibqbra' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:info  2023-01-09 13:23:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c1385220b21e6a","ieee_address":"0xa4c1385220b21e6a","status":"successful","supported":false},"type":"device_interview"}'
Zigbee2MQTT:debug 2023-01-09 13:23:16: Received Zigbee message from '0xa4c1385220b21e6a', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-01-09 13:23:16: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ytibqbra'
Zigbee2MQTT:warn  2023-01-09 13:23:16: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2023-01-09 13:23:19: Received Zigbee message from '0xa4c1385220b21e6a', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":0,"appVersion":72}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-01-09 13:23:19: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ytibqbra'
Zigbee2MQTT:warn  2023-01-09 13:23:19: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2023-01-09 13:23:20: Received Zigbee message from '0xa4c1385220b21e6a', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1}],"seq":2048}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-01-09 13:23:20: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ytibqbra'
Zigbee2MQTT:warn  2023-01-09 13:23:20: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2023-01-09 13:23:20: Received Zigbee message from '0xa4c1385220b21e6a', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":2304}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-01-09 13:23:20: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ytibqbra'
Zigbee2MQTT:warn  2023-01-09 13:23:20: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2023-01-09 13:23:20: Received Zigbee message from '0xa4c1385220b21e6a', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":16}],"seq":2560}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-01-09 13:23:20: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ytibqbra'
Zigbee2MQTT:warn  2023-01-09 13:23:20: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2023-01-09 13:23:20: Received Zigbee message from '0xa4c1385220b21e6a', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":17}],"seq":2816}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-01-09 13:23:20: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ytibqbra'
Zigbee2MQTT:warn  2023-01-09 13:23:20: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2023-01-09 13:23:20: Received Zigbee message from '0xa4c1385220b21e6a', type 'attributeReport', cluster 'genBasic', data '{"65503":"��N+f��N+i��N+\u0012"}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-01-09 13:23:20: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ytibqbra'
Zigbee2MQTT:warn  2023-01-09 13:23:20: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2023-01-09 13:23:21: Received Zigbee message from '0xa4c1385220b21e6a', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-01-09 13:23:21: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ytibqbra'

menzbenz-jeedom, 2023-01-13:
Hello, I have the same issue with _TZE204_ztc6ggyl . My zigbee network is spammed. 
I use this universal box : https://www.domadoo.fr/fr/box-domotique/5878-jeedom-controleur-domotique-jeedom-atlas-zigbee.html
Is there any solution to apply from any side (client or server  Zigbee2MQTT,MQTT,Mosquitto-PHP (i don't no which is client or server ,I'm sorry )) in order to ignore those annoying message please ? 

`Received Zigbee message from 'detecteur de presence CH', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,215],"type":"Buffer"},"datatype":2,"dp":9}],"seq":50176}' from endpoint 1 with groupID 0
Debug 2023-01-13 17:07:53Received Zigbee message from 'detecteur de presence CH', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,216],"type":"Buffer"},"datatype":2,"dp":9}],"seq":50432}' from endpoint 1 with groupID 0
Debug 2023-01-13 17:07:54Received Zigbee message from 'detecteur de presence CH', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,224],"type":"Buffer"},"datatype":2,"dp":9}],"seq":50688}' from endpoint 1 with groupID 0
Debug 2023-01-13 17:07:55Received Zigbee message from 'detecteur de presence CH', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,239],"type":"Buffer"},"datatype":2,"dp":9}],"seq":50944}' from endpoint 1 with groupID 0
Debug 2023-01-13 17:07:56Received Zigbee message from 'detecteur de presence CH', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,1],"type":"Buffer"},"datatype":2,"dp":9}],"seq":51200}' from endpoint 1 with groupID 0
Debug 2023-01-13 17:07:57Received Zigbee message from 'detecteur de presence CH', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,33],"type":"Buffer"},"datatype":2,"dp":9}],"seq":51456}' from endpoint 1 with groupID 0
Debug 2023-01-13 17:07:58Received Zigbee message from 'detecteur de presence CH', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,42],"type":"Buffer"},"datatype":2,"dp":9}],"seq":51712}' from endpoint 1 with groupID 0
Debug 2023-01-13 17:07:59Received Zigbee message from 'detecteur de presence CH', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,36],"type":"Buffer"},"datatype":2,"dp":9}],"seq":51968}' from endpoint 1 with groupID 0
Debug 2023-01-13 17:08:00Received Zigbee message from 'detecteur de presence CH', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,231],"type":"Buffer"},"datatype":2,"dp":9}],"seq":52224}' from endpoint 1 with groupID 0
Debug 2023-01-13 17:08:01Received Zigbee message from 'detecteur de presence CH', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,227],"type":"Buffer"},"datatype":2,"dp":9}],"seq":52480}' from endpoint 1 with groupID 0
Debug 2023-01-13 17:08:02Received Zigbee message from 'detecteur de presence CH', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,228],"type":"Buffer"},"datatype":2,"dp":9}],"seq":52736}' from endpoint 1 with groupID 0
Debug 2023-01-13 17:08:03Received Zigbee message from 'detecteur de presence CH', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,226],"type":"Buffer"},"datatype":2,"dp":9}],"seq":52992}' from endpoint 1 with groupID 0
Debug 2023-01-13 17:08:04`

Thank you!

closed 2023-02-20 by github-actions[bot]

Yakie996, 2023-03-26:
Hi, has someone been able to make this supported?