Koenkk/zigbee2mqtt #16057: Unit of measurement for VOC seems to be wrong with current Home Assistant

jmerifjKriwe, 2023-01-09:
### What happened?

Seems that the unit of measurement for VOC shouldn't be 'ppb' if you use the current version of Home Assistant (2023.1). 

### What did you expect to happen?

According to https://www.home-assistant.io/integrations/sensor/ it's only allowed to be:

`volatile_organic_compounds: Concentration of volatile organic compounds in µg/m³`

### How to reproduce it (minimal and precise)

Use any sensor with voc, e.g. Xiaomi VOCKQJK11LM

will result in:
`Entity sensor.mysensor_voc (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'ppb' which is not a valid unit for the device class ('volatile_organic_compounds') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22`

### Zigbee2MQTT version

1.29.1 commit: 7d67ffc

### Adapter firmware version

20221102

### Adapter

zStack3x0

sjorge, 2023-01-09:
ppb -> µg/m3 is not an easy conversion as you need to know stuff about the exact gas mix it is detecting.

I gave up when adding support for https://www.develcoproducts.com/products/sensors-and-alarms/air-quality-sensor/, it also reports ppb and it even has a table but none of the formulas I found even came close to the table they listed.

I imagine it would be similar for the Xiaomi one.

dreimer1986, 2023-01-25:
New Beta 2023.2.0B0 now shows this Warning:

```
Entity sensor.co2sensor_voc (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'ppb' which is not a valid unit for the device class ('volatile_organic_compounds') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```

If it is not convertible, maybe the better idea would be to talk to the HA authors?

IvovanWilligen, 2023-02-14:
If it is not convertible, maybe the better idea would be to talk to the HA authors?

Is second this solution :)

closed 2023-03-04 by Koenkk

Koenkk, 2023-03-05:
Pushed a fix for this, `device_class` will be removed now if `unit_of_measurement` != `ug/m3`.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

satter, 2023-03-16:
I switched my VOCKQJK11LM sensor to display VOC in mg/m3 on screen and compared it with ppb value reported via zigbee

ppb | mg/m3 | coef ppb-> μg/m3
-- | -- | --
312 | 0.71 | 2.28
306 | 0.7 | 2.29
300 | 0.69 | 2.30
218 | 0.5 | 2.29
1304 | 2.99 | 2.29
2111 | 4.84 | 2.29
126 | 0.29 | 2.30
108 | 0.25 | 2.31
92 | 0.21 | 2.28
79 | 0.18 | 2.28
56 | 0.13 | 2.32
49 | 0.11 | 2.24
42 | 0.1 | 2.38
35 | 0.09 | 2.57
19 | 0.04 | 2.11
33 | 0.08 | 2.42
84 | 0.19 | 2.26
92 | 0.21 | 2.28
69 | 0.15 | 2.17

On average coefficient is 2.3, on lower values variance is greater  
I will try to record more data with different temperatures, but 2.3 is looks fine for me to convert ppb ->  μg/m3