Koenkk/zigbee2mqtt #16060: [New device support]:

LukasMendez, 2023-01-09:
### Link

https://www.aliexpress.com/item/1005004705172997.html?spm=a2g0o.order_list.order_list_main.85.21ef1802lYK1Az

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0xa4c13863968a30e9","nwkAddr":44147,"manufId":4417,"manufName":"_TZ3000_cipibmnp","powerSource":"DC Source","modelId":"TS0224","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[1,3,4,5,6,768,1280,1282,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0224","manufacturerName":"_TZ3000_cipibmnp","powerSource":132,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x0000000000000000","zoneState":0}},"genOnOff":{"attributes":{"onOff":0}},"ssIasWd":{"attributes":{"1":100,"2":50,"4":0,"5":0,"6":50,"57344":0,"maxDuration":60}},"lightingColorCtrl":{"attributes":{"57600":[1000,0],"tuyaRgbMode":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1673283059724,"defaultSendRequestWhen":"immediate"}

### Comments

Model: TS0224 from iAlarm

[image]

In Smart Life you have to create a scene and activate the scene in order to start the alarm.

LukasMendez, 2023-01-14:
I've tried to make my own converter using the documentation for supporting new devices. I have a Tuya Zigbee Gateway and was able to get all of the datapoints / code numbers. 

This is the custom converter I've created: 

```const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0224',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_cipibmnp'
        },
    ],
    model: 'TS0224',
    vendor: 'TuYa',
    description: 'Tuya Smart Light & Sound Siren ',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        exposes.binary('alarm', ea.STATE_SET, true, false).withDescription('Enable the alarm'),
        exposes.enum('alarm_volume', ea.ALL, ['low', 'middle', 'high', 'mute']).withDescription('Volume of siren'),
        exposes.numeric('alarm_time', ea.STATE_SET).withValueMin(60).withValueMax(3600).withValueStep(1)
                .withUnit('s').withDescription('Alarm time')],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'alarm', tuya.valueConverterBasic.lookup({false: tuya.enum(3), true: tuya.enum(2)})], // <- mapped the datapoint
            [5, 'alarm_volume', tuya.valueConverter.raw],
            [7, 'alarm_time', tuya.valueConverter.raw],
        ],
    }
};

module.exports = definition;
```

I can now see the exposes, but none of them seem to work. I keep getting this "No converter available for 'x'" error message. 

[image]

These are the options I have when going to Tuya IoT Cloud and debug: 

[image]

Does anyone have any suggestion? Perhaps @Koenkk

Koenkk, 2023-01-15:
Below `const definition`, add:

```
const tzDatapoints = {
  ...tuya.tz.datapoints,
  key: ['alarm', 'alarm_volume', 'alarm_time'],
}
```

and change `toZigbee: [tuya.tz.datapoints],` to `toZigbee: [tzDatapoints],`

LukasMendez, 2023-01-15:
> tzDatapoints

I'm sorry. Which file? Just inside of the custom converter? And should there be any additional code inside the `tzDatapoints` block? I'm very new to this, and trying my best to do as much as possible on my own. 

This is what I've done so far based on your reply, but it's not working: 

```const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const tzDatapoints = {
    key: ['alarm', 'alarm_volume', 'alarm_time'],
}

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0224',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_cipibmnp'
        },
    ],
    model: 'TS0224',
    vendor: 'TuYa',
    description: 'Tuya Smart Light & Sound Siren ',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tzDatapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        exposes.binary('alarm', ea.STATE_SET, true, false).withDescription('Enable the alarm'),
        exposes.enum('alarm_volume', ea.ALL, ['low', 'middle', 'high', 'mute']).withDescription('Volume of siren'),
        exposes.numeric('alarm_time', ea.STATE_SET).withValueMin(60).withValueMax(3600).withValueStep(1)
                .withUnit('s').withDescription('Alarm time'),
        exposes.binary('muffling', ea.STATE_SET, true, false).withDescription("Muffling")],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'alarm', tuya.valueConverterBasic.lookup({false: tuya.enum(3), true: tuya.enum(2)})], // <- mapped the datapoint
            [5, 'alarm_volume', tuya.valueConverter.raw],
            [7, 'alarm_time', tuya.valueConverter.raw],
            [16, 'muffling', tuya.valueConverter.raw]
        ],
    }
};

module.exports = definition;
```

Koenkk, 2023-01-15:
I've updated your comment (https://github.com/Koenkk/zigbee2mqtt/issues/16060#issuecomment-1383145012)

LukasMendez, 2023-01-15:
Oh my godness.. Didn't realize that using three dots were actually a thing in JavaScript 😂 It gives no errors anymore, but I can't make it trigger any siren.

I've tried to modify the 'alarm' part according to the options you actually have in Tuya

Adding this as the expose instead: 

`exposes.enum('alarm', ea.ALL, ['alarm_sound', 'alarm_light', 'alarm_sound_light', 'normal']).withDescription('Volume of siren')`

As this is what it looks like in Tuya IoT Platform

[image]

But now I'm getting this error: 

[image]

LukasMendez, 2023-01-16:
Alright... I realized that I had been trying to use 'valueConverterBasic.raw' instead of 'valueConverter.raw'. 

However it doesn't do anything. I can click on all of the buttons, but the device itself does not do anything.
No errors are displayed though. 

I have updated my code to the following: 

```
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: ['alarm', 'alarm_volume', 'alarm_time'],
}

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0224',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_cipibmnp'
        },
    ],
    model: 'TS0224',
    vendor: 'TuYa',
    description: 'Tuya Smart Light & Sound Siren ',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tzDatapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        exposes.enum('alarm', ea.ALL, ['alarm_sound', 'alarm_light', 'alarm_sound_light', 'normal']).withDescription('The alarm state'),
        exposes.enum('alarm_volume', ea.ALL, ['low', 'middle', 'high', 'mute']).withDescription('Volume of siren'),
        exposes.numeric('alarm_time', ea.STATE_SET).withValueMin(60).withValueMax(3600).withValueStep(1)
                .withUnit('s').withDescription('Alarm time')],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'alarm', tuya.valueConverterBasic.lookup({'alarm_sound': tuya.enum(0), 'alarm_light': tuya.enum(1), 'alarm_sound_light': tuya.enum(2), 'normal': tuya.enum(3)})],
            [5, 'alarm_volume', tuya.valueConverterBasic.lookup({'low': tuya.enum(0), 'middle': tuya.enum(1), 'high': tuya.enum(2), 'mute': tuya.enum(2)})],
            [7, 'alarm_time', tuya.valueConverter.raw]
        ],
    }
};

module.exports = definition;
```

How do I check that the command is fired, and what can I do myself to diagnose this?

Koenkk, 2023-01-16:
That means the wrong payload is send, can you try with integers instead? `tuya.enum(0)` -> `0`

LukasMendez, 2023-01-16:
@Koenkk I've updated my code as suggested, but still nothing happens. Really weird. I feel like we are getting so close. It looks a lot like the other examples I've seen

LukasMendez, 2023-01-17:
This is what the standard instruction set looks like in Tuya IoT Cloud in JSON: 

[image]

Koenkk, 2023-01-17:
It is unclear what value is send with alarm_state normal. Is it possible to send this via the TuYa platform?

LukasMendez, 2023-01-17:
@Koenkk Yes if I send this it will start making sounds and flashing the light

[image]

LukasMendez, 2023-01-19:
@Koenkk Yes. This is what I see: 

[image]

Koenkk, 2023-01-19:
What if you set the alarm and check the device logs as explained on https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#_9-find-your-data-point, does it show the data that was sent?

LukasMendez, 2023-01-19:
Hmm doesn't really look like what we are after, but this is what it shows:

[image]

LukasMendez, 2023-01-19:
And this is the payload of the event filter 

[image]

Koenkk, 2023-01-21:
I logged into the TuYa platform myself, but I cannot find how to enums are mapped to the datapoints. Given you are sending 3 things at once with the platform (`alarm_time`, `alarm_state` and `alarm_volume`), can you try to do the same via z2m? (what happens if you set the `alarm_time`?) For the `alarm_state` try with both:

- `[1, 'alarm_state', tuya.valueConverterBasic.lookup({'alarm_sound': tuya.enum(0), 'alarm_light': tuya.enum(1), 'alarm_sound_light': tuya.enum(2), 'normal': tuya.enum(3)})],`
- - `[1, 'alarm_state', tuya.valueConverterBasic.lookup({'alarm_sound': tuya.enum(3), 'alarm_light': tuya.enum(2), 'alarm_sound_light': tuya.enum(1), 'normal': tuya.enum(0)})],`

LukasMendez, 2023-01-21:
@Koenkk Yeah it's a weird logic. The way the alarm states should be understood is: 

Normal = No sound, flashes or anything (So basically the alarm is off)
Alarm light = Triggers the alarm, but only thing that will happen is that the light will flash and in the time specified in `alarm_time`
Sound / Sound Light = Triggers the alarm and will sound at the volume specified in `alarm_volume` and the duration specified for `alarm_time`

I've tried to test it using the lines you just mentioned, but it still doesn't do anything 😞

Koenkk, 2023-01-22:
Any chance you could sniff the traffic with the TuYa gateway when turning on just the alarm? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

LukasMendez, 2023-01-22:
Is it correctly understood that I need an extra zigbee dongle? I was also considering sponsoring an alarm for you if you could make it work yourself

Koenkk, 2023-01-23:
Yes you need a separate dongle. I'm wondering if the device is still available, https://www.aliexpress.com/item/1005004705172997.html?spm=a2g0o.order_list.order_list_main.85.21ef1802lYK1Az shows it is not.

LukasMendez, 2023-01-23:
If you want to, I can send you one. That's the least I can do as you've already done a lot for the community 👍

LukasMendez, 2023-01-24:
@Koenkk Free Shipping apparently 

[image]

Koenkk, 2023-01-25:
I've ordered the device (22.49 euros) total

disclaimer: there is not guarantee that I will get it working.

LukasMendez, 2023-01-25:
Well okay then 😂 I was just trying to contribute in some kind of way. But that's nice of you. I look forward to updates on your progress then 👍 And thank you for your willingness to help

Koenkk, 2023-01-25:
In case you want to make a contribution, check the `Sponsors` button at the top of the page 😄

LukasMendez, 2023-01-25:
Done 😊👍

lisecons, 2023-02-09:
I'm looking forward to any progress with this device and many thanks in advance.
I used to code posted above to try it out. 
Can confirm it is not working.
I did get the no converter message. (see below)
If i can help out let me know. I am very new to this however.

Debug 2023-02-09 14:06:13Received Zigbee message from 'Living room Alarm', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":2,"batteryVoltage":2}' from endpoint 1 with groupID 0
Debug 2023-02-09 14:06:13No converter available for 'TS0224' with cluster 'genPowerCfg' and type 'attributeReport' and data '{"batteryPercentageRemaining":2,"batteryVoltage":2}'
Debug 2023-02-09 14:09:27Received Zigbee message from 'Living room Alarm', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":2,"batteryVoltage":2}' from endpoint 1 with groupID 0

Koenkk, 2023-02-20:
It seems this device does not use the TuYa api but a combination of standard zigbee commands with some manufacturer specific writes. Please try out with the following external converter: https://gist.github.com/Koenkk/04b9a1ff8c0ca2c046e3e00cfe469f10

For the warning I did not try all the various combinations, mode `burglar` is what the TuYa bridge uses when you select `voice + light alarm`.

LukasMendez, 2023-02-21:
@Koenkk Nice work! 

I've tested your external converter. Had to add the `fingerprint` key in order for Z2M to recognize it as supported. 

Unfortunately I can't really get anything to work other than turning the light on and off. I'm not sure if there's actually happening anything when tweaking all the other parameters, as the volume level doesn't seem to work at all. 

[image]

Maybe it is doing something, but is just muted.. But right now, as mentioned. I can only turn the light on and off.

LukasMendez, 2023-02-21:
@Koenkk Nevermind... Just realized that I had to give a value for all parameters and then click on "Apply". Almost shit my pants

LukasMendez, 2023-02-21:
Is there any way to perhaps pre-define a combination of settings and then just expose it as a toggle or something like that? Right now the "Warning" entity is not really adjustable from HA itself, unless I'm not seeing something. 

[image]

For example I would just like to have "Burglar" + "Medium" + "Medium" + True + 5 seconds + 3600 or something like that. And then just be able to turn it on and off, or something similar.

LukasMendez, 2023-02-21:
> you guys are awesome. After including the fingerprint part in the convertor i can confirm i'm now able to set off the alarm.
> 
> Thanks

Have you also found a way to control it in HA?

lisecons, 2023-02-21:
i just publish a mqtt message.
I took the msg content from the debug log after using the "exposes" tab in z2mqtt.
Start and stop samples below:
Debug 2023-02-21 14:33:31Received MQTT message on 'zigbee2mqtt/LivingRoomAlarm-01/set' with data '{"warning":{"duration":121,"level":"high","mode":"burglar","strobe":true,"strobe_duty_cycle":6,"strobe_level":"high"}}'

Debug 2023-02-21 14:34:39Received MQTT message on 'zigbee2mqtt/LivingRoomAlarm-01/set' with data '{"warning":{"duration":121,"level":"high","mode":"stop","strobe":true,"strobe_duty_cycle":6,"strobe_level":"high"}}'
Debug 2023-02-21 14:34:39Publishing 'set' 'warning' to 'LivingRoomAlarm-01'

LukasMendez, 2023-02-21:
> i just publish a mqtt message.
>
> I took the msg content from the debug log after using the "exposes" tab in z2mqtt.
>
> Start and stop samples below:
>
> Debug 2023-02-21 14:33:31Received MQTT message on [...]

I'm not so familiar with the exposes tab. Does it give you a button/toggle in Home Assistant that you can use? If yes, could you please show me exactly what you did. I tried myself use the implementation of "warning" using hardcoded values for "mode" and setting it to "burglar" or "stop" depending on the on/off state, but for some reason I'm being told that no converter exists. 

Ideally there should be a way to change all the different states (duration, strobe, volume etc) and then it should use them as variables when activating the alarm / sending a warning. But right now my goal is just to be able to turn the alarm on and off with some hardcoded values

lisecons, 2023-02-22:
[image]
if i would press the apply button in the above image. the device would flash and sound the alarm for 120 seconds.
Important to note that all field shown must have a value. And do not touch the field below the apply button since they are not implemented.
After you press apply you will get an debug line in the log.

2 examples:
Alarm + light  flashing
Debug 2023-02-21 14:33:31Received MQTT message on 'zigbee2mqtt/LivingRoomAlarm-01/set' with data '{"warning":{"duration":121,"level":"high","mode":"burglar","strobe":true,"strobe_duty_cycle":6,"strobe_level":"high"}}'

No Sound, Flash light for 60 seconds (plan to use this while arming and disarming the alarm)
Debug 2023-02-21 21:47:55Received MQTT message on 'zigbee2mqtt/Alarm-LivingRoom-001/set' with data '{"warning":{"duration":61,"level":"low","mode":"fire","strobe":true,"strobe_duty_cycle":6,"strobe_level":"very_high"}}'

Next you take this information and use the topic and data part to operate the device from home assistant.
[image]

[image]

Keep in mind with the topic of the device to include /set at the end

[image]

i hope this clarifies the way you can operate the device

LukasMendez, 2023-02-22:
> [image] if i would press the apply button in the above image. the device would flash and sound the alarm for 120 seconds. Important to note that all field shown must have a value. And do not touch [...]

Do you use any integration/addon to do publish the MQTT messages as you show in the bottom screenshots? Sorry, but I haven't really used it, so I'm not familiar with that interface. And do they show up as an entity on your device in HA? Because that's what I'm trying to achieve

LukasMendez, 2023-02-22:
Okay I managed to make it work by adding the following: 

```
const alarmConverter = {
alarm: {
    key: ['alarm'],
    convertSet: async (entity, key, value, meta) => {
        const mode = {'stop': 0, 'burglar': 1, 'fire': 2, 'emergency': 3, 'police_panic': 4, 'fire_panic': 5, 'emergency_panic': 6};
        const level = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};
        const strobeLevel = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};

        const values = {
            mode: value.toLowerCase() === 'on' ? 'burglar' : 'stop',
            level: 'medium',
            strobe: true,
            duration: 3600,
            strobeDutyCycle: 10,
            strobeLevel: 3,
        };

        let info;
        // https://github.com/Koenkk/zigbee2mqtt/issues/8310 some devices require the info to be reversed.
        if (['SIRZB-110', 'SRAC-23B-ZBSR', 'AV2010/29A', 'AV2010/24A'].includes(meta.mapped.model)) {
            info = (mode[values.mode]) + ((values.strobe ? 1 : 0) << 4) + (level[values.level] << 6);
        } else {
            info = (mode[values.mode] << 4) + ((values.strobe ? 1 : 0) << 2) + (level[values.level]);
        }

        await entity.command(
            'ssIasWd',
            'startWarning',
            {startwarninginfo: info, warningduration: values.duration,
                strobedutycycle: values.strobeDutyCycle, strobelevel: values.strobeLevel},
            utils.getOptions(meta.mapped, entity));
    }
}
};

  // Merge the custom converters with the existing ones
Object.assign(tz, alarmConverter);

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0224',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_cipibmnp'
        },
    ],
    zigbeeModel: 'TS0224',
    model: 'TS0224',
    vendor: 'TuYa',
    description: 'Tuya smart light & sound siren',
    fromZigbee: [],
    toZigbee: [tz.warning, tzLocal.TS0224, alarmConverter.alarm],
    exposes: [e.warning(), 
        exposes.binary('light', ea.ALL, 'ON', 'OFF').withDescription('Turn the light ON/OFF'),
        exposes.binary('alarm', ea.ALL, 'ON', 'OFF').withDescription('Turn the alarm ON/OFF'),
        exposes.numeric('duration', ea.STATE_SET).withValueMin(60).withValueMax(3600).withValueStep(1).withUnit('s')
            .withDescription('Duration of the alarm'),
        exposes.enum('volume', ea.STATE_SET, ['mute', 'low', 'medium', 'high'])
            .withDescription('Volume of the alarm'),
    ],
};
```

Only problem now is I need to find out how to save the state. So when I turn it on, it should know that it is on. Right now it doesn't. 

[image]

Koenkk, 2023-02-22:
```js
        await entity.command(
            'ssIasWd',
            'startWarning',
            {startwarninginfo: info, warningduration: values.duration,
                strobedutycycle: values.strobeDutyCycle, strobelevel: values.strobeLevel},
            utils.getOptions(meta.mapped, entity));
       return {state: {[key]: value}}; // <-- add this line
    }
```

lisecons, 2023-02-23:
> ```js
>         await entity.command(
>             'ssIasWd',
>             'startWarning',
>             {startwarninginfo: info, warningduration: values.duration,
>                 strobedutycycle: [...]

That works for me, thanks

Koenkk, 2023-02-25:
There are two issues with exposing the alarm switch:
- This is not really a switch, the alarm will turn off after 3600 seconds, the switch will stay ON
- Different users might want different values (level, strobe) etc.

Therefore it seems to make more sense to let users define there own MQTT switch in their Home Assistant `configuration.yaml`.

LukasMendez, 2023-02-25:
> There are two issues with exposing the alarm switch:
>
> * This is not really a switch, the alarm will turn off after 3600 seconds, the switch will stay ON
> * Different users might want different values [...]

Yes I also realized that. I think I agree with you on this one

Koenkk, 2023-02-26:
I've integrated this device, will be supported out-of-the-box in the next z2m release on 1 February.

closed 2023-02-26 by Koenkk

karel314, 2023-04-22:
So I've got this working, however triggering it is still a hassel. Home Assistan does see other devices as a Siren. But this one is not seen as one. And therefor cannot be trigger out of the box as a Siren in HA. Is this something that can be looked into? 

https://www.home-assistant.io/integrations/siren/

shayan, 2023-05-07:
> So I've got this working, however triggering it is still a hassel. Home Assistan does see other devices as a Siren. But this one is not seen as one. And therefor cannot be trigger out of the box as [...]

Same issue.
HA does not recognize this device as a SIREN.

alfbas, 2023-05-08:
I have this device and it is true that HA does not recognize it at a Siren....but you can use it any creating some MQTT scenes, that's what I have done and it works fine

shayan, 2023-05-08:
Yes, that's possible. I was able to produce some sound of this siren trough Z2M ui panel [like this](https://github.com/Koenkk/zigbee2mqtt/issues/16060#issuecomment-1439621740) (working only in burglar mode and low volume)

However, within the HA panel, apart from turning on the LIGHT of the siren, nothing else can be done trough Automations or Scripts or Dashboard & ....

alfbas, 2023-05-08:
nope, I trigger the alarm directly from HA. What I have done is:

Create two Mqqt scenes;

Done with this entries in the configuration.yaml file:

mqtt:
  scene:
    - name: Siren on
      unique_id: Siren_on
      command_topic: "zigbee2mqtt/**Alarma**/set"
      payload_on: '{"warning":{"duration":1,"level":"low","mode":"burglar","strobe":false,"strobe_duty_cycle":1,"strobe_level":"low"}}'    
    - name: Siren off
      unique_id: Siren_off
      command_topic: "zigbee2mqtt/**Alarma**/set"
      payload_on: '{"warning":{"duration":1,"level":"low","mode":"stop","strobe":false,"strobe_duty_cycle":1,"strobe_level":"low"}}'    

being **alarma** the name of the alarm device in zigbee2mqtt.

Then, you need to create a toggle helper:
[image]

and finally, create two these two automation linked to the helper status:

alias: Alarma OFF
description: ""
trigger:
  - platform: state
    entity_id:
      - input_boolean.alarma
    from: "on"
    to: "off"
condition: []
action:
  - service: scene.turn_on
    target:
      entity_id: scene.siren_off
    metadata: {}
mode: single

alias: Alarma ON
description: ""
trigger:
  - platform: state
    entity_id:
      - input_boolean.alarma
    from: "off"
    to: "on"
condition: []
action:
  - service: scene.turn_on
    target:
      entity_id: scene.siren_on
    metadata: {}
mode: single

And, that's all. You can add the helper to a dasboard to test it, works like a charm. And, of course, you use the helper in any automation, in node-red,.....everywhere

shayan, 2023-05-09:
Actually its not "Direct" from HA. by "Direct" i mean an exposed item (like a switch, a light, a siren or etc) entity inside HA. Like all other devices. now (in latest HA) I have a these entities for this device. It seems there is a missing "Siren ON/OFF switch" entity for this device inside HA. So there is no need to define MQTT Scene or other tricks to turn ON & OFF the sound.

[image]

DaeMonSxy, 2023-09-03:
It seems the device does not recognised by zigbee2mqtt, any hints to try?
[image]

rt400, 2023-11-14:
> Okay I managed to make it work by adding the following:
>
> ```
> const alarmConverter = {
> alarm: {
>     key: ['alarm'],
>     convertSet: async (entity, key, value, meta) => {
>         const mode = {'stop': [...]

Can you share the Full convert js ?
Includes the consts..

rt400, 2023-11-16:
@LukasMendez can you help?

LukasMendez, 2023-11-16:
> @LukasMendez can you help?

Sure. Haven't touched it for a long time, so don't know if it works and some of the code might also be unnecessary, but here it is: 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    TS0224: {
        key: ['light', 'alarm_duration'],
        convertSet: async (entity, key, value, meta) => {
            let payload = null;
            if (key === 'light') {
                await entity.command('genOnOff', value.toLowerCase() === 'on' ? 'on' : 'off', {}, utils.getOptions(meta.mapped, entity));
            } else if (key === 'duration') {
                await entity.write('ssIasWd', {'maxDuration': value}, utils.getOptions(meta.mapped, entity));
            } else if (key === 'volume') {
                const lookup = {'mute': 0, 'low': 10, 'medium': 30, 'high': 50};
                value = value.toLowerCase();
                utils.validateValue(value, Object.keys(lookup));
                await entity.write('ssIasWd', {0x0002: {value: lookup[value], type: 0x0a}}, utils.getOptions(meta.mapped, entity));
            } 
            return {state: {[key]: value}};
        },
    },
};

const alarmConverter = {
alarm: {
    key: ['alarm'],
    convertSet: async (entity, key, value, meta) => {
        const mode = {'stop': 0, 'burglar': 1, 'fire': 2, 'emergency': 3, 'police_panic': 4, 'fire_panic': 5, 'emergency_panic': 6};
        const level = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};
        const strobeLevel = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};

        const values = {
            mode: value.toLowerCase() === 'on' ? 'emergency' : 'stop',
            level: 'medium',
            strobe: true,
            duration: 3600,
            strobeDutyCycle: 10,
            strobeLevel: 3,
        };

        let info;
        // https://github.com/Koenkk/zigbee2mqtt/issues/8310 some devices require the info to be reversed.
        if (['SIRZB-110', 'SRAC-23B-ZBSR', 'AV2010/29A', 'AV2010/24A'].includes(meta.mapped.model)) {
            info = (mode[values.mode]) + ((values.strobe ? 1 : 0) << 4) + (level[values.level] << 6);
        } else {
            info = (mode[values.mode] << 4) + ((values.strobe ? 1 : 0) << 2) + (level[values.level]);
        }

        await entity.command(
            'ssIasWd',
            'startWarning',
            {startwarninginfo: info, warningduration: values.duration,
                strobedutycycle: values.strobeDutyCycle, strobelevel: values.strobeLevel},
            utils.getOptions(meta.mapped, entity));
    }
}
};

  // Merge the custom converters with the existing ones
Object.assign(tz, alarmConverter);

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0224',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_cipibmnp'
        },
    ],
    zigbeeModel: 'TS0224',
    model: 'TS0224',
    vendor: 'TuYa',
    description: 'Tuya smart light & sound siren',
    fromZigbee: [],
    toZigbee: [tz.warning, tzLocal.TS0224, alarmConverter.alarm],
    exposes: [e.warning(), 
        exposes.binary('light', ea.ALL, 'ON', 'OFF').withDescription('Turn the light ON/OFF'),
        exposes.binary('alarm', ea.ALL, 'ON', 'OFF').withDescription('Turn the alarm ON/OFF'),
        exposes.numeric('duration', ea.STATE_SET).withValueMin(60).withValueMax(3600).withValueStep(1).withUnit('s')
            .withDescription('Duration of the alarm'),
        exposes.enum('volume', ea.STATE_SET, ['mute', 'low', 'medium', 'high'])
            .withDescription('Volume of the alarm'),
    ],
};

module.exports = definition;
```

rt400, 2023-11-16:
@LukasMendez Thank you...
Did you able to mange it from HA ?

LukasMendez, 2023-11-16:
> @LukasMendez Thank you...
> 
> Did you able to mange it from HA ?

Yes. This is what I have.
[image]

rt400, 2023-11-16:
So there is no mode selector and the volume didn't work...

rt400, 2023-11-18:
After testing the device, it does not support any modes except "burglar mode" which only sounds a siren and with a light and "fire mode" which turns on the light without the siren. The volume cannot be changed at all. In my opinion, you can download all the other options because they are not relevant at all.

Stratagm, 2023-11-25:
I just acquired this siren today and appear to have no way to sound the siren.

jsalatiel, 2023-12-02:
Is there anyway to mute the volume on this thing and trigger only the lights?
Trying to set volume = mute in zigbee2mqq throws an error:
[image]

danielrosehill, 2024-04-03:
Anyone figured out how to get the scenes on this to work? I can create scenes in Z2M with the relevant parameters for the siren. But upon activation, they do nothing.

danielrosehill, 2024-04-03:
> After testing the device, it does not support any modes except "burglar mode" which only sounds a siren and with a light and "fire mode" which turns on the light without the siren. The volume cannot [...]

Ah - my friend from the wonderful Shabbat integration! What a small world!

Did you figure it out in the end? I'm also not getting it to work.

[2 more replies]