Koenkk/zigbee2mqtt #16070: [Feature request]: Moes ts0601_thermostat SET position

terrusso, 2023-01-09:
### Is your feature request related to a problem? Please describe

Tvr valve is internally managed as a PID but proportional, integral and derivative parameters can’t be managed. As result, this standard configuration can’t satisfy both right temperature and energy management

### Describe the solution you'd like

Actually valve position (%) is exposed but no way to set the value. It would be great to have an input number/dedicated mqtt topic to set valve position. In this way the PID behaviour could be adatped to every situation

### Describe alternatives you've considered

N.a.

### Additional context

N.a.

JacekKac, 2023-01-10:
as far as i know it's not possible to do it. My workarround is to set auto mode and heating setpoint and use the pid alogritm inside the valve. In addition i am measuring temp with additional temp sensor and publish the temp recalibration parameter to the valve to keep the temp reading synchronized

terrusso, 2023-01-10:
> as far as i know it's not possible to do it. My workarround is to set auto mode and heating setpoint and use the pid alogritm inside the valve. In addition i am measuring temp with additional temp sensor and publish the temp recalibration parameter to the valve to keep the temp reading synchronized

The same solution i tryied before, but batteries were off in 3 weeks. As i know position can be set since in zha i could make a service call to the cluster to set the position, so i imagined it would be possibile to add in zb2mqtt

JacekKac, 2023-01-10:
i was using it with ON / OFF force command for 2 months (battery life) then i switched to internal auto mode with is using 25/50/75/100 position and i noticed much better battery performance. 
so if it is possible i would love to have it - this type of radiator valve is very often used by users
@Koenkk maybe u can help ? it seems to be quite easy to adopt converters

terrusso, 2023-01-10:
i don't know if this can help, but when i used to set position from zha i called cluster = 61184 and attrid = 621 giving the desired potision set

malosaa, 2023-01-12:
what i don't understand, when i set the TS0601 to 15c and inside its 20 the valve remains open to 100% when the device is on auto mode. 
Even when using on other presets nothing changes..

regards

JacekKac, 2023-01-12:
@malosaa hi! Its a little bit of topic, maybe you can try on Discusion. 
its important to me to stick to the future request.

closed 2023-02-19 by github-actions[bot]