Koenkk/zigbee2mqtt #16084: All paired devices unavailable after "Error: ZdoRequest error"

hhenkel, 2023-01-10:
### What happened?

All devices have been working fine till an error occurred in the log - afterwards no devices are shown in the web frontend, even though the container and the tasmota based coordinator have been rebooted.
 
The log of the zigbee2mqtt container shows the following error:
```
Zigbee2MQTT:info  2023-01-10 17:50:04: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe13a85a', payload '{"current":0,"energy":35.95,"indicator_mode":"off/on","linkquality":172,"power":0,"power_outage_memory":"restore","state":"OFF","voltage":232}'
Zigbee2MQTT:info  2023-01-10 17:50:05: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe139733', payload '{"current":0,"energy":20.4,"indicator_mode":"off/on","linkquality":164,"power":0,"power_outage_memory":"restore","state":"OFF","voltage":230}'
Error: ZdoRequest error
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:571:19)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at EZSPAdapter.nodeDescriptorInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:373:28)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:363:32)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Using '/app/data' as data directory
Zigbee2MQTT:info  2023-01-10 17:53:04: Logging to console and directory: '/app/data/log/2023-01-10.17-53-04' filename: log.txt
Zigbee2MQTT:info  2023-01-10 17:53:04: Starting Zigbee2MQTT version 1.29.1 (commit #7d67ffc)
Zigbee2MQTT:info  2023-01-10 17:53:04: Starting zigbee-herdsman (0.14.83)
Zigbee2MQTT:info  2023-01-10 17:53:14: zigbee-herdsman started (reset)
Zigbee2MQTT:info  2023-01-10 17:53:14: Coordinator firmware version: '{"meta":{"maintrel":"9 ","majorrel":"6","minorrel":"7","product":8,"revision":"6.7.9.0 build 405"},"type":"EZSP v8"}'
Zigbee2MQTT:info  2023-01-10 17:53:14: Currently 0 devices are joined:
Zigbee2MQTT:warn  2023-01-10 17:53:14: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2023-01-10 17:53:14: Allowing new devices to join.
Zigbee2MQTT:warn  2023-01-10 17:53:14: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2023-01-10 17:53:14: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2023-01-10 17:53:14: Connecting to MQTT server at mqtt://mosquitto_pod
Zigbee2MQTT:info  2023-01-10 17:53:14: Connected to MQTT server
Zigbee2MQTT:info  2023-01-10 17:53:14: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-01-10 17:53:15: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2023-01-10 17:53:15: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-01-10 17:53:15: Zigbee2MQTT started!
Zigbee2MQTT:info  2023-01-10 17:53:15: MQTT publish: topic 'homeassistant/switch/0x842e14fffe1395a9/switch/config', payload 'null'
Zigbee2MQTT:info  2023-01-10 17:53:15: MQTT publish: topic 'homeassistant/switch/0xccccccfffee1e059/switch/config', payload 'null'
```

### What did you expect to happen?

All paired devices should reappear

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

6.7.9.0 build 405

### Adapter

EZSP v8 / eWeLink Zigbee Ethernet Gateway (https://zigbee.blakadder.com/eWeLink_ZB-GW03.html)

mccasian, 2023-01-15:
I am having exactly the same problem with a eWeLink Zigbee Bridge(https://ewelinkstore.com/product/sonoff-zigbee-bridge-smart-zigbee-wifi-bridge/?v=fa868488740a)

The version of Zigbee2MQTT is [1.28.4], Adapter EZSP v8, Coordinator: 6.7.9.0 build 405

I will upgrade to the latest available version, although this seems to be still an open issue here, so I expect that the fix is not included in the update as of today.

mccasian, 2023-01-15:
update... I have just noticed that I am using Coordinator 6.7.9.0 build 405 just like you on one of my ZbBridges, but I have a second bridge connected to the same HA instance and it's running fine without any issue since months, and that's using same versions everywhere except the coordinator.

I also noticed that the 6.7.9 version it's declared "under testing" while the 6.7.8 it's the "recommended stable version". Source:
https://github.com/arendst/Tasmota/tree/development/tools/fw_SonoffZigbeeBridge_ezsp

I have just downgraded my EZSP to 6.7.8.0 build 373. I hope that's the solution :D

mccasian, 2023-01-21:
as I mentioned about a week ago, I have downgraded the Coordinator to 6.7.8 and not a single issue since then; this really looks to be solution, case when, the issue should be raised to Tasmota project instead of zigbee2mqtt.

closed 2023-03-01 by github-actions[bot]

oywino, 2023-11-29:
> as I mentioned about a week ago, **I have downgraded the Coordinator to 6.7.8 and I did not had a single issue since then**; this really looks to be solution, case when, the issue should be raised to Tasmota project instead of zigbee2mqtt.

I too have the same problem and as suggested I downgraded to 6.7.8 but the problem remains.
It all began after I paired a device that required a retention value - which was set automatically during pairing - and which caused Z2M to crash with the following error message:

`
MQTT retention requires protocol version 5
`
So, I added  `version: 5` to the _configuration.yaml_ file and all hell broke loose. Z2M did start, but has since crashed randomly in 3-10 hours intervals.  So I don't think the problem is 6.7.9 alone.

mccasian, 2023-12-31:
Hi. I can confirm that the issue is present in 6.7.8 too. While after downgrade, for weeks it worked well for me, after a time I started to notice the same issues.