Koenkk/zigbee2mqtt #16087: ikea bulbs stops responding and "Became unavailable"

ctrld-pl, 2023-01-10:
### What happened?

I use quite a few IKEA bulbs (48 to be exact) in my environment:

- TRADFRI bulb GU10 WS 400lm: 43
- TRADFRI remote control: 2
- FLOALT panel WS 30x90: 1
- TRADFRI bulb E27 CWS opal 600lm: 1
- FLOALT panel WS 60x60: 1

unfortunatelly in random time those bulbs jst disapears from the network
The only solution is to restart them and everything goes back to normal.

Let's focus for example for the *Lobby11* bulb (it became unavailable at *2023-01-10T19:23*)
[image]

What can you do to prevent it?

What I've done so far:
- changed zigbee to channel 25 to avoid any interferences with wifi
- tried second Sonoff adapter - exactly the same issue

I've attached full log from the container - from 2022-01-10T19:30 (few minutes before that) - but I just don't see inside anything.

### How to reproduce it (minimal and precise)

Just wait some time and the bulbs with disapeare from the network.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Sonoff Dongle Plus Zigbee 3.0 Usb 

### Debug log

[full_log_2.txt.zip](https://github.com/Koenkk/zigbee2mqtt/files/10387126/full_log_2.txt.zip)

untergeek, 2023-01-10:
Nevermind. My issue seems to be in #15856

ctrld-pl, 2023-01-18:
Still have it:

```
2023-01-18 13:36:24Publish 'set' 'state' to 'KonradDesk' failed: 'Error: Command 0x680ae2fffe11119e/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4472 - 1 - 23 - 6 - 11 after 10000ms)'
error 2023-01-18 13:36:50Publish 'set' 'state' to 'KonradDesk' failed: 'Error: Command 0x680ae2fffe11119e/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4472 - 1 - 26 - 6 - 11 after 10000ms)'
error 2023-01-18 13:37:13Failed to read state of 'KonradDesk' after reconnect (Read 0x680ae2fffe11119e/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4472 - 1 - 27 - 6 - 1 after 10000ms))
error 2023-01-18 13:37:36Publish 'set' 'brightness' to 'KonradDesk' failed: 'Error: Command 0x680ae2fffe11119e/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4472 - 1 - 28 - 8 - 11 after 10000ms)'
error 2023-01-18 13:37:59Publish 'set' 'brightness' to 'KonradDesk' failed: 'Error: Command 0x680ae2fffe11119e/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4472 - 1 - 33 - 8 - 11 after 10000ms)'
```

closed 2023-02-26 by github-actions[bot]