Koenkk/zigbee2mqtt #16088: GIDEALED ZC05M Wrong color_temp pre sets

firewhank, 2023-01-10:
### What happened?

The controller is connected to a led strip with RGBWW and the neutral position(value 370) outputs the max warmest color of the led

the positions warm and warmest don’t do anything.

the cool position (250), I believe is more accurate to be the neutral position

is it possível to adjust the values my self?

### Zigbee2MQTT version

1.29.1-1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

Cyberbeni, 2023-01-12:
This should be already fixed on dev: https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

firewhank, 2023-01-12:
Great! I will wait for it to come the stable release. Thanks

closed 2023-02-19 by github-actions[bot]