Koenkk/zigbee2mqtt #16090: State Legrand 067776

icagn, 2023-01-11:
### What happened?

The indicated status is not correct. The state is reversed from reality. The affected device is 067776.
The previous update fixed the error on device 067776_001a.

### Zigbee2MQTT version

1.29.1-1

### Adapter firmware version

20230104.0 - latest

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/10390461/debug.txt)

Koenkk, 2023-01-12:
Can you provide the data/database.db entry of this device?

icagn, 2023-01-12:
[Uploading database.txt…]()

icagn, 2023-01-12:
"id":4

Koenkk, 2023-01-13:
I think something went wrong when uploading the file, can just copy-paste it?

icagn, 2023-01-14:
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0024c8c9ac","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x04cf8cdf3c7f2f79","nwkAddr":23202,"manufId":4718,"manufName":"XIAOMI","powerSource":"Battery","modelId":"lumi.sen_ill.mgl01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[0,1024,3,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.sen_ill.mgl01","manufacturerName":"XIAOMI","powerSource":3,"zclVersion":3,"appVersion":26,"stackVersion":2,"hwVersion":1,"dateCode":"20200305","swBuildId":"2019\u0000www."}},"genPowerCfg":{"attributes":{"batteryVoltage":30}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":28574}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8c9ac","endpointID":1},{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8c9ac","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1024,"attrId":0,"minRepIntval":15,"maxRepIntval":3600,"repChange":500}],"meta":{}}},"appVersion":26,"stackVersion":2,"hwVersion":1,"dateCode":"20200305","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{"configured":855622161},"lastSeen":1673704989927,"defaultSendRequestWhen":"immediate"}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d000807d430","nwkAddr":55346,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_switch","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3002},{"elmType":33,"elmVal":5032},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":634},{"elmType":32,"elmVal":90}],"modelId":"lumi.sensor_switch","appVersion":10}},"genOnOff":{"attributes":{"32768":2,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":10,"interviewCompleted":true,"meta":{},"lastSeen":1673704416131,"defaultSendRequestWhen":"immediate"}
{"id":4,"type":"Router","ieeeAddr":"0x00047400007af8f9","nwkAddr":27114,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Shutter switch with neutral\u0000\u0000\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,258,4,5,15],"outClusterList":[0,25],"clusters":{"genBasic":{"attributes":{"modelId":" Shutter switch with neutral\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":0,"stackVersion":64,"hwVersion":1,"dateCode":" \u000020180330\u0000\u0000\u0000\u0000\u0000","swBuildId":"00d\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u00000\u0012\u0002\u0000\t\u0007\u0000\u0018\u0002\u0003\b\u0000 \u00132\u0000\u0000\u0000\u0000X\u0002\n\u0000\u0000\u0000\u0000d\u0017\u0000\u0018\u0000"}},"genIdentify":{"attributes":{"identifyTime":60}},"genBinaryInput":{"attributes":{"presentValue":1}},"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":100}}},"binds":[{"cluster" [... cut ...] outClusterList":[],"clusters":{"genBasic":{"attributes":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3022},{"elmType":33,"elmVal":17320},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":23926},{"elmType":32,"elmVal":94}],"modelId":"lumi.sensor_switch","appVersion":10}},"genOnOff":{"attributes":{"32768":3,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":10,"interviewCompleted":true,"meta":{},"lastSeen":1673704835069,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-01-15:
I see you have multiple, can you provide the `ieeeAddress` of the ones where the state is currently not correct? (e.g. `0x00047400007af8f9`)

icagn, 2023-01-15:
0x00047400007af8f9

Koenkk, 2023-01-16:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    // swbuildid 001a requires coverInverted: https://github.com/Koenkk/zigbee2mqtt/issues/15101#issuecomment-1356787490
    fingerprint: [
        {modelID: ' Shutter switch with neutral\u0000\u0000\u0000', softwareBuildID: '001a'},
        {modelID: ' Shutter switch with neutral\u0000\u0000\u0000', softwareBuildID: "00d\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u00000\u0012\u0002\u0000\t\u0007\u0000\u0018\u0002\u0003\b\u0000 \u00132\u0000\u0000\u0000\u0000X\u0002\n\u0000\u0000\u0000\u0000d\u0017\u0000\u0018\u0000"},
    ],
    model: '067776_001a',
    vendor: 'Legrand',
    description: 'Netatmo wired shutter switch',
    fromZigbee: [fz.identify, fz.ignore_basic_report, fz.legrand_binary_input_moving, fz.cover_position_tilt],
    toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.legrand_identify, tz.legrand_settingEnableLedInDark],
    meta: {coverInverted: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBinaryInput', 'closuresWindowCovering', 'genIdentify']);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
    exposes: [e.cover_position()],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

icagn, 2023-01-16:
I want to perform the manipulation, but how should I proceed?

icagn, 2023-01-16:
I got an error message when checking the configuration of Home Assistant

Koenkk, 2023-01-16:
Thanks, pushed the fix. You can keep using the external converter until the next release.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-16 by icagn