Koenkk/zigbee2mqtt #16091: Temperature step and max value

ZTHawk, 2023-01-11:
### What happened?

1. I cannot set the Max Temperature (it is always NULL).
2. The temperature can only be set in steps of 1 degree. But the device supports steps of 0.5 degrees.

I have the following thermostat (https://www.aliexpress.com/item/1005004996552864.html):
- Zigbee Model
  - TS0601
- Zigbee Manufacturer
  - _TZE200_aoclfnxz
- Description
  - Moes BHT series Thermostat
- Support status
  - Supported
- Manufacturer
  - [Moes](https://www.zigbee2mqtt.io/supported-devices/#v=Moes)
- Model
  - [BHT-002-GCLZB](https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#moes-bht-002-gclzb)

Here is the state of the device.
```
{
    "child_lock": "LOCK",
    "current_heating_setpoint": 15,
    "heat": "OFF",
    "linkquality": 244,
    "local_temperature": 22,
    "min_temperature_limit": 5,
    "preset": "hold",
    "preset_mode": "hold",
    "running_state": "idle",
    "sensor": "IN",
    "system_mode": "off",
    "deadzone_temperature": null,
    "local_temperature_calibration": null,
    "max_temperature_limit": null
}
```

### What did you expect to happen?

1. I can change the max temperature.
2. I can change the current temperature in 0.5 steps.

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (ZBDongle-E)

ZTHawk, 2023-02-11:
This issue still exists using v1.30.1-1

ArnSu, 2023-02-12:
> This issue still exists using v1.30.1-1

I am having the same issues. I tried Z2M Edge, same problem, cannot set deadzone temp, shows Null.
Any solution?
Thanks

dropsmisha, 2023-02-22:
I am having the same issues.
"deadzone_temperature": null,
"max_temperature_limit": null,
Any solution?
Thanks

phenixar, 2023-02-23:
Same issue here with a MOES Serie 006 ZIgbee, it was recognise like a BHT-002
Impossible to change Min / Max value and deadzone.
Other point, when I set a decimal value like 18.5 for example, the value is set to 0 or a strange value like "the max value of integer".
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 18,
    "deadzone_temperature": null,
    "local_temperature": 18,
    "max_temperature_limit": null,
    "min_temperature_limit": 1,
It's become uregent to find a solution because actually I need to manage my device manually to avoid power on boiler every 10 minutes :(
The deadzone is really mandatory to avoid damage a boiler or other device behind.
Thanks.

M450NC, 2023-02-26:
Same issue here with BHT-002-GCLZB.

Cannot set deadzone, max temp or adjust the programme mode setpoints/time. Apply button is greyed out and cannot get it to save any adjustments made in Z2M...

insipiens, 2023-03-01:
the device physically [using the buttons] supports 0.5 steps but for some reason Mioes/Tuya imp[lemented the zigbee part of the firmware (not Z2m) with steps of 1 (integers). Annoying but not something that can change through Z2M

{
    "current_heating_setpoint": 11,
    "deadzone_temperature": 1,
    "heat": "OFF",
    "linkquality": 131,
    "local_temperature": 15.5,
    "local_temperature_calibration": -2,
    "max_temperature_limit": 30,
    "min_temperature_limit": 5,
    "preset": "hold",
    "preset_mode": "hold",
    "running_state": "idle",
    "sensor": "IN",
    "system_mode": "heat",
    "child_lock": null
}

adorobis, 2023-03-01:
Setpoint step is a known limitation for zigbee communication for this device. 
And here is the information from Moes support confirming above limitation:
https://github.com/Koenkk/zigbee2mqtt/issues/14965#issuecomment-1368179306

M0DDer, 2023-03-12:
> Same issue here with 006 series, shows as BHT-002-GCLZB in Z2M.
> 
> Cannot set deadzone, max temp or adjust the programme mode setpoints/time. Apply button is greyed out and cannot get it to save any adjustments made in Z2M...

I have the same! Two devices not work(

Zigbee2MQTT 1.30.2 & 20221226

ZTHawk, 2023-04-12:
Still applicable

ZTHawk, 2023-05-13:
still applicable

M450NC, 2023-05-13:
Yep, I've decided to bin mine and choose a different thermostat. No idea why this hasn't been resolved when support for other devices seem to be resolved in a matter of days/weeks.

ZTHawk, 2023-06-14:
still applicable

iliesh, 2023-07-02:
I have the same issue, after updating zigbee2mqtt to the latest version (1.32.0) - I'm getting the following error:
```
zigbee-herdsman-converters:Moes BHT-002: existing program state not set.
```

csszokolai, 2023-07-11:
> I have the same issue, after updating zigbee2mqtt to the latest version (1.32.0) - I'm getting the following error:
> 
> ```
> zigbee-herdsman-converters:Moes BHT-002: existing program state not set.
> ```

I have same issue with this comment. I had removed / repaired the device, it does not solved the issue. BTW the other null-s are settable if I manually set the value in.the field not by the slider.

zeljkopopovic, 2023-08-15:
Hi guys, I am new in the community.
Is there any update about MOES  BHT-006? I just planning to order it and now I am thinking about which thermometer should I buy for underfloor water heating.
MOES  BHT-006 or  BHT-002 ?
Because here they mention that 002 works perfectly (https://github.com/fashberg/WThermostatBeca)
Thank you

closed 2023-09-23 by github-actions[bot]

jayanezj, 2023-10-31:
still applicable v1.33.1

adorobis, 2023-10-31:
This is not going to be implemented. It is a limitation of the device, not the z2m (at least the temp. step - not sure about the max temp limit).

jayanezj, 2023-10-31:
What about the deadzone? This is important

adorobis, 2023-10-31:
I can set the deadzone temp as well. The allowed range is from 0 to 5:
[image]

adorobis, 2023-10-31:
BTW, everything is described in the device documentation:
https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#moes-bht-002-gclzb

jayanezj, 2023-10-31:
The problem is that the device referred to in the ticket is not the BHT-002-GCLZB, it is BHT-006GAZB.
[image]

Although it is recognized by Zigbee2mqtt as a BHT-002-GCLZB. If necessary I can open a new ticket adding debug logs so that the correct compatibility for the model is added.
[image]

Even if values are adjusted in the sliders nothing happens, it still appears as null in the status and the slider is reset.

[image]

```
{
    ...
    "running_state": "idle",
    "sensor": "IN",
    "system_mode": "off",
    "deadzone_temperature": null,
    "local_temperature_calibration": null,
    "max_temperature_limit": null
}
```

adorobis, 2023-10-31:
Where? The ticket is opened for a different device, check the first entry by the opener:
[image]

adorobis, 2023-10-31:
In that case you should follow the process to add new device as described:
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

lukeskymuh, 2024-01-14:
This is also a problem if you try to use it with "better thermostat". If you set "Local temperature calibration" or "Current heating setpoint" to a non integer  value it results into some crazy random number. It looks like a integer to float conversion problem. As "Better thermostat" manipulates the values with float numbers it results in a random behavior. Is it possible to fix this in Z2M?

My device: MOES BHT-006 Zigbee

effgen, 2024-01-22:
I have the same problem, was a solution found or not?

adorobis, 2024-01-22:
As you can see this issue is closed. This is not z2m but a device limitation.. It only allows for 1 degree steps when controlled via zigbee.

jayanezj, 2024-01-22:
I was doing some research to try to create a new component with full compatibility. I integrated the device with TuYa to check the exact endpoints and found that both the degree step problems, the temperature limits and dead zone work equally bad, so it's not even a limitation of zigbee2mqtt, it's of TuYa's own software. Just make sure Zigbee2mqtt recognizes the device as [BHT002/BHT006](https://www.zigbee2mqtt.io/devices/BHT-002_BHT-006.html#moes-bht-002%252Fbht-006), you're not going to get better results than that.

lukeskymuh, 2024-01-22:
> As you can see this issue is closed. This is not z2m but a device limitation.. It only allows for 1 degree steps when controlled via zigbee.

Correct , but Z2M cold solve the problem by rounding the numbers for these devices.