Koenkk/zigbee2mqtt #16093: Device left network

Gerwintje, 2023-01-11:
### What happened?

Paired devices in Z2M randomly disconnect/leaving network. 

For example an Ikea (battery) device is successfully paired and interviewed. When the device is idle (no input) it the device is randomly disconnected from Z2M. Z2M reports below message in the log. 

Warning 2023-01-11 10:03:37Device '0xb43a31fffe555ad5' left the network.

Also, when rebooting Z2M al devices are 'missing' until they report back. Thereby I mean; when a motion sensor detects motion, it's connected and interviewed again bij Z2M. The device is listed again, but the above proces repeats itself. 

Z2M is installed through Docker on a QNAP NAS. Zigbee stick is attached to docker and placed away from NAS through a 5m extension cable.

Tried several updates (dev and non dev), re-pairing, changing channel with no succes. I this stage I'm a bit lost. 

Also I can observe that the interviewed Ikea devices have no updates available. However, when I check the Ikea site, there IS an update available. Z2M doesn't seem to detect is for some reason. Perhaps this has something to do with my issue. 

### What did you expect to happen?

Devices to maintain in Z2M.

### How to reproduce it (minimal and precise)

Reset Zigbee device, pair with Z2M and then wait. 'Leaving the network' occurs random. 

### Zigbee2MQTT version

1.29.1-dev commit: bf33963

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff | Zigbee 3.0 | Dongle Plus-E

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10391004/log.txt)

kizio1000, 2023-01-12:
Same here. Devices that are rarely used like door sensors go offline after some time and only repairing helps. LQI for those devices in my case is around 150-200 but still they disappear from network. This is somehow related with 1.8-1.9 updates as earlier I did not experienced such problems.

RemyyB, 2023-01-15:
I have the same issue, for now I only experienced this multiple times with 2 Ikea Styrbar Remotes and 1 Philips Hue Dimmer Switch v2. I don't really know if this is a coordinator issue or something else. But listing it here for sure :)
I didn't have Zigbee2MQTT installed previously so don't know if it could be the latest update.

amasuke55, 2023-01-15:
Same issue here, my Aqara Xiaomi WXKG15LM disconnected randomly and didn#t work until I paired it again. Also using a Sonoff Zigbee Dongle-E

MattWestb, 2023-01-15:
@amasuke55 If the Aqara sensor is having  the coordinator as its parent and your host system is not online and the sensor is trying talking with the coordinator it cant do it and leaving the network for 110%.
Always paring sleeping end device with good routers and not with the coordinator.

joshuevic, 2023-01-15:
I have the same issue with some devices. A Hue Switch, Ikea Blind and a Sonoff temperature sensor. 
But in my case it happens when I upgrade to new version of Zigbee2MQTT (i am using the docker version) or when I unplug the dongle from the usb port. I am using the Sonoff Dongle-E. I think there's a problem with that dongle. With Conbee II work perfectly.

Gerwintje, 2023-01-20:
Tired of all workarounds and possible solutions, I ordered a new Sonoff ZigBee stick, but now based on the CC2652P chip (ZBDongle-P). 

After repairing, all the above issues are resolved.

Simperdy, 2023-01-22:
Same here for the new Home Assistant SkyConnect stick i just received and two ikea motion sensors.
When the device is idle (no input) the device is randomly disconnected from Z2M. Z2M reports below message in the log.

Warning 2023-01-22 14:07:45 Device 'Bewegungsmelder Küche' left the network
Warning 2023-01-22 14:08:15 Device 'Bewegungsmelder Flur' left the network

On motion the devices get inverviewed again but my automation (turn on lamp) in HA is not triggered.

My z2m is version 1.29.2 on synology nas with docker.

Zigbee2MQTT Version: 1.29.2 commit: bb3e8f6
Coordinator-Typ: EZSP v9
Coordinator-Version: 7.1.1.0 build 273

I just started with z2m and this is my 4/5 device I migrated from zha.

ottoketney, 2023-01-28:
I have similar behaviour with Sonoff ZBDongle-E in Home Assistant.

tsspmq, 2023-01-28:
I had that issue when using an ezsp stick, not the Sonoff one so can confirm another stick in that category.   This was the HUSBZ1 dual Zigbee/ZWave stick that did it using 6.8.7 I think.   

I changed to a Sonoff Zigbee 3.0 -P stick a few weeks back and it stopped having that problem and a few others: Map timeouts when refreshing, interviewing was more successful on devices, Z2M crashing when a device left and joined on its own, and devices did not go zombie hours before falling off network.

My devices that did this were all at the farther end of the house and without any evidence I just figured they were trying a new route or something along the lines of that.   If I had reporting for the devices set too long it happened more frequently.   I have a custom handler for all devices so I can tweak reporting intervals for this large and busy network.

MattWestb, 2023-02-09:
@Archer36 After restart is the motion sensor updating its status in 2 hours without being triggered and going on line or is being complete silent for days ?

Archer36, 2023-02-10:
@MattWestb If I recall correctly I left it up for several hours possibly a day and the sensors never re-joined despite being triggered by my family moving around the house.

To facilitate testing/debugging I can setup a dedicated zigbee2mqtt host and pair a sensor to it via the SkyConnect adapter if that would help.

MattWestb, 2023-02-10:
Then sniffing is the only way to getting to knowing what is happening.

Archer36, 2023-02-10:
What is the best way to sniff the network data. A long time ago I used an SDR and GNU Radio to sniff some zigbee traffic and check for free channels, but I'm not sure if that is the best way to handle it to debug a situation like this. Do you have any suggestions or is there a recommended way that this project prefers?

MattWestb, 2023-02-10:
SDR you need one very good radio for receiving  2.4GHz and you cant analyze the data.
One cc-2531 with sniffer firmware or one EZSP and wireschark. Look in the Z2M Wiki.

Archer36, 2023-02-10:
Yeah I was using a limesdr mini and was able to get traffic into wireshark, but as you mentioned it was a lot of work and better suited to a dedicated sniffer. I picked up a cc-2531 module and will get it flashed to sniff. Thanks!

kirovilya, 2023-02-11:
@ottoketney @Simperdy @amasuke55 @Gerwintje please check on the latest dev/edge version of z2m, I made some changes for ezsp adapter

kizio1000, 2023-02-13:
For all of you having issues with devices leaving network, disconnections of coordinator and other issues when using Sonoff dongle E, please consider updating your coordinator firmware. All issues are gone for me after updating from EZSP v8 -
6.10.3.0 build 297 to EZSP v9 - 7.1.1.0 build 273 (I strongly advice to not load higher version than 7.1.1.0 as I had really BIG issues with stability after updating coordinator to any higher build). 

Easy step by step manual I found in following video:
https://www.youtube.com/watch?v=suK7BzKFFt4&t=330s

ottoketney, 2023-02-13:
where can I find the  Sonoff Dongle-E  EZSP v9 - 7.1.1.0 build 273 firmware?

kizio1000, 2023-02-13:
I have posted above link to youtube movie explaining it all and yet you ask w/o watching it, please do so and all will be clear...

Archer36, 2023-02-13:
Looks like my SkyConnect is running 7.1.1.0 and thanks to @kizio1000 post it inspired me to do some more googling and found the NabuCasa repo with the firmware https://github.com/NabuCasa/silabs-firmware. I'll try updating to 7.2.1.0 and see if that makes any difference.

Archer36, 2023-02-21:
Just an update I updated my SkyConnect to 7.1.4.0 and then spun up a zigbee2mqtt container on a separate host (version 1.30.1 commit: eb878d3) and finally paired one of my Iris sensors to it. Seems to be working much better, I can stop the container for an hour or more and then restart it and the Iris sensor reconnects almost instantly when the container starts up. I'll try moving the SkyConnect over to my Raspberry Pi running HAOS and see what happens.

yarosman, 2023-03-09:
Have similar issues with aqara contact sensor on cc2652P coordinator not sonoff with revision 20221226

maxime1992, 2023-04-10:
Up

wastez, 2023-04-29:
Did anybody try to use the new 7.2.2 Firmware?

Dutchy-79, 2023-04-29:
Same problem with 5 230v zigbee gu10 spots

wastez, 2023-04-29:
> Same problem with 5 230v zigbee gu10 spots 

With which firmware? The 7.2.2?

Dutchy-79, 2023-04-29:
Don't know how to see the firmware version. On OTA they don't show up.

wastez, 2023-04-29:
You have to go to settings - about in z2m. 
But if you don't know you will have 6.10.3.

Dutchy-79, 2023-04-29:
> You have to go to settings - about in z2m. But if you don't know you will have 6.10.3.

[image]

wastez, 2023-05-07:
> > You have to go to settings - about in z2m. But if you don't know you will have 6.10.3.
> 
> 
> 
> [image]
> 
> 

That's interesting. 
In the main post the problem happens in combination with ezsp adapter, your adapter is a different one.

closed 2023-06-15 by github-actions[bot]