Koenkk/zigbee2mqtt #16096: Problem with set the temperature "current_heating_setpoint" on TZE200_5toc8efa

VoltickVL, 2023-01-11:
### What happened?

I use dev channel (Zigbee2Mqtt Edge) and add new external converter for TZE200_5toc8efa.

When I set the temperature in the current_heating_setpoint menu: 30 degrees Celsius, my thermostat sets 3.0 degrees Celsius.
If I set 25 degrees Celsius, 2.5 degrees Celsius is set on the thermostat.
And when I set the temperature to 300 degrees Celsius, 30 degrees are set on the thermostat...
Tell me how to fix this problem?

### What did you expect to happen?

I update external converter and now when i set temp to 25 *C at device set same temp, but in z2m i see value 250.
I am not a programmer, please tell me how to edit the code so that the temperature of 25 degrees Celsius and not 250 degrees Celsius is displayed in z2m...

I have already tried different options, but I can't make the device set the temperature according to the formula temp = value *10, and only display temp. In this case, the temperature is set correctly on the device, but the display is not. 

[image]
[image]
[image]

### How to reproduce it (minimal and precise)

### My external converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposesLocal = {
    hour: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('h').withValueMin(0).withValueMax(23),
    minute: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('m').withValueMin(0).withValueMax(59),
    program_temperature: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('°C')
        .withValueMin(5).withValueMax(35).withValueStep(0.5),
};

const tzLocal = {
   thermostat_current_heating_setpoint: {
        key: ['current_heating_setpoint'],
        convertSet: async (entity, key, value, meta) => {
            const temp = Math.round(value * 10);
            await tuya.sendDataPointValue(entity, tuya.dataPoints.moesHeatingSetpoint, temp);
        },
   },
};

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_5toc8efa',
        },
    ],
    model: 'BHT-002-GCLZB',
    vendor: 'Moes',
    description: 'Moes BHT series Thermostat',
    fromZigbee: [fz.moes_thermostat],
    toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_mode,
            tz.moes_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration,
            tz.moes_thermostat_deadzone_temperature, tz.moes_thermostat_max_temperature_limit, tz.moes_thermostat_min_temperature_limit,
            tz.moes_thermostat_program_schedule, tzLocal.thermostat_current_heating_setpoint],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(), e.min_temperature_limit(),
            exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE).withLocalTemperatureCalibration(-30, 30, 1, ea.STATE_SET)
                .withSystemMode(['off', 'heat'], ea.STATE_SET).withRunningState(['idle', 'heat', 'cool'], ea.STATE)
                .withPreset(['hold', 'program']),
            e.temperature_sensor_select(['IN', 'AL', 'OU']),
            exposes.composite('program', 'program').withDescription('Time of day and setpoint to use when in program mode')
                .withFeature(exposesLocal.hour('weekdays_p1_hour'))
                .withFeature(exposesLocal.minute('weekdays_p1_minute'))
                .withFeature(exposesLocal.program_temperature('weekdays_p1_temperature'))
                .withFeature(exposesLocal.hour('weekdays_p2_hour'))
                .withFeature(exposesLocal.minute('weekdays_p2_minute'))
                .withFeature(exposesLocal.program_temperature('weekdays_p2_temperature'))
                .withFeature(exposesLocal.hour('weekdays_p3_hour'))
                .withFeature(exposesLocal.minute('weekdays_p3_minute'))
                .withFeature(exposesLocal.program_temperature('weekdays_p3_temperature'))
                .withFeature(exposesLocal.hour('weekdays_p4_hour'))
                .withFeature(exposesLocal.minute('weekdays_p4_minute'))
                .withFeature(exposesLocal.program_temperature('weekdays_p4_temperature'))
                .withFeature(exposesLocal.hour('saturday_p1_hour'))
                .withFeature(exposesLocal.minute('saturday_p1_minute'))
                .withFeature(exposesLocal.program_temperature('saturday_p1_temperature'))
                .withFeature(exposesLocal.hour('saturday_p2_hour'))
                .withFeature(exposesLocal.minute('saturday_p2_minute'))
                .withFeature(exposesLocal.program_temperature('saturday_p2_temperature'))
                .withFeature(exposesLocal.hour('saturday_p3_hour'))
                .withFeature(exposesLocal.minute('saturday_p3_minute'))
                .withFeature(exposesLocal.program_temperature('saturday_p3_temperature'))
                .withFeature(exposesLocal.hour('saturday_p4_hour'))
                .withFeature(exposesLocal.minute('saturday_p4_minute'))
                .withFeature(exposesLocal.program_temperature('saturday_p4_temperature'))
                .withFeature(exposesLocal.hour('sunday_p1_hour'))
                .withFeature(exposesLocal.minute('sunday_p1_minute'))
                .withFeature(exposesLocal.program_temperature('sunday_p1_temperature'))
                .withFeature(exposesLocal.hour('sunday_p2_hour'))
                .withFeature(exposesLocal.minute('sunday_p2_minute'))
                .withFeature(exposesLocal.program_temperature('sunday_p2_temperature'))
                .withFeature(exposesLocal.hour('sunday_p3_hour'))
                .withFeature(exposesLocal.minute('sunday_p3_minute'))
                .withFeature(exposesLocal.program_temperature('sunday_p3_temperature'))
                .withFeature(exposesLocal.hour('sunday_p4_hour'))
                .withFeature(exposesLocal.minute('sunday_p4_minute'))
                .withFeature(exposesLocal.program_temperature('sunday_p4_temperature')),
        ],

};

module.exports = definition;
```

### Zigbee2MQTT version

Zigbee2MQTT 1.29.0-dev commit: 3351bc2

### Adapter firmware version

20211217

### Adapter

SONOFF ZIGBEE 3.0 USB DONGLE PLUS (zStack3x0)

### Debug log

```shell
Warning 2023-01-09 21:31:11 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:31:50 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:31:53 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:32:11 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:32:46 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:32:49 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:33:17 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,24]}}
Warning 2023-01-09 21:34:18 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,19]}}
Warning 2023-01-10 00:04:21 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,4]}}
Warning 2023-01-10 00:05:21 zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,9]}}
```

VoltickVL, 2023-01-14:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5323

Koenkk, 2023-01-14:
Thanks for the fix!

closed 2023-01-14 by Koenkk

VoltickVL, 2023-01-14:
Koennk, i mb need also edit this code section:
```shell
            case tuya.dataPoints.moesHeatingSetpoint:
                return {current_heating_setpoint: value};
```
Because current_heating_setpoint don't work again :(   current_heating_setpoint that is 10 time lower than expected (2.5 instead of 25.0°).

VoltickVL, 2023-01-15:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5334