Koenkk/zigbee2mqtt #16099: Sengled Devices won't update firmware via OTA anymore.

Lymkin, 2023-01-11:
### What happened?

Sengled devices don't seem to update anymore.

### What did you expect to happen?

Expected the OTA update to work as it did previously.

### How to reproduce it (minimal and precise)

Request OTA update of Sengled Bulb (E11-N1EA).  This use to work previously so I can only assume my update to 1.29.1 is what broke it.

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2 (EZSP v8)

### Debug log

Error 2023-01-11 08:12:05 Update of 'D1' failed (Device didn't respond to OTA request)

gdst4rp, 2023-01-14:
Similar issue, though for some reason I have 12 of these bulbs, but 2 of them updated, and the rest failed. I just updated to MQTT 1.29.2 and it seems to be working so far.

Lymkin, 2023-02-06:
This now seems to be working again with v1.30.1

closed 2023-02-06 by Lymkin