Koenkk/zigbee2mqtt #16100: Innr BY 266 lights pair fine, but do not respond to commands

shanetuohy, 2023-01-11:
### What happened?

INNR BY 266 lights respond with failed (no response received) when any command is issued to them, or Z2M queries their status. 

Bulbs pair no problem, and also announce themselves onto the network so some communication is happening, but commands cannot be sent. 

The bulbs worked for a short period after first pairing, but dropped off the network and when re connected exhibited this issue. 

Example of the device announcing on the network, and failing to respond. 

`
debug 2023-01-11 12:20:11 Device '0xc49886000007f789' announced itself
info 2023-01-11 12:20:11 MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xc49886000007f789","ieee_address":"0xc49886000007f789"},"type":"device_announce"}'
info 2023-01-11 12:20:11MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xc49886000007f789"},"type":"device_announced"}'
debug 2023-01-11 12:20:13 Retrieving state of '0xc49886000007f789' after reconnect
error 2023-01-11 12:21:14 Failed to read state of '0xc49886000007f789' after reconnect (Read 0xc49886000007f789/0 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received))
`

 

Some guesses as to what is happening, since these did work for a period of time; 

- Perhaps there is a reference within the software to the bulb that's conflicting? 
- Since the last message seen is the bulb requesting an OTA, that it apparently doesn't support, maybe that has something to do with the device not responding? 

Please let me know if there is any other debugging I can help with. 

Discussion on the issue here - https://github.com/Koenkk/zigbee2mqtt/discussions/12985

Per the discussion, these bulbs work on ZHA with the same coordinator. 

 

### What did you expect to happen?

I expected that the light would turn on or off, or change brightness when commands are issued. 

### How to reproduce it (minimal and precise)

Pair an Innr BY 266, then attempt to control the device via Z2M. Commands time out. 

### Zigbee2MQTT version

1.29.1-1

### Adapter firmware version

0x26580700

### Adapter

ConbeeII

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10393178/log.txt)

Device is ID - 0xc49886000007f789 .

Log contains a pairing attempt.

shanetuohy, 2023-01-17:
Happy to do some more investigation on this one, if someone could give me some pointers on where to look?

closed 2023-02-24 by github-actions[bot]

julianfs, 2023-09-17:
See #18961 for the fix.