Koenkk/zigbee2mqtt #16107: CC2652P - network commissioning timed out - #12633 follow up

ughh12, 2023-01-11:
### What happened?

It is recurrence of https://github.com/Koenkk/zigbee2mqtt/issues/12633

I have SLZB-06 device, debug log is same like igvk reported, Error: network commissioning timed out. PanID + Network key are autogenerated.

I am not able to find Z-Stack_Home_3.x.0-20210708 to test. Neither 20211217 nor 20220219 works.

I tried to flash firmware multiple times using two different tools, tried to connect device via Ethernet and then USB, initiate it via Z2M as well as directly in ZHA, initiate without antenna in metal box – still the same error.

Please was there some solution for #12633? Could you, please, navigate me to 20210708 to try it?

Thanks.

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20220219

### Adapter

Smlight SLZB-06 (CC2562P)

### Debug log

2023-01-11T22:29:39.829Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[***],"panID":45996,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[20]},"serialPort":{"baudRate":115200,"path":"tcp://192.168.0.15:6638","adapter":"zstack"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-01-11T22:29:39.830Z zigbee-herdsman:adapter:zStack:znp:log Opening TCP socket with 192.168.0.15:6638
2023-01-11T22:29:39.833Z zigbee-herdsman:adapter:zStack:znp:log Socket connected
2023-01-11T22:29:39.833Z zigbee-herdsman:adapter:zStack:znp:log Socket ready
2023-01-11T22:29:39.834Z zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload
2023-01-11T22:29:39.834Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2023-01-11T22:29:40.837Z zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352
2023-01-11T22:29:40.839Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-01-11T22:29:40.841Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-01-11T22:29:40.847Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2023-01-11T22:29:40.847Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2023-01-11T22:29:40.848Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2023-01-11T22:29:40.848Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2023-01-11T22:29:40.849Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-11T22:29:40.849Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2023-01-11T22:29:40.849Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2023-01-11T22:29:40.854Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,10,97,2,2,1,2,7,1,59,137,52,1,0,233]
2023-01-11T22:29:40.854Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,10,97,2,2,1,2,7,1,59,137,52,1,0,233]
2023-01-11T22:29:40.854Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 10 - 3 - 1 - 2 - [2,1,2,7,1,59,137,52,1,0] - 233
2023-01-11T22:29:40.855Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20220219}
2023-01-11T22:29:40.855Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-11T22:29:40.855Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 16
2023-01-11T22:29:40.855Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20220219})
2023-01-11T22:29:40.855Z zigbee-herdsman:adapter:zStack:startup beginning znp startup
2023-01-11T22:29:40.856Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - getExtAddr - {}
2023-01-11T22:29:40.856Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,4,37]
2023-01-11T22:29:40.861Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,8,97,4,71,85,209,38,0,75,18,0,209]
2023-01-11T22:29:40.861Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,8,97,4,71,85,209,38,0,75,18,0,209]
2023-01-11T22:29:40.861Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 8 - 3 - 1 - 4 - [71,85,209,38,0,75,18,0] - 209
2023-01-11T22:29:40.862Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - getExtAddr - {"extaddress":"0x00124b0026d15547"}
2023-01-11T22:29:40.862Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-11T22:29:40.862Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":130}
2023-01-11T22:29:40.863Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,130,0,178]
2023-01-11T22:29:40.871Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,19,24,0,104]
2023-01-11T22:29:40.871Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,19,24,0,104]
2023-01-11T22:29:40.871Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 19 - [24,0] - 104
2023-01-11T22:29:40.871Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvLength - {"length":24}
2023-01-11T22:29:40.871Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-11T22:29:40.872Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvReadExt - {"id":130,"offset":0}
2023-01-11T22:29:40.872Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,33,28,130,0,0,0,187]
2023-01-11T22:29:40.887Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,97,28,0,24,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,127]
2023-01-11T22:29:40.887Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,97,28,0,24,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,127]
[... 316 lines cut ...]
2023-01-11T22:29:46.856Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- APP_CNF - bdbComissioningNotifcation - {"status":1}
2023-01-11T22:29:46.857Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-11T22:29:47.127Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,69,192,8,140]
2023-01-11T22:29:47.127Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,69,192,8,140]
2023-01-11T22:29:47.127Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 2 - 5 - 192 - [8] - 140
2023-01-11T22:29:47.128Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - stateChangeInd - {"state":8}
2023-01-11T22:29:47.128Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-11T22:29:49.815Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0]
2023-01-11T22:29:49.816Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0]
2023-01-11T22:29:49.818Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,65,128,0,2,1,2,7,1,192]
2023-01-11T22:29:49.819Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,254,6,65,128,0,2,1,2,7,1,192]
2023-01-11T22:29:49.819Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 6 - 2 - 1 - 128 - [0,2,1,2,7,1] - 192
2023-01-11T22:29:49.819Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- SYS - resetInd - {"reason":0,"transportrev":2,"productid":1,"majorrel":2,"minorrel":7,"hwrev":1}
2023-01-11T22:29:49.819Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:error 2023-01-11 23:30:46: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-01-11 23:30:46: Failed to start zigbee
Zigbee2MQTT:error 2023-01-11 23:30:46: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-01-11 23:30:46: Exiting...
Zigbee2MQTT:error 2023-01-11 23:30:46: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

ughh12, 2023-01-21:
I was investigating further on this issue:
-	I tried development branch firmware (20221102) – no success.
-	I installed Z2M on laptop and tried to start with device connected via USB on outdoor place (no Wifi/RF interference) – no success.
-	I borrowed Sonoff USB dongle – both P and E variants. Both started and operated without any issue with same setup like I tried for my device (I updated adapter connection details only). P variant that has same Zigbee chip like my SMLIGHT reports back 20210708 firmware version – I don’t want to flash recent version to not to potentially break Sonoff device. 

So, my conclusion is error message from Z2M or ZHA is misleading for sure. I cannot decide whether there is some hardware problem on my SMLIGHT device or issue is with firmware versions that support amplifier configuration in CC2652P (maybe in combination with some hw setup). 

I would appreciate if someone can provide CC1352P2_CC2652P_other_coordinator_20210708.zip file to continue in investigation.

Koenkk, 2023-01-22:
You can download `20210708` from [here](https://github.com/Koenkk/Z-Stack-firmware/tree/afd5ac4666732a7822611cbacef9ed571af99e01/coordinator/Z-Stack_3.x.0/bin) however I doubt if it fixes the problem. Usually this problem occurs when there is another network on the same panid/channel combination and therefore is most of the times fixed by changing the channel in your zigbee2mqtt `configruation.yaml`. So try and see if changing the channel fixes the problem, if not I guess there is either heavy interference of something close to the SLZB-06 or it is broken.

ughh12, 2023-01-22:
Thanks a lot. After fimware downgrade to 20210708 it works like a charm under same conditions like when it was failing before. So, I was able to exactly reproduce problem reported by igvk in https://github.com/Koenkk/zigbee2mqtt/issues/12633, incl. workaround he applied. It is very unlikely external interference as I was playing with various channels, PanIDs, cables, power supplies, network/usb mode, wifi-free area, etc. Obviously it does not mean error cannot be triggered by some hw defect in combination with firmware version (my strong suspicion is it has something to do with transmit power setup introduced by  later fw version). 
I am available to follow some instructions to troubleshoot problem further if required.