Koenkk/zigbee2mqtt #16109: LQI for devices connected to ZBDongle-P show dramatically less than when connected to ZBDongle-E

etlweather, 2023-01-12:
### What happened?

LQI for devices connected to ZBDongle-P show dramatically less than when connected to ZBDongle-E.

I have a few devices in a relatively small area, each at different distances - most are within a few feet. This is really a dev environment. But I am seeing similar discrepancies in a live environment.

With the ZBDongle-P, LQI for the devices is between 80-195 - all less than 200.

With the ZBDongle-E, LQI is 240-255.

### What did you expect to happen?

I expect LQI to be similar regardless of the coordinator in this scenario. Also, the devices are within a few feet so less than 200 LQI does not make any sense.

### How to reproduce it (minimal and precise)

ZBDongle-P with a few devices nearby.

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

20220219

### Adapter

ZBDongle-P

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10398507/log.txt)

Koenkk, 2023-01-12:
You cannot compare measurements between different stacks as measuring methods can vary.

> With the ZBDongle-P, LQI for the devices is between 80-195 - all less than 200.

80-195 is perfectly fine, (in my experience everything above 30)

etlweather, 2023-01-12:
Thanks. So if I get this right, it isn't really a scale from 0-255? Because if the device is within a few feet, it reads below 200.

I am trying to give non-tech people a sort of range that is meaningful.

Koenkk, 2023-01-13:
It is, but how this is computed depends on the stack. 

Take away message is to not care about this value when it is larger then 30-50 and not having any problems.

etlweather, 2023-01-13:
Ok, thank you!

closed 2023-01-13 by etlweather