Koenkk/zigbee2mqtt #16112: TS011F_plug_1 behaves like a TS011F_plug_3

ilmisu, 2023-01-12:
### What happened?

Hi all,
I've got this model of smart plug, that should advertise Voltage, current, power and energy beyond the on-off switch.
the problem is that it's not advertising any value, exposing properly only switch and setup variables, as you suggested in the issue related to TS011F_plug_3.

### What did you expect to happen?

I'd like to see power consumption values.

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

20200327

### Adapter

 zStack30x

Koenkk, 2023-01-12:
You will find it next to the zigbee2mqtt `configuration.yaml`.

ilmisu, 2023-01-12:
Alright in a few minutes I'll send it out.

ilmisu, 2023-01-12:
"id":22,"type":"Router","ieeeAddr":"0x70b3d52b6001bc7d","nwkAddr":12772,"manufId":4660,"manufName":"_TZ3000_okaz9tjs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,6,3,4,5,57345,2820,1794],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_okaz9tjs","powerSource":1,"zclVersion":2,"appVersion":100,"stackVersion":1,"hwVersion":3,"dateCode":"20210625","swBuildId":"20B+TZSKT11BS108"}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"moesStartUpOnOff":2,"tuyaBacklightMode":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}},"seMetering":{"attributes":{"currentSummDelivered":[0,128]}}},"binds":[{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0006104dc3","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":3,"dateCode":"20210625","swBuildId":"20B+TZSKT11BS108","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1673546608801,"defaultSendRequestWhen":"immediate"}

here we go.

thanks in advance for your kind support.

Koenkk, 2023-01-13:
Thanks, the issue is already fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/5287 which will be included in the 1 February release and is already available in the dev branch.  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ilmisu, 2023-01-13:
> Koenkk/zigbee-herdsman-converters#5287

I'm sorry but my instance is already switched on dev branch, but it is
still non working.
Could you please tell me how to check if I properly set up the dev branch?
And could you link me a guide in how to update the instance once the new
version will be released?
Thank you.

Koenkk, 2023-01-14:
@ilmisu how are you running z2m?

ilmisu, 2023-01-14:
I tun it into a raspberry pi 3b running stock OS.

Koenkk, 2023-01-14:
In that case, once switched to the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html), you can follow [this guide](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#for-later-update-zigbee2mqtt-to-the-latest-version)

ilmisu, 2023-01-14:
Thanks for the hint, but is the one I already followed.
I got an error at the "npm ci" instruction... But the rest has been
completed.

Koenkk, 2023-01-14:
`npm ci` required, what is the error?

ilmisu, 2023-01-14:
Command not found. Which package must I install then?

Koenkk, 2023-01-14:
NodeJs must be installed, see https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing

ilmisu, 2023-01-14:
Now it works like a charm.
Thanks a lot koen. And keep going, you're awesome! 👌😉

Koenkk, 2023-01-15:
Great!

closed 2023-01-15 by Koenkk

papperone, 2023-02-28:
I've exact same problem with 2 new ZB plug which are recognized as "[TS011F_plug_1]" but no report of W and A (only V) and I'm not able to activate polling mode :(

here is it how they are setup in **database.db**, can you help?

{"id":3,"type":"Router","ieeeAddr":"0xa4c138bc5e412a00","nwkAddr":15129,"manufId":4417,"manufName":"_TZ3000_gvn91tmx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0012/�+\u0017","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_gvn91tmx","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":218,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,72]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2643544,"53251":125,"53252":11554,"53253":1361,"powerOnBehavior":1,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe3bd3cb","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe3bd3cb","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe3bd3cb","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1677612962092,"defaultSendRequestWhen":"immediate"}

papperone, 2023-03-04:
> Could you check if the issue is fixed with the following external converter:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     fingerprint: [192, 160, 100, 69, 68, 65, 64, 66].map((applicationVersion) => {
>         return {modelID: 'TS011F', applicationVersion, priority: -1};
>     }),
>     model: 'TS011F_plug_3',
>     description: 'Smart plug (with power monitoring by polling)',
>     vendor: 'TuYa',
>     whiteLabel: [{vendor: 'VIKEFON', model: 'TS011F'}, {vendor: 'BlitzWolf', model: 'BW-SHP15'},
>         {vendor: 'Avatto', model: 'MIUCOT10Z'}, {vendor: 'Neo', model: 'NAS-WR01B'}, {vendor: 'Neo', model: 'PLUG-001SPB2'}],
>     ota: ota.zigbeeOTA,
>     extend: tuya.extend.switch({electricalMeasurements: true, powerOutageMemory: true, indicatorMode: true, childLock: true}),
>     configure: async (device, coordinatorEndpoint, logger) => {
>         await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
>         const endpoint = device.getEndpoint(1);
>         endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
>         endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
>         device.save();
>     },
>     options: [exposes.options.measurement_poll_interval()],
>     onEvent: (type, data, device, options) =>
>         tuya.onEventMeasurementPoll(type, data, device, options,
>             device.applicationVersion !== 66, // polling for voltage, current and power
>             [66, 100, 160].includes(device.applicationVersion), // polling for energy
>         ),
> };
> 
> module.exports = definition;
> ```
> 
>     * save this as file next to `configuration.yaml` as `ext_converter.js`
> 
>     * add it to `configuration.yaml`:
> 
> 
> ```yaml
> external_converters:
>   - ext_converter.js
> ```
> 
>     * start z2m, check if issue is fixed

hi Koen,
Sorry for the delay in answering; at the moment I'm still waiting as all of a sudden one of the 2 plugs started to report values for W and A as well, without me taking any actions. I'll try to diagnose what can have happened and eventually try the external converter you provided; I will keep you posted, THANKS!

Flashlab2, 2023-05-03:
I have a similar problem with _TZ3000_gvn91tmx. It seems to get stuck on values and doesn't update. Seems like TS011F_plug_3 behaviour to me. Can I force polling?

boesing, 2023-09-08:
Hm, today I received my SHP15 which should have power measuring with report function.
I've seen the documentation entry:

> Starting with firmware version 1.0.5 (which comes pre-flashed on plugs produced since Q4 2021) core functionality on this plug is broken. TuYa has disabled the automatic reporting of power, voltage and current values meaning they need to be polled instead. The poll interval can be controlled through the measurement_poll_interval option.
>
> If your plug is affected, it will be detected as [TS011F_plug_3](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html) instead of TS011F_plug_1.

So obv. my device is affected - but does that mean that due to this change to the firmware that this device will not report any values and therefore polling is required? Or is there any chance to have this enabled at some point?

dickman666, 2024-01-17:
Good evening, I have the same problem with 4 devices, it does not show consumption data. You support me? send database.db

`{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0024c3cb96","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x000d6ffffe6bad8d","nwkAddr":57024,"manufId":0,"manufName":"TUYATEC-HaoiuWzy","powerSource":"Battery","modelId":"RH3052","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,1026,1029],"outClusterList":[3,1026,1029],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2173}},"msRelativeHumidity":{"attributes":{"measuredValue":6073}},"genBasic":{"attributes":{"modelId":"RH3052","manufacturerName":"TUYATEC-HaoiuWzy","appVersion":71,"powerSource":3,"zclVersion":2,"stackVersion":0,"hwVersion":1,"dateCode":"20180608"}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":71,"stackVersion":0,"hwVersion":1,"dateCode":"20180608","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1705471313773,"defaultSendRequestWhen":"immediate"}
{"id":3,"type":"EndDevice","ieeeAddr":"0x000d6ffffe6ed138","nwkAddr":18861,"manufId":0,"manufName":"TUYATEC-HaoiuWzy","powerSource":"Battery","modelId":"RH3052","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,1026,1029],"outClusterList":[3,1026,1029],"clusters":{"genBasic":{"attributes":{"modelId":"RH3052","manufacturerName":"TUYATEC-HaoiuWzy","powerSource":3,"appVersion":71,"zclVersion":2,"stackVersion":0,"hwVersion":1,"dateCode":"20180608"}},"genPowerCfg":{"attributes":{"batteryVoltage":29,"batteryPercentageRemaining":170}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2032}},"msRelativeHumidity":{"attributes":{"measuredValue":3943}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":71,"stackVersion":0,"hwVersion":1,"dateCode":"20180608","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1705471768281,"defaultSendRequestWhen":"immediate"}
{"id":4,"type":"Router","ieeeAddr":"0xa4c13801407a04e4","nwkAddr":40242,"manufId":4417,"manufName":"_TZ3000_o81odtrp","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":2,"inClusterList":[0,3,4,5,6,10,57344,4096],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"32768":0,"tuyaBacklightMode":1,"onOff":0,"moesStartUpOnOff":2,"onTime":0,"offWaitTime":0}},"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_o81odtrp","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"haElectricalMeasurement":{"attributes":{}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2123427102},"lastSeen":1705472238843,"defaultSendRequestWhen":"immediate"}
[... cut ...] eeAddress":"0x00124b0024c3cb96","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0,"manufacturerCode":null},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0,"manufacturerCode":null}],"meta":{}}},"appVersion":5,"hwVersion":1,"dateCode":"20211103","zclVersion":1,"interviewCompleted":true,"meta":{"configured":-1949790169},"lastSeen":1705471722834,"defaultSendRequestWhen":"immediate"}`

[image]

Thanks

ansantos, 2025-11-03:
Hi @Koenkk, is there any workaround for the issue above?

ansantos, 2025-11-05:
[log.log](https://github.com/user-attachments/files/23362421/log.log)
Hi, check the log attached. There are thousands of log messages in just a few seconds in debug mode (really don't know why), so I'm not sharing all the logs from the 5 minutes you asked, but just for the few seconds after I reconfigured the device.
Even then, I suggest you search for "Tomada do Vicente" in order to find the messages related to that device. You'll see the reconfigure command around line 242 and then some errors a few seconds after. Maybe this can give you a clue.

ansantos, 2025-11-06:
UPDATE: I plugged something requesting more load and now I get measurements! I was using a USB charger that powers a small led, but maybe the load is not big enough to trigger sending measurement data. I now plugged another equipment and it seems to be working ok!
Could it be it was already working without the external converter but maybe it needs some extra configuration for it to be a little more sensitive?

ansantos, 2025-11-06:
UPDATE 2: I tested without the external converter and in fact it was already working, but not reporting voltage. With your external converter I can get the voltage. Still not sure about the total energy. Still zero in any of the scenarios.

One other thing, I can see that it seems there is an OTA update available, but the update always fails:

[image]

Koenkk, 2025-11-07:
Updated https://gist.github.com/Koenkk/1234e786d16ac1e0bde09bc83fc72d37, can you try to re-configure, see if it passes, wait 24 hours with a load attached and see if it reports energy?