Koenkk/zigbee2mqtt #16117: Bseed thermostat TS0601 TZE200_5toc8efa electric floor heating. Help to make a converter.

VladimirTuzovGitHub, 2023-01-12:
### Link

https://aliexpress.com/item/1005004644629279.html

### Database entry

{"id":36,"type":"Router","ieeeAddr":"0x003c84fffed33230","nwkAddr":52420,"manufId":4098,"manufName":"_TZE200_5toc8efa","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":31,"65508":1,"stackVersion":0,"dateCode":"","appVersion":72}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1672557341252,"defaultSendRequestWhen":"immediate"}

### Comments

Hello! Please help me to make a converter for thermostat BSEED _TZE200_5toc8efa, works with TuYa. Its software is very similar to BHT-002-GCLZB.. It is not currently supported in z2m.

VladimirTuzovGitHub, 2023-01-13:
I tried to build a converter based on the Moes BHT-002-GCLZB converter. I am doing this for the first time and I think that I make mistakes. Below I will describe what problems and errors come out.

Converter code:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const exposesLocal = {
    hour: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('h').withValueMin(0).withValueMax(23),
    minute: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('m').withValueMin(0).withValueMax(59),
    program_temperature: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('°C')
        .withValueMin(5).withValueMax(35).withValueStep(0.5),
};

const definition =     {
	fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aoclfnxz'},
		{modelID: 'TS0601', manufacturerName: '_TZE200_ztvwu4nk'},
		{modelID: 'TS0601', manufacturerName: '_TZE200_5toc8efa'},
		{modelID: 'TS0601', manufacturerName: '_TZE200_ye5jkfsb'},
		{modelID: 'TS0601', manufacturerName: '_TZE200_5toc8efa'},
		{modelID: 'TS0601', manufacturerName: '_TZE200_u9bfwha0'}],
	model: 'BHT-002-GCLZB',
	vendor: 'Moes',
	description: 'Moes BHT series Thermostat',
	fromZigbee: [fz.moes_thermostat],
	toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode,
		tz.moes_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration,
		tz.moes_thermostat_deadzone_temperature, tz.moes_thermostat_max_temperature_limit, tz.moes_thermostat_min_temperature_limit,
		tz.moes_thermostat_program_schedule],
	exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(), e.min_temperature_limit(),
		exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
			.withLocalTemperature(ea.STATE).withLocalTemperatureCalibration(-30, 30, 0.1, ea.STATE_SET)
			.withSystemMode(['off', 'heat'], ea.STATE_SET).withRunningState(['idle', 'heat', 'cool'], ea.STATE)
			.withPreset(['hold', 'program']),
		e.temperature_sensor_select(['IN', 'AL', 'OU']),
		exposes.composite('program', 'program').withDescription('Time of day and setpoint to use when in program mode')
			.withFeature(exposesLocal.hour('weekdays_p1_hour'))
			.withFeature(exposesLocal.minute('weekdays_p1_minute'))
			.withFeature(exposesLocal.program_temperature('weekdays_p1_temperature'))
			.withFeature(exposesLocal.hour('weekdays_p2_hour'))
			.withFeature(exposesLocal.minute('weekdays_p2_minute'))
			.withFeature(exposesLocal.program_temperature('weekdays_p2_temperature'))
			.withFeature(exposesLocal.hour('weekdays_p3_hour'))
			.withFeature(exposesLocal.minute('weekdays_p3_minute'))
			.withFeature(exposesLocal.program_temperature('weekdays_p3_temperature'))
			.withFeature(exposesLocal.hour('weekdays_p4_hour'))
			.withFeature(exposesLocal.minute('weekdays_p4_minute'))
			.withFeature(exposesLocal.program_temperature('weekdays_p4_temperature'))
			.withFeature(exposesLocal.hour('saturday_p1_hour'))
			.withFeature(exposesLocal.minute('saturday_p1_minute'))
			.withFeature(exposesLocal.program_temperature('saturday_p1_temperature'))
			.withFeature(exposesLocal.hour('saturday_p2_hour'))
			.withFeature(exposesLocal.minute('saturday_p2_minute'))
			.withFeature(exposesLocal.program_temperature('saturday_p2_temperature'))
			.withFeature(exposesLocal.hour('saturday_p3_hour'))
			.withFeature(exposesLocal.minute('saturday_p3_minute'))
			.withFeature(exposesLocal.program_temperature('saturday_p3_temperature'))
			.withFeature(exposesLocal.hour('saturday_p4_hour'))
			.withFeature(exposesLocal.minute('saturday_p4_minute'))
			.withFeature(exposesLocal.program_temperature('saturday_p4_temperature'))
			.withFeature(exposesLocal.hour('sunday_p1_hour'))
			.withFeature(exposesLocal.minute('sunday_p1_minute'))
[... 58 lines cut ...]
        "weekdays_p1_hour": 7,
        "weekdays_p1_minute": 6,
        "weekdays_p1_temperature": 0,
        "weekdays_p2_hour": 0,
        "weekdays_p2_minute": 59,
        "weekdays_p2_temperature": 25,
        "weekdays_p3_hour": 23,
        "weekdays_p3_minute": 0,
        "weekdays_p3_temperature": 25,
        "weekdays_p4_hour": 13,
        "weekdays_p4_minute": 30,
        "weekdays_p4_temperature": 25
    }
}
Device exposes:
[image]
[image]
[image]
Device:
[image]
[image]

I use Russian there, I think everything will be clear anyway.
It can be seen that the temperature setting and setting and max_temperature_limit and current_heating_setpoint are indicated in hundredths and should be in tenths. Also, there are problems with setting the days of the week for program mode. For the rest, I still see that it works correctly, but datatype errors periodically appear.
Please help me understand and make the correct converter. I would even agree to remove the program mode since I do not use the temperature setting by time and day of the week at all.

VladimirTuzovGitHub, 2023-01-13:
I also found that Hold and Programm are confused. I put the manual mode shows Programm, I put the program of the week shows Hold.

VoltickVL, 2023-01-14:
```shell
    moes_thermostat: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'moes_thermostat');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
            let temperature;
            /* See tuyaThermostat above for message structure comment */
            switch (dp) {
            case tuya.dataPoints.moesSchedule:
                return {
                    program: {
                        weekdays_p1_hour: value[0],
                        weekdays_p1_minute: value[1],
                        weekdays_p1_temperature: value[2] / 2,
                        weekdays_p2_hour: value[3],
                        weekdays_p2_minute: value[4],
                        weekdays_p2_temperature: value[5] / 2,
                        weekdays_p3_hour: value[6],
                        weekdays_p3_minute: value[7],
                        weekdays_p3_temperature: value[8] / 2,
                        weekdays_p4_hour: value[9],
                        weekdays_p4_minute: value[10],
                        weekdays_p4_temperature: value[11] / 2,
                        saturday_p1_hour: value[12],
                        saturday_p1_minute: value[13],
                        saturday_p1_temperature: value[14] / 2,
                        saturday_p2_hour: value[15],
                        saturday_p2_minute: value[16],
                        saturday_p2_temperature: value[17] / 2,
                        saturday_p3_hour: value[18],
                        saturday_p3_minute: value[19],
                        saturday_p3_temperature: value[20] / 2,
                        saturday_p4_hour: value[21],
                        saturday_p4_minute: value[22],
                        saturday_p4_temperature: value[23] / 2,
                        sunday_p1_hour: value[24],
                        sunday_p1_minute: value[25],
                        sunday_p1_temperature: value[26] / 2,
                        sunday_p2_hour: value[27],
                        sunday_p2_minute: value[28],
                        sunday_p2_temperature: value[29] / 2,
                        sunday_p3_hour: value[30],
                        sunday_p3_minute: value[31],
                        sunday_p3_temperature: value[32] / 2,
                        sunday_p4_hour: value[33],
                        sunday_p4_minute: value[34],
                        sunday_p4_temperature: value[35] / 2,
                    },
                };
            case tuya.dataPoints.state: // Thermostat on standby = OFF, running = ON
                if (model.model === 'BAC-002-ALZB') {
                    return {system_mode: value ? 'cool' : 'off'};
                } else {
                    return {system_mode: value ? 'heat' : 'off'};
                }
            case tuya.dataPoints.moesChildLock:
                return {child_lock: value ? 'LOCK' : 'UNLOCK'};
            case tuya.dataPoints.moesHeatingSetpoint:
                return {current_heating_setpoint: value};
            case tuya.dataPoints.moesMinTempLimit:
                return {min_temperature_limit: value};
            case tuya.dataPoints.moesMaxTempLimit:
                return {max_temperature_limit: value};
            case tuya.dataPoints.moesMaxTemp:
                return {max_temperature: value};
            case tuya.dataPoints.moesDeadZoneTemp:
                return {deadzone_temperature: value};
            case tuya.dataPoints.moesLocalTemp:
                temperature = value & 1<<15 ? value - (1<<16) + 1 : value;
                if (!['_TZE200_ztvwu4nk', '_TZE200_ye5jkfsb'].includes(meta.device.manufacturerName)) {
                    // https://github.com/Koenkk/zigbee2mqtt/issues/11980
                    temperature = temperature / 10;
                }

                return {local_temperature: parseFloat(temperature.toFixed(1))};
            case tuya.dataPoints.moesTempCalibration:
                temperature = value;
                // for negative values produce complimentary hex (equivalent to negative values)
                if (temperature > 4000) temperature = temperature - 4096;
                return {local_temperature_calibration: temperature};
            case tuya.dataPoints.moesHold: // state is inverted, preset_mode is deprecated
                return {preset_mode: value ? 'program' : 'hold', preset: value ? 'program' : 'hold'};
            case tuya.dataPoints.moesScheduleEnable: // state is inverted, preset_mode is deprecated
                return {preset_mode: value ? 'hold' : 'program', preset: value ? 'hold' : 'program'};
            case tuya.dataPoints.moesValve:
                return {heat: value ? 'OFF' : 'ON', running_state: (value ? 'idle' : (model.model === 'BAC-002-ALZB' ? 'cool' : 'heat'))};
            case tuya.dataPoints.moesSensor:
                switch (value) {
                case 0:
                    return {sensor: 'IN'};
                case 1:
                    return {sensor: 'AL'};
                case 2:
                    return {sensor: 'OU'};
                default:
                    return {sensor: 'not_supported'};
                }
            case tuya.dataPoints.bacFanMode:
                return {fan_mode: tuya.fanModes[value]};
            default: // DataPoint 17 is unknown
                meta.logger.warn(`zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #${
                    dp} with data ${JSON.stringify(dpValue)}`);
            }
        },
    },
```

VoltickVL, 2023-01-14:
same problem https://github.com/Koenkk/zigbee2mqtt/issues/11980

Koenkk, 2023-01-14:
`_TZE200_5toc8efa` is supported in the dev branch (and the next release on 1 February) (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-14 by Koenkk

VladimirTuzovGitHub, 2023-01-14:
I installed the z2m dev branch and reconnected the thermostat, I also removed my converter from the configuration. The thermostat is connected and says that it is supported in z2m, but all the problems I described above are also present, the thermostat does not work correctly.

VladimirTuzovGitHub, 2023-01-14:
> `_TZE200_5toc8efa`поддерживается в ветке dev (и следующий выпуск 1 февраля) ( https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html )

I installed the z2m dev branch and reconnected the thermostat, I also removed my converter from the configuration. The thermostat is connected and says that it is supported in z2m, but all the problems I described above are also present, the thermostat does not work correctly.

VladimirTuzovGitHub, 2023-01-14:
And the log gives the same errors.

2023-01-14 19:09:19zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,250]}}
2023-01-14 19:09:19zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #103 with data {"dp":103,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,1]}}
2023-01-14 19:09:19zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #104 with data {"dp":104,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,194]}}
2023-01-14 19:09:19zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #110 with data {"dp":110,"datatype":0,"data":{"type":"Buffer","data":[1,6,0,0,200,11,30,0,200,13,30,0,200,17,30,0,200]}}
2023-01-14 19:09:19zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #109 with data {"dp":109,"datatype":0,"data":{"type":"Buffer","data":[2,6,0,0,200,11,30,0,200,13,30,0,200,17,30,0,200]}}
2023-01-14 19:09:19zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #108 with data {"dp":108,"datatype":0,"data":{"type":"Buffer","data":[3,6,0,0,200,11,30,0,200,13,30,0,200,17,30,0,200]}}
2023-01-14 19:09:19zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #107 with data {"dp":107,"datatype":0,"data":{"type":"Buffer","data":[4,6,0,0,200,11,30,0,200,13,30,0,200,17,30,0,200]}}
2023-01-14 19:09:20zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #106 with data {"dp":106,"datatype":0,"data":{"type":"Buffer","data":[5,6,0,0,200,11,30,0,200,13,30,0,200,17,30,0,200]}}
2023-01-14 19:09:20zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #106 with data {"dp":106,"datatype":0,"data":{"type":"Buffer","data":[5,6,0,0,200,11,30,0,200,13,30,0,200,17,30,0,200]}}
2023-01-14 19:09:20zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #105 with data {"dp":105,"datatype":0,"data":{"type":"Buffer","data":[6,6,0,0,200,11,30,0,200,13,30,0,200,17,30,0,200]}}

VoltickVL, 2023-01-15:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5334

VladimirTuzovGitHub, 2023-01-15:
> [Koenkk/zigbee-пастух-преобразователи#5334](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5334)

Так удалось решить? Ато что-то я не пойму.

VoltickVL, 2023-01-15:
Пока нет, но я думаю мы на верном пути

directenser, 2023-01-18:
Добрый день!
Удалось завершить работы по конвертеру?

VoltickVL, 2023-01-20:
> Добрый день! Удалось завершить работы по конвертеру?

Добрый день я внёс правки в свой конвертер и всё работает https://github.com/Koenkk/zigbee2mqtt/issues/16044

directenser, 2023-01-22:
> > Добрый день! Удалось завершить работы по конвертеру?
> 
> Добрый день я внёс правки в свой конвертер и всё работает #16044

Спасибо!