Koenkk/zigbee2mqtt #16118: New device EARU din smart relay TS0601 TZE204_davzgqq0. Add support or converter.

VladimirTuzovGitHub, 2023-01-12:
### Link

https://aliexpress.com/item/1005004894566311.html

### Database entry

{"id":45,"type":"Router","ieeeAddr":"0x84b4dbfffec08548","nwkAddr":18204,"manufId":4098,"manufName":"_TZE204_davzgqq0","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��\u0010+i��\u0010+\u0013�l\u000f+\u0013�l\u000f+i�l\u000f+i��\u0010+i�l\u000f+i��\u0010+i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_davzgqq0","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1673517086744,"defaultSendRequestWhen":"immediate"}

### Comments

Hello! Please help me to make a converter for EARU din smart relay TS0601 TZE204_davzgqq0, works with TuYa. Its functions are similar to TuYa TS011F_din_smart_relay. It is not currently supported in z2m.

Avdeich, 2023-01-13:
Hello. I took as a basis the converter from DAC2161C. Not everything works.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const fzLocal = {
    tuya_xocadinrail_switch: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            for (const dpValue of msg.data.dpValues) {
                const value = tuya.getDataValue(dpValue);
                const dp = dpValue.dp;
                
                switch (dp) {
                    case 16: // DPID that we added to common
                       return {   state: value ? 'ON' : 'OFF'};
                       
                      
                    case 1:
                        return {
                            energy: value / 100
                        };
                    case 6:   
                     return {
                            current: ((value[4] | value[3] << 8) / 1000), voltage: ((value[1] | value[0] << 8) / 10), power: ((value[7] | value[6] << 8))
                        };
                        
                    case 17:
                      const lookup = {
                       0: 'not set',
                       1: 'Over current threshold',
                       3: 'Over voltage threshold',
                                      };
                        return {
                            threshold_1: lookup[value[0]],
                            threshold1_value : (value[3] | value[2] << 8),
                            threshold_2: lookup[value[4]],
                            threshold2_value : (value[7] | value[6] << 8),                            
                    };    
                        
                      case 10:{
                          const lookup = {
                          0: 'clear',
                          1: 'over current threshold',
                          2: 'over power threshold',
                          4: 'over voltage threshold',
                          8: 'wrong frequency threshold',
                           };
                           return { alarm : lookup[value]} ;
                    }
                }
            }
        },
    },
};

const tzLocal = {
    state: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointBool(entity, 16, value === 'ON');
        },
    },
};

const definition = {
    fingerprint: [{
        modelID: 'TS0601',
        manufacturerName: '_TZE204_davzgqq0'
    }],
    model: 'EAYCBM-Z_2P',
    vendor: 'Tuya',
    extend: extend.switch(),
    description: 'DIN Rail Smart Energy Meter',
    fromZigbee: [fzLocal.tuya_xocadinrail_switch, ],
    toZigbee: [tzLocal.state],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.switch().setAccess('state', ea.STATE_SET), 
              e.voltage(), 
              e.power(), 
              e.current(), 
              e.energy(),
              exposes.text('threshold_1', ea.STATE),
              exposes.text('threshold1_value', ea.STATE),
              exposes.text('threshold_2', ea.STATE),
              exposes.text('threshold2_value', ea.STATE),
              exposes.text('alarm', ea.STATE),
              ] 
};

module.exports = definition;

DNX, 2023-01-13:
@Avdeich please attach here your `external_converter` file, I'll test it on my device.

Igos439, 2023-01-14:
Good afternoon. not everything works for me either . I took DAC2161C as a basis.
[avtomat.txt](https://github.com/Koenkk/zigbee2mqtt/files/10417883/avtomat.txt)

Avdeich, 2023-01-14:
[TZE204_davzgqq0_2.txt](https://github.com/Koenkk/zigbee2mqtt/files/10417985/TZE204_davzgqq0_2.txt)

thomas-schwieters, 2023-02-06:
I also have a circuit breaker with this ID. I just tried the latest converter from @Avdeich and it seems to work.
However I'm still getting some log messages about unknown messages. E.g.
`Zigbee2MQTT:warn  2023-02-06 20:03:59: zigbee-herdsman-converters:: NOT RECOGNIZED DP #18 with data {"seq":22016,"dpValues":[{"dp":18,"datatype":0,"data":{"type":"Buffer","data":[1,1,0,35,3,1,1,9,4,1,0,180]}}]} VALUE = #	�
Zigbee2MQTT:warn  2023-02-06 20:03:59: zigbee-herdsman-converters:: NOT RECOGNIZED DP #19 with data {"seq":22272,"dpValues":[{"dp":19,"datatype":3,"data":{"type":"Buffer","data":[89,75,76,68,45,90,67,72,32,86,49,46,48]}}]} VALUE = YKLD-ZCH V1.0`

Do you also have this Problem? Is there any way to ignore these messages? Because they keep popping up in the web-ui.

Also I have a question about the parameters. My circuit breaker is labeled as a 230V 16A 1p.
Now there are these two parameters:  threshold_1 and threshold_2. Do they mean, when the circuit will break?
If so, then it seems to be wrong for mine, because mine is set to:

```
{
    "current": 0,
    "energy": 0,
    "last_seen": "2023-02-06T20:07:00+01:00",
    "linkquality": 123,
    "power": 0,
    "state": "OFF",
    "threshold1_value": 30,
    "threshold2_value": 0,
    "voltage": 232.3
}
```
Which would mean 30A...? Or what does this threshold mean?

skillix, 2023-03-02:
> 
> ```
> {
>     "current": 0,
>     "energy": 0,
>     "last_seen": "2023-02-06T20:07:00+01:00",
>     "linkquality": 123,
>     "power": 0,
>     "state": "OFF",
>     "threshold1_value": 30,
>     "threshold2_value": 0,
>     "voltage": 232.3
> }
> ```
> 
> Which would mean 30A...? Or what does this threshold mean?

hello, wouldn't the 30 be rather the 30mA protection?

[image]

I also have this module, I can't wait for it to be available to control my hot water tank

ZHNovell, 2023-03-03:
> 

Не появилось решение? у меня так же эта ошибка.

timewolf, 2023-03-05:
hello i've the same circuit breaker.
using avtomat.txt seems to works fine but i've a lot of "info" logs
`MQTT publish: topic 'zigbee2mqtt/Prese Allarme', payload '{"clear_fault":null,"current":0.09,"device":{"applicationVersion":70,"dateCode":"","friendlyName":"Prese Allarme","hardwareVersion":1,"ieeeAddr":"0x84b4dbfffec09000","manufacturerID":4098,"manufacturerName":"_TZE204_davzgqq0","model":"TS0601_din_2","networkAddress":38706,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":0.99,"fault":null,"linkquality":168,"meter_id":`

what does it means?

kva4991, 2023-03-07:
Hello, I also have this device, I can test your solution if necessary)

Creo2005, 2023-03-13:
I have the same relay. They helped me, they wrote a converter, but they settled on voltage and current values ​​\u200b\u200bthat cannot be recorded. Here is my converter, maybe it will help to complete the work.
[image]
[image]
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const libColor = require('zigbee-herdsman-converters/lib/color');
const utils = require('zigbee-herdsman-converters/lib/utils');
const zosung = require('zigbee-herdsman-converters/lib/zosung');
const fzZosung = zosung.fzZosung;
const tzZosung = zosung.tzZosung;
const ez = zosung.presetsZosung;
const globalStore = require('zigbee-herdsman-converters/lib/store');
const {Buffer} = require('buffer');

//function thresholdParser(v, statusLookup, stateLookup) {
//  const buffer = Buffer.from(v, 'base64');
//  return {
//    status: tuya.valueConverterBasic.lookup(statusLookup).from(buffer[0]),
//    state: tuya.valueConverterBasic.lookup(stateLookup).from(buffer[1]),
//    value: (buffer[3] | buffer[2] << 8),
//  };
//}

const valueConverter = {
  ...tuya.valueConverter,
  thresholdVariant1: (statusLookup, stateLookup) => {
    return {
      from: (v, meta) => {
        const buffer = Buffer.from(v, 'base64');
        meta.logger.debug(`TEST2: ${JSON.stringify(buffer)}`);
        return {
          status: buffer[0],//tuya.valueConverterBasic.lookup(statusLookup).from(buffer[0]),
          state: buffer[1],//tuya.valueConverterBasic.lookup(stateLookup).from(buffer[1]),
          value: (buffer[3] | buffer[2] << 8),
        };
      },
      to: (v) => {
        const payload = [];
        payload.push(/*tuya.valueConverterBasic.lookup(statusLookup).to(v.status)*/v.status);
        payload.push(/*tuya.valueConverterBasic.lookup(stateLookup).to(v.state)*/v.state);
        payload.push(...tuya.convertDecimalValueTo2ByteHexArray(v.value));
        return payload;
      },
    };
  },
  thresholdVariant2: (names, statusLookup, stateLookup) => {
    return {
      from: (v, meta) => {
        const thresholds = {};
        const chunkSize = 4;
        meta.logger.debug(`TEST0: ${JSON.stringify(v)}`);
        for (const [i, name] of Object.entries(names)) {
          const idx = i * chunkSize;
          const thresholdPayload = v.slice(idx, idx + chunkSize);
          meta.logger.debug(`TEST1: name: ${name}, idx: ${idx}, payload: ${JSON.stringify(thresholdPayload)}`);
          thresholds[`${name}_threshold`] = valueConverter.thresholdVariant1(statusLookup, stateLookup).from(thresholdPayload, meta);
        }
        return thresholds;
      },
      to: (v) => {
        const payload = [];
        for (const threshold of Object.values(v)) payload.push(...valueConverter.thresholdVariant1(statusLookup, stateLookup).to(threshold));
        return payload;
[... 37 lines cut ...]
        .withFeature(exposes.numeric('value', ea.STATE_SET).withUnit('V').withValueMax(300).withValueMin(50).withValueStep(1)
          .withPreset('default', 180, 'Default value').withDescription('Value of trigger for undervoltage threshold'))),
    exposes.text('breaker_id', ea.STATE).withDescription('Device identificator from manufacturer'),
  ],
  meta: {
      tuyaDatapoints: [
          [1, 'energy', tuya.valueConverter.divideBy100],
          [6, null, tuya.valueConverter.phaseVariant2],
          [16, 'state', tuya.valueConverter.onOff],
          // {"type":"Buffer","data":[4,1,0,30]} - leakage enabled
          // {"type":"Buffer","data":[4,0,0,30]} - leakage disabled
          [17, 'leakage_threshold', valueConverter.thresholdVariant1({4: 4}, {'ON': 1, 'OFF': 0})],
          //[17, 'leakage_threshold', valueConverter.threshold2],
          // {"type":"Buffer","data":[1,1,0,30,3,0,0,250,4,0,0,150]} - overcurr enabled 30A | overvolt enabled 250V | undervolt enabled 150V
          // {"type":"Buffer","data":[1,0,0,21,3,0,0,250,4,0,0,150]} - overcurr disabled 21A | overvolt disabled 250V | undervolt disabled 150V
          [18, 'thresholds', valueConverter.thresholdVariant2(['overcurrent', 'overvoltage', 'undervoltage'], {1: 1, 3: 3, 4: 4}, {'ON': 1, 'OFF': 0})],
          //[18, 'overcurrent_threshold', valueConverter.thresholdVariant2([], {1: 1}, {'ON': 1, 'OFF': 0})],
          //[18, 'overvoltage_threshold', valueConverter.thresholdVariant2([], {3: 3}, {'ON': 1, 'OFF': 0})],
          //[18, 'undervoltage_threshold', valueConverter.thresholdVariant2([], {4: 4}, {'ON': 1, 'OFF': 0})],
          [19, 'breaker_id', tuya.valueConverter.raw],
      ],
  },
};

module.exports = definition;

Avdeich, 2023-03-13:
Hello. Download converter file? The fact that the above plane must be edited.

kva4991, 2023-03-14:
[Creo2005](https://github.com/Creo2005) 
https://github.com/Koenkk/zigbee2mqtt/issues/16118#issuecomment-1465844355
Hello. Your code gives me an error in the z2m logs, could you help me fix it, please?

[image]

Creo2005, 2023-03-14:
> [Creo2005](https://github.com/Creo2005) [#16118 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/16118#issuecomment-1465844355) Hello. Your code gives me an error in the z2m logs, could you help me fix it, please? p.s.: I'm not good at programming) [image] [image]

this is a test code for developers, it still needs to be finalized

lucasimons, 2023-03-15:
@Creo2005 Can you post the working code?

Creo2005, 2023-03-15:
> @Creo2005 Can you post the working code?

Sorry, no, this code started make a friend, but never finished. Maybe someone here can fix it :)

lucasimons, 2023-03-19:
@VladimirTuzovGitHub can you adjust your code with the @Creo2005 's coda?

sash1024-cloud, 2023-04-12:
Dear @Koenkk,
What can I do to help debugging?

Saxtus, 2023-04-17:
Unfortunately, @Creo2005 copied/pasted the code instead of attaching it, so GitHub converted it using markdown and these automatic changes made the code unusable.

I took the liberty to un-markdown his code and attach it here [_TZE204_davzgqq0.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/11247935/_TZE204_davzgqq0.js.txt)
(remember to remove the .txt extension as GitHub didn't allow me to upload it directly as .js)

I have no idea how it works, but at least it doesn't produce any errors.
Enjoy...

Creo2005, 2023-04-17:
> Unfortunately, @Creo2005 copied/pasted the code instead of attaching it, so GitHub converted it using markdown and these automatic changes made the code unusable.
> 
> I took the liberty to un-markdown his code and attach it here [_TZE204_davzgqq0.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/11247935/_TZE204_davzgqq0.js.txt) (remember to remove the .txt extension as GitHub didn't allow me to upload it directly as .js)
> 
> I have no idea how it works, but at least it doesn't produce any errors. Enjoy...
> 
> Now if only someone could build upon it to expose the switch about remembering the last state after a power failure, that would be sweet.

[ts0601.js.zip](https://github.com/Koenkk/zigbee2mqtt/files/11248077/ts0601.js.zip)
Code in file.

scaytrase, 2023-05-10:
I can confirm this JS kind of works, but getting errors

```
Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}
```

and doesn't report any current or power consumption

TolianIPB, 2023-05-21:
Add support pleaze.

Koenkk, 2023-05-22:
Did somebody find out the datapoints of this device? https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

laplac, 2023-06-27:
> [ts0601.js.zip](https://github.com/Koenkk/zigbee2mqtt/files/11248077/ts0601.js.zip) Code in file.

Tested on EARU EAYCBM-Z-1P https://www.aliexpress.com/item/1005004894129906.html 
Power , Current , Voltage and sum of energy working perfectly. 
Ignored leakage alarm as it only supported on 2P devices. 

Thanks

meganiekje, 2023-07-03:
@Koenkk  Hi Koen, I love your work and effort you put into this !!!

I als o have the TS0601_din_1 _TZE204_davzgqq0  device, and used the above ts0601.js as an external convertor.
It works partly: the voltage, amps, power and on/off switch work fine.

The energy does not work, it keeps increasing even if the amps are 0.
It looks like it has some kind of timer running that counts the seconds the breaker is switched ON.
That timer resets every 12 hours.
Maybe, the energy uses that timer to report a daily usage ?

Also, and that is what worries me: I can not switch on the overcurrent threshold alarm:

[image]
 
I can move the off/on switch, and click 'apply', but the next time I check the status, it is OFF again.

In the debug I see no change, the output stays like this:

2023-07-03 15:59:46Received Zigbee message from 'KookplaatFuse', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1,0,0,24,3,0,1,144,4,0,0,50],"type":"Buffer"},"datatype":0,"dp":18}],"seq":60672}' from endpoint 1 with groupID 0

2023-07-03 15:59:46MQTT publish: topic 'zigbee2mqtt/KookplaatFuse', payload '{"breaker_id":"YKLD-ZCH V1.0","current":0,"energy":14.34,"leakage_threshold":{"state":1,"status":4,"value":30},"linkquality":66,"power":0,"state":"ON","thresholds":{"overcurrent_threshold":{"state":0,"status":1,"value":24},"overvoltage_threshold":{"state":0,"status":3,"value":400},"undervoltage_threshold":{"state":0,"status":4,"value":50}},"voltage":239.6}'

Now I'm afraid the breaker is not going to switch off, if more than 25Amps is running.
that would result in a dangerous situation.

Please Help !!

jampez77, 2023-07-05:
I just bought 2 of these devices (2 poles version) and so far it mostly works but here are a few issues. 

Energy doesn't work.

on/off switch is sometimes unresponsive. switch state is not updated when you turn it off using the test function.

Setting the limits don't work.

meganiekje, 2023-08-05:
This problem is still present.
It would be very nice if it can be fixed

Nordranei, 2023-08-08:
I manage to get state, energy, power, voltage and current to work.
[image]
But rest of the states do not work
[image]

Lotambena, 2023-09-02:
The switch lacks the function remering its states. When ever there is a blackout the switch will turn off and will not turn back on when power returs. Is there a data point for this function? Am currently using an automation to to turn it on

paris21aek, 2023-09-11:
Does anyone find any working solution for this switch? It came today and I can't work it.

Koenkk, 2023-09-24:
Did somebody already grab the datapoints of this device? https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

Creo2005, 2023-09-24:
> Did somebody already grab the datapoints of this device? https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

[ts0601.js.zip](https://github.com/Koenkk/zigbee2mqtt/files/12709381/ts0601.js.zip)

beliy1985, 2023-09-24:
> [](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html)

I don’t have a gateway, I would do it right now, people connect @meganiekje @VladimirTuzovGitHub @Avdeich

vipmenable, 2023-09-25:
automatic script output for 1P device:
{"1":"Total forward energy","6":"Phase A","16":"Switch","17":"Alarm set1","18":"Alarm set2","19":"Breaker id"}

beliy1985, 2023-09-25:
> automatic script output for 1P device: {"1":"Total forward energy","6":"Phase A","16":"Switch","17":"Alarm set1","18":"Alarm set2","19":"Breaker id"}

@Koenkk

Koenkk, 2023-09-26:
The datapoints match the converter which was provided here https://github.com/Koenkk/zigbee2mqtt/issues/16118#issuecomment-1732640264 . But if I understand correctly, this has the issue that the energy resets every day? Is this also the case when using it with the TuYa gateway`/

danielszilagyi, 2023-09-26:
> The datapoints match the converter which was provided here [#16118 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/16118#issuecomment-1732640264) . But if I understand correctly, this has the issue that the energy resets every day? Is this also the case when using it with the TuYa gateway?

Negative, at least on the device I am using (1P) energy is measured correctly for more than a week now. Only problem is that it is not turning back on.

```yaml
breaker_id: YKLD-ZCH V1.0
current: 0.28
device:
  applicationVersion: 70
  dateCode: ''
  friendlyName: Szerver_Klima
  hardwareVersion: 1
  ieeeAddr: '0x84b4dbfffec0851a'
  manufacturerID: 4098
  manufacturerName: _TZE204_davzgqq0
  model: TS0601_din_1
  networkAddress: 16555
  powerSource: Mains (single phase)
  stackVersion: 0
  type: Router
  zclVersion: 3
energy: 68.6
leakage_threshold:
  state: 1
  status: 4
  value: 30
linkquality: 135
power: 33
thresholds:
  overcurrent_threshold:
    state: 1
    status: 1
    value: 35
  overvoltage_threshold:
    state: 1
    status: 3
    value: 265
  undervoltage_threshold:
    state: 1
    status: 4
    value: 180
voltage: 236.1
friendly_name: Szerver_Klima
`

beliy1985, 2023-09-26:
> Negative, at least on the device I am using (1P) energy is measured correctly for almost a month now

Please share your converter, it doesn’t work for me: thresholds, energy ...

danielszilagyi, 2023-09-26:
> > Negative, at least on the device I am using (1P) energy is measured correctly for almost a month now
> 
> Please share your converter, it doesn’t work for me: thresholds, energy ...

https://github.com/Koenkk/zigbee2mqtt/issues/16118#issuecomment-1732640264

beliy1985, 2023-09-26:
@danielszilagyi Does overcurrent_threshold work for you?

danielszilagyi, 2023-09-26:
> @danielszilagyi Does overcurrent_threshold work for you? my energy does not work, it keeps increasing even if the amps are 0

Didn’t try. Energy works as expected.

meganiekje, 2023-09-26:
In my case the voltage,  current and power work.
Energy and current threshold do not work.
Current treshold is important, otherwise the device does not function as a circuit breaker.

Koenkk, 2023-09-27:
@danielszilagyi 

>  Only problem is that it is not configurable to turn back on after a power failure (but it is a circuit breaker anyway)

According to the datapoints, the device doesn't support such feature

> Current treshold is important, otherwise the device does not function as a circuit breaker.

What can be configured via the TuYa app?

danielszilagyi, 2023-09-27:
> @danielszilagyi Does overcurrent_threshold work for you? my energy does not work, it keeps increasing even if the amps are 0

The converter must have a bug, as I get an error around `convertDecimalValueTo2ByteHexArray`
when trying to set the overcurrent value.
The overcurrent on/off toggle works. 

@Koenkk, I do not have a tuya hub. But I believe the provided datapoints are correct, just some coding mistake in the converter.

Anything else we can provide as input?

Koenkk, 2023-09-27:
@danielszilagyi can you try with https://gist.github.com/Koenkk/4477f1d8bce028ec5654833168607e1d 

I guess you also uncomment: 

```
          //[18, 'overcurrent_threshold', valueConverter.thresholdVariant2([], {1: 1}, {'ON': 1, 'OFF': 0})],
          //[18, 'overvoltage_threshold', valueConverter.thresholdVariant2([], {3: 3}, {'ON': 1, 'OFF': 0})],
          //[18, 'undervoltage_threshold', valueConverter.thresholdVariant2([], {4: 4}, {'ON': 1, 'OFF': 0})],
```

Or how do you set the thresholds?

danielszilagyi, 2023-09-27:
> @danielszilagyi can you try with https://gist.github.com/Koenkk/4477f1d8bce028ec5654833168607e1d
>
> I guess you also uncomment:
>
> ```
>           //[18, 'overcurrent_threshold', [...]

Honestly I have not tried voltage/amp limits at all before today evening, it was someone else's comment. I'll take a look and report back soon.

danielszilagyi, 2023-09-27:
> @danielszilagyi can you try with https://gist.github.com/Koenkk/4477f1d8bce028ec5654833168607e1d
>
> I guess you also uncomment:
>
> ```
>           //[18, 'overcurrent_threshold', [...]

Looks OK from UI, no errors on any of the over/undervoltage parameter changes, but I cannot confirm the amp limit is working, as I cannot generate even 10A load on the device which is the minimum setting.

I think the over/undervoltage part is not behaving correctly, got 512V for default value of undervoltage, then tried to set it to another value, if you look below at overvoltage_threshold/undervoltage_threshold I think it explains itself.

Energy monitoring works properly.

Tried setting obvious under/overvoltage settings but the device has not tripped after about 5 minutes. If I understand correctly it should switch off, or would "alarm" mean just some property changing here?

```json
{
    "breaker_id": "YKLD-ZCH V1.0",
    "current": 2.57,
    "energy": 70.58,
    "leakage_threshold": {
        "state": 1,
        "status": 4,
        "value": 30
    },
    "linkquality": 129,
    "power": 584,
    "state": "ON",
    "thresholds": {
        "overcurrent_threshold": {
            "state": 0,
            "status": 1,
            "value": 204
        },
        "overvoltage_threshold": {
            "state": 1,
            "status": 3,
            "value": 264
        },
        "undervoltage_threshold": {
            "state": 170,
            "status": 4,
            "value": 512
        }
    },
    "voltage": 232.4,
```

Koenkk, 2023-09-27:
@danielszilagyi 

> If I understand correctly it should switch off, or would "alarm" mean just some property changing here?

I don't know how the device should behave, this has to be checked with the original TuYa gateway

danielszilagyi, 2023-09-27:
> @danielszilagyi
> 
> > If I understand correctly it should switch off, or would "alarm" mean just some property changing here?
> 
> I don't know how the device should behave, this has to be checked with the original TuYa gateway

Yes, totally understood. For me the working functionality is fine. Thank you.

beliy1985, 2023-09-27:
> > @danielszilagyi
> > > Если я правильно понимаю, он должен выключиться, или «тревога» будет означать просто изменение какого-то свойства здесь?
> > 
> > 
> > Не знаю, как должно себя вести устройство, это надо сверять с оригинальным шлюзом ТуЯ
> 
> Да, полностью понял. Для меня рабочий функционал в порядке. Спасибо.

Doesn't your energy increase if you turn off the relay?

beliy1985, 2023-09-27:
> @danielszilagyi
> 
> > If I understand correctly it should switch off, or would "alarm" mean just some property changing here?
> 
> I don't know how the device should behave, this has to be checked with the original TuYa gateway

In TuYa, current and voltage limits also work:(https://www.youtube.com/watch?v=u2FraPijDus&t=668s)

meganiekje, 2023-09-27:
I just tried the externel converter. I can set de overcurrent value, and move the switch to ON.
When i set the overcurrent value to 10, and i set my induction kookplaat to 13 amps, the breaker does NOT switch off.
When i refresh the zigbee2mqtt webpage, the overcurrent switch is in OFF position.
I have a debug log of this action, if needed.
I do not have a tuya bridge.

beliy1985, 2023-09-27:
> I just tried the externel converter. I can set de overcurrent value, and move the switch to ON. When i set the overcurrent value to 10, and i set my induction kookplaat to 13 amps, the breaker does [...]

I have the same problem

meganiekje, 2023-09-27:
Also, now the device reports that it has used 34166 kWh of energy  ;)

beliy1985, 2023-09-27:
> Also, now the device reports that it has used 34166 kWh of energy ;)

And continues to increase when the relay is off?

danielszilagyi, 2023-09-27:
> Doesn't your energy increase if you turn off the relay?

I kept it off for 30 minutes and it didn’t, not sure how much increase you would expect

beliy1985, 2023-09-27:
> I kept it off for 30 minutes and it didn’t, not sure how much increase you would expect

[image]

meganiekje, 2023-09-27:
> > Also, now the device reports that it has used 34166 kWh of energy ;)
> 
> And continues to increase when the relay is off?

I have seen that happening a few times, but it changes if i disconnect the device from the 230volt, and reconect it again. Sometimes it does increase the energy even when no power is consumed.

danielszilagyi, 2023-09-27:
No, nothing like this.

danielszilagyi, 2023-09-27:
> [image] [ [image] ](https://user-images.githubusercontent.com/43289664/271096069-2fe4d690-53cb-4a1a-8b1c-c8e1611e6f34.gif) [ ](https://user-images.githubusercontent.com/43289664/271096069-2fe4d690-53cb-4a1a-8b1c-c8e1611e6f34.gif)

No, nothing like this.

meganiekje, 2023-09-27:
[image]
I have even seen this behavior on the energy value. LOL

[164 more replies]

closed 2025-07-05 by github-actions[bot]