Koenkk/zigbee2mqtt #16121: New device EARU power and energy meter in kWh din2 smart relay TS0001 _TZ3000_ajv2vfow. Add support or converter.

VladimirTuzovGitHub, 2023-01-12:
### Link

https://aliexpress.com/item/1005004828497741.html

### Database entry

{"id":45,"type":"Router","ieeeAddr":"0xa4c1381c9d677932","nwkAddr":41224,"manufId":4417,"manufName":"_TZ3000_ajv2vfow","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���*\u0013���*\u0012\u0014\u0000\u0000\u0000\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3000_ajv2vfow","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":1,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c95141","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":153093895},"lastSeen":1673530738988,"defaultSendRequestWhen":"immediate"}

### Comments

When connected to z2m, it does not determine it correctly, it defines it as a TuYa TS0001 - 1 gang switch. But this is a smart relay on a DIN rail with the function of an energy monitor and voltage input. Please help me make a converter for it.

Koenkk, 2023-01-12:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    fingerprint: tuya.fingerprint('TS0001', ['_TZ3000_xkap8wtb', '_TZ3000_qnejhcsu', '_TZ3000_ajv2vfow', '_TZ3000_x3ewpzyr', '_TZ3000_mkhkxx1p']),
    model: 'TS0001_power',
    description: 'Switch with power monitoring',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report,
        tuya.fz.power_outage_memory, tuya.fz.switch_type],
    toZigbee: [tz.on_off, tuya.tz.power_on_behavior, tuya.tz.switch_type],
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE), e.energy(), tuya.exposes.switchType(),
        exposes.enum('power_outage_memory', ea.ALL, ['on', 'off', 'restore']).withDescription('Recover state after power outage')],
};

module.exports = definition;

```
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

VladimirTuzovGitHub, 2023-01-12:
Gives an error message:
Failed to configure '0xa4c1381c9d677932', attempt 1 (Error: ConfigureReporting 0xa4c1381c9d677932/1 haElectricalMeasurement([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {" sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0," manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 3179 - 1 - 39 - 2820 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/ utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))

It can only be turned on and off. Volt, Amp, Voltage, Energy shows Null.

VladimirTuzovGitHub, 2023-01-13:
The seller wrote that they had an error and that the zigbee version without an energy monitor, only has WiFi an energy monitor. I apologize for the inconvenience, I'm closing the appeal.

closed 2023-01-13 by VladimirTuzovGitHub