Koenkk/zigbee2mqtt #16129: Silvercrest 3 gang switch (HG06338) socket state go to null

maxime-killinger, 2023-01-12:
### What happened?

I'm having issue using my new Silvercrest 3 gang switch (HG06338). 
When I toggle the one socket, the state of the others is lost.
In the logs we can see that when I toggle on a socket, zigbee2mqtt send an update for all three sockets with one at on the to other at null.

### What did you expect to happen?

Keep the state of the unused sockets.

### How to reproduce it (minimal and precise)

- Connect Silvercrest 3 gang switch (HG06338). 
- Go to the "Exposes" tab
- Toggle on one socket, the two other should be in "?" state
- Toggle another one, the first should be now in "?"  state

### Zigbee2MQTT version

1.29.1-dev

### Adapter firmware version

20220219

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coordinator

### Debug log

```
Info 2023-01-12 20:47:20MQTT publish: topic 'zigbee2mqtt/Multiprise', payload '{"device":{"applicationVersion":69,"dateCode":"","friendlyName":"Multiprise","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffe3c3287","manufacturerID":4098,"manufacturerName":"_TZ3000_vzopcetz","model":"HG06338","networkAddress":42093,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":163,"state_l1":null,"state_l2":"ON","state_l3":null}'
Info 2023-01-12 20:47:20MQTT publish: topic 'zigbee2mqtt/Multiprise', payload '{"device":{"applicationVersion":69,"dateCode":"","friendlyName":"Multiprise","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffe3c3287","manufacturerID":4098,"manufacturerName":"_TZ3000_vzopcetz","model":"HG06338","networkAddress":42093,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":160,"state_l1":null,"state_l2":"ON","state_l3":null}'
Info 2023-01-12 20:47:21MQTT publish: topic 'zigbee2mqtt/Multiprise', payload '{"device":{"applicationVersion":69,"dateCode":"","friendlyName":"Multiprise","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffe3c3287","manufacturerID":4098,"manufacturerName":"_TZ3000_vzopcetz","model":"HG06338","networkAddress":42093,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":163,"state_l1":null,"state_l2":null,"state_l3":"ON"}'
Info 2023-01-12 20:47:21MQTT publish: topic 'zigbee2mqtt/Multiprise', payload '{"device":{"applicationVersion":69,"dateCode":"","friendlyName":"Multiprise","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffe3c3287","manufacturerID":4098,"manufacturerName":"_TZ3000_vzopcetz","model":"HG06338","networkAddress":42093,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":163,"state_l1":null,"state_l2":null,"state_l3":"ON"}'
Info 2023-01-12 20:47:22MQTT publish: topic 'zigbee2mqtt/Multiprise', payload '{"device":{"applicationVersion":69,"dateCode":"","friendlyName":"Multiprise","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffe3c3287","manufacturerID":4098,"manufacturerName":"_TZ3000_vzopcetz","model":"HG06338","networkAddress":42093,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":163,"state_l1":"ON","state_l2":null,"state_l3":null}'
Info 2023-01-12 20:47:22MQTT publish: topic 'zigbee2mqtt/Multiprise', payload '{"device":{"applicationVersion":69,"dateCode":"","friendlyName":"Multiprise","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffe3c3287","manufacturerID":4098,"manufacturerName":"_TZ3000_vzopcetz","model":"HG06338","networkAddress":42093,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":163,"state_l1":"ON","state_l2":null,"state_l3":null}'
Info 2023-01-12 20:47:23MQTT publish: topic 'zigbee2mqtt/Multiprise', payload '{"device":{"applicationVersion":69,"dateCode":"","friendlyName":"Multiprise","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffe3c3287","manufacturerID":4098,"manufacturerName":"_TZ3000_vzopcetz","model":"HG06338","networkAddress":42093,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":163,"state_l1":null,"state_l2":"ON","state_l3":null}'
Info 2023-01-12 20:47:23MQTT publish: topic 'zigbee2mqtt/Multiprise', payload '{"device":{"applicationVersion":69,"dateCode":"","friendlyName":"Multiprise","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffe3c3287","manufacturerID":4098,"manufacturerName":"_TZ3000_vzopcetz","model":"HG06338","networkAddress":42093,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":163,"state_l1":null,"state_l2":"ON","state_l3":null}'
Info 2023-01-12 20:47:24MQTT publish: topic 'zigbee2mqtt/Multiprise', payload '{"device":{"applicationVersion":69,"dateCode":"","friendlyName":"Multiprise","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffe3c3287","manufacturerID":4098,"manufacturerName":"_TZ3000_vzopcetz","model":"HG06338","networkAddress":42093,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":160,"state_l1":null,"state_l2":null,"state_l3":"ON"}'
Info 2023-01-12 20:47:24MQTT publish: topic 'zigbee2mqtt/Multiprise', payload '{"device":{"applicationVersion":69,"dateCode":"","friendlyName":"Multiprise","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffe3c3287","manufacturerID":4098,"manufacturerName":"_TZ3000_vzopcetz","model":"HG06338","networkAddress":42093,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":160,"state_l1":null,"state_l2":null,"state_l3":"ON"}'
```

terragit, 2023-02-04:
I had the same problem with zigbee2mqtt version 1.22.2. Update to 1.30.0 fixed it.

maxime-killinger, 2023-02-04:
I just tried the 1.30.0 and still no luck for me. Did you do something to make it work ?

terragit, 2023-02-04:
> I just tried the 1.30.0 and still no luck for me. Did you do something to make it work ?

In short: Unfortunately, no. 

Long story:

0. got the brand new device today
1. connected it with mains (close to my zigbee controller). Instant pairing but only with L1 working.
2. Due to the errors, I pressed the power button 5 seconds to re-pair. Again, not working.
3. Searched for the error online and found your post. 
4. Wrote a "confirmation message" but realized that I had to post my zigbee2mqtt version. 
5. Hence, I looked it up and decided to make the update. I issued the following command (Debian 10):
```
cd /opt/zigbee2mqtt
sudo ./update.sh
```
zigbee2mqtt restarted automatically. I didn't even re-pair the device. 

6. Opened web-UI and found the all-working HG06338

Maybe it has something to do with the controller or the firmware of the controller? Unfortunately, I don't know which (firmware) version I have nor where I can look it up. 

Hope you find a solution.

closed 2023-03-15 by github-actions[bot]