Koenkk/zigbee2mqtt #16131: Zigbee2mqtt Failed to configure thermostate

l0ttek, 2023-01-12:
### What happened?

I used a danfoss ally thermostate and all of them not worki g after a few days, not responding for any change. I tried to reconfigure device but got this error. 

2023-01-12 21:48:04Failed to configure 'thermostate-bedroom', attempt 1 (Error: Bind 0x0c4314fffe5ddfc6/1 genPowerCfg from '0x00124b00228133fa/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))

Previously when I restarted zigbee2mqtt or whole device everything back to normal for a while, but now not working anymore..

Software version
01.18.0008 01.18
Producer
Danfoss
Model
014G2461

### What did you expect to happen?

I would like to change themperature on my thermostate. 

### How to reproduce it (minimal and precise)

Add device danfoss ally, try to change temperature after day or few days normalną work. 

### Zigbee2MQTT version

1.29.1-1

### Adapter firmware version

20220219

### Adapter

CC2652P

paszczaq, 2023-01-15:
Are you using any routers in your zigbee network?
Take a look at zigbee map and see if your TRV is connected directly to cooridinator - if yes take a look here: #13478 
Latest reports are showing that upgrade to firmware 20221226 on sonoff dongle plus helps

l0ttek, 2023-01-16:
Yes, it is connected directly to my coordinator. Thanks, I will look at this and upgrade coordinator.

l0ttek, 2023-01-16:
@paszczaq Hi, now start working, but changing temperature on this thermostate takes 3-4minutes, on others in two seconds..

l0ttek, 2023-01-19:
@paszczaq everything back to normal after upgrade dongle software to dev version, thank you.

l0ttek, 2023-01-19:
Back to normal after upgrade dongle software to 20221226 version

Gerry33, 2023-01-24:
@l0ttek : Where did you find the firmware ''20221226 '. 

I  can only find a version ''20220219'   at ' https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

Thanks 
Gerry

paszczaq, 2023-01-24:
@Gerry33: just switch branch to develop 
[image]

Gerry33, 2023-01-25:
@paszczaq : Thank you very much for the hint. GITHUB is sometimes a maze ... or I'm too dull  ...

May I also ask which one of the 2 firmwares for the CC1352P  you flashed :
- CC1352P2_CC2652P_launchpad_coordinator_20221226.zip
- CC1352P2_CC2652P_other_coordinator_20221226.zip

.. and which tool did you use ? 

I'm not familar with all this terminology. 
Again Thanks,
Gerry

l0ttek, 2023-02-01:
> @paszczaq : Thank you very much for the hint. GITHUB is sometimes a maze ... or I'm too dull ...
>
> May I also ask which one of the 2 firmwares for the CC1352P you flashed :
>
> * [...]

Hi Gerry,
sorry for late response.
You should look at the table at the bottom and find your dongle coordinator. The third column tells you which file to download.

I used this tutorial for upgrade software
https://www.youtube.com/watch?v=4S_c_m6z-RY&ab_channel=digiblurDIY

and it works :)
have fun and if you have any question, feel free to ask

l0ttek, 2023-02-01:
@paszczaq the problem still occurs, but less frequently than before. After two weeks, my two thermostats are not responding to any changes..

Gerry33, 2023-02-01:
@paszczaq : Thank you very much for your  attention. 
With your hint I got the the stick updated with the latest firmware version.

But as @l0ttek  already said above, the problem persists also in my configuration.
Will try to re-pair, remove routers, a.s.o.  Lets see.

l0ttek, 2023-02-01:
@Gerry33 After re-pair everything back to normal for a while, week, two weeks..

paszczaq, 2023-02-01:
@Gerry33 @l0ttek if possible make sure that your TRV connects via router (zigbee socket or bulb) - that should stabilise connection until Koenkk can solve this issue.
The fun fact from other people reports is that Ally works fine on ZHA :)

l0ttek, 2023-02-01:
@paszczaq Yes, each TRV are connected to router, as you can see on the image below.

https://postimg.cc/hXZvHkzD

Gerry33, 2023-02-01:
Still no luck.  Ensured connection via a lightbulb router. 
Will now try to fiddle around with devices firmware updates with no great hope.

The problem is discussed in details also here  [https://github.com/Koenkk/zigbee2mqtt/issues/13478]. Maybe we head over there, to avoid duplication.

closed 2023-03-11 by github-actions[bot]