Koenkk/zigbee2mqtt #16132: Eglo Fueva-Z mis-identified as lightbulb after firmware update

capturetheventure, 2023-01-12:
### What happened?

Similar to [#12281](https://github.com/Koenkk/zigbee2mqtt/issues/12281), I have a few ceiling mounted Eglo lights that are recognised as light bulbs and don't support all the functions of the device. This problem started after I updated the lights with the AwoX app to firmware version 2.3.13 and repaired all of them afterwards.

I did the firmware update because there were already a few issues with these lights before, e.g. they sometimes turned themself off when setting the brightness to values below 20%.

They are now tagged as: [33951/33948](https://www.zigbee2mqtt.io/devices/33951_33948.html#awox-33951%252F33948)
I don't remember exactly as which device they were recognised before, but it was some form of AwoX light bulb with variable color temperature (which was probably wrong already).

Here's a link to the device: https://www.amazon.de/dp/B09FSZ9C2H

If this should be a new device addition, I'd be happy to provide more information if needed.

Here's the log:

```
Zigbee2MQTT:info  2023-01-12 22:24:15: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2023-01-12 22:24:15: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"bmu2t-3"}'
Zigbee2MQTT:info  2023-01-12 22:24:24: Device '0xa4c138f74d3a2af4' joined
Zigbee2MQTT:info  2023-01-12 22:24:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138f74d3a2af4","ieee_address":"0xa4c138f74d3a2af4"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-01-12 22:24:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c138f74d3a2af4"},"type":"device_connected"}'
Zigbee2MQTT:info  2023-01-12 22:24:24: MQTT publish: topic 'homeassistant/light/0xa4c138f74d3a2af4/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"command_topic":"zigbee2mqtt/0xa4c138f74d3a2af4/set","device":{"identifiers":["zigbee2mqtt_0xa4c138f74d3a2af4"],"manufacturer":"AwoX","model":"LED white (33951/33948)","name":"0xa4c138f74d3a2af4","sw_version":"0122052017"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"json_attributes_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","name":"0xa4c138f74d3a2af4","schema":"json","state_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","unique_id":"0xa4c138f74d3a2af4_light_zigbee2mqtt"}'
Zigbee2MQTT:info  2023-01-12 22:24:24: MQTT publish: topic 'homeassistant/select/0xa4c138f74d3a2af4/effect/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0xa4c138f74d3a2af4/set/effect","device":{"identifiers":["zigbee2mqtt_0xa4c138f74d3a2af4"],"manufacturer":"AwoX","model":"LED white (33951/33948)","name":"0xa4c138f74d3a2af4","sw_version":"0122052017"},"enabled_by_default":false,"icon":"mdi:palette","json_attributes_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","name":"0xa4c138f74d3a2af4 effect","options":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"unique_id":"0xa4c138f74d3a2af4_effect_zigbee2mqtt"}'
Zigbee2MQTT:info  2023-01-12 22:24:24: MQTT publish: topic 'homeassistant/sensor/0xa4c138f74d3a2af4/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138f74d3a2af4"],"manufacturer":"AwoX","model":"LED white (33951/33948)","name":"0xa4c138f74d3a2af4","sw_version":"0122052017"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","name":"0xa4c138f74d3a2af4 power on behavior","state_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","unique_id":"0xa4c138f74d3a2af4_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
Zigbee2MQTT:info  2023-01-12 22:24:24: MQTT publish: topic 'homeassistant/select/0xa4c138f74d3a2af4/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0xa4c138f74d3a2af4/set/power_on_behavior","device":{"identifiers":["zigbee2mqtt_0xa4c138f74d3a2af4"],"manufacturer":"AwoX","model":"LED white (33951/33948)","name":"0xa4c138f74d3a2af4","sw_version":"0122052017"},"entity_category":"config","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","name":"0xa4c138f74d3a2af4 power on behavior","options":["off","on","toggle","previous"],"state_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","unique_id":"0xa4c138f74d3a2af4_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
Zigbee2MQTT:info  2023-01-12 22:24:24: MQTT publish: topic 'homeassistant/sensor/0xa4c138f74d3a2af4/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138f74d3a2af4"],"manufacturer":"AwoX","model":"LED white (33951/33948)","name":"0xa4c138f74d3a2af4","sw_version":"0122052017"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","name":"0xa4c138f74d3a2af4 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","unique_id":"0xa4c138f74d3a2af4_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2023-01-12 22:24:24: Starting interview of '0xa4c138f74d3a2af4'
Zigbee2MQTT:info  2023-01-12 22:24:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138f74d3a2af4","ieee_address":"0xa4c138f74d3a2af4","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2023-01-12 22:24:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c138f74d3a2af4"},"type":"pairing"}'
Zigbee2MQTT:debug 2023-01-12 22:24:24: Device '0xa4c138f74d3a2af4' announced itself
Zigbee2MQTT:info  2023-01-12 22:24:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138f74d3a2af4","ieee_address":"0xa4c138f74d3a2af4"},"type":"device_announce"}'
Zigbee2MQTT:info  2023-01-12 22:24:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xa4c138f74d3a2af4"},"type":"device_announced"}'
Zigbee2MQTT:debug 2023-01-12 22:24:24: Received MQTT message on 'homeassistant/light/0xa4c138f74d3a2af4/light/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"command_topic":"zigbee2mqtt/0xa4c138f74d3a2af4/set","device":{"identifiers":["zigbee2mqtt_0xa4c138f74d3a2af4"],"manufacturer":"AwoX","model":"LED white (33951/33948)","name":"0xa4c138f74d3a2af4","sw_version":"0122052017"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"json_attributes_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","name":"0xa4c138f74d3a2af4","schema":"json","state_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","unique_id":"0xa4c138f74d3a2af4_light_zigbee2mqtt"}'
Zigbee2MQTT:debug 2023-01-12 22:24:24: Received MQTT message on 'homeassistant/select/0xa4c138f74d3a2af4/effect/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0xa4c138f74d3a2af4/set/effect","device":{"identifiers":["zigbee2mqtt_0xa4c138f74d3a2af4"],"manufacturer":"AwoX","model":"LED white (33951/33948)","name":"0xa4c138f74d3a2af4","sw_version":"0122052017"},"enabled_by_default":false,"icon":"mdi:palette","json_attributes_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","name":"0xa4c138f74d3a2af4 effect","options":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"unique_id":"0xa4c138f74d3a2af4_effect_zigbee2mqtt"}'
Zigbee2MQTT:debug 2023-01-12 22:24:24: Received MQTT message on 'homeassistant/sensor/0xa4c138f74d3a2af4/power_on_behavior/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138f74d3a2af4"],"manufacturer":"AwoX","model":"LED white (33951/33948)","name":"0xa4c138f74d3a2af4","sw_version":"0122052017"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","name":"0xa4c138f74d3a2af4 power on behavior","state_topic":"zigbee2mqtt/0xa4c138f74d3a2af4","unique_id":"0xa4c138f74d3a2af4_power_on_behavior_zigbee2mqtt","value_templat [... cut ...]
Zigbee2MQTT:info  2023-01-12 22:24:24: Successfully configured '0xa4c138f74d3a2af4'
```
Color temperature can still be changed via bluetooth in the AwoX app.
[image]
[image]

### What did you expect to happen?

I was expecting to be able to change the colour temperature on the devices after firmware update, but could only switch it ON/OFF or change the brightness because they were identified as the wrong device.

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

20210708

### Adapter

Slaesh CC2652RB

capturetheventure, 2023-01-12:
Here are the database.db entries of the devices:

```
{"id":3,"type":"Router","ieeeAddr":"0xa4c138c026e8669f","nwkAddr":46951,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"appVersion":2,"hwVersion":60}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genLevelCtrl":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":2,"hwVersion":60,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1673559389340,"defaultSendRequestWhen":"immediate"}

{"id":5,"type":"Router","ieeeAddr":"0xa4c138e466044d9e","nwkAddr":7995,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"appVersion":2,"hwVersion":60}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorMode":2,"colorTemperature":333}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":193}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":2,"hwVersion":60,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1673559396642,"defaultSendRequestWhen":"immediate"}

{"id":13,"type":"Router","ieeeAddr":"0xa4c138f74d3a2af4","nwkAddr":37841,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":2,"hwVersion":60,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1673559356524,"defaultSendRequestWhen":"immediate"}

{"id":18,"type":"Router","ieeeAddr":"0xa4c138219287882e","nwkAddr":24884,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":2,"hwVersion":60,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1673559361199,"defaultSendRequestWhen":"immediate"}
```

Koenkk, 2023-01-13:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            type: 'Router', manufacturerName: 'AwoX', modelID: 'TLSR82xx', endpoints: [
                {ID: 1, profileID: 260, deviceID: 268, inputClusters: [0, 3, 4, 5, 6, 8, 768, 4096, 64599], outputClusters: [6]},
                {ID: 3, profileID: 4751, deviceID: 268, inputClusters: [65360, 65361], outputClusters: [65360, 65361]},
            ],
        },
        {
            type: 'Router', manufacturerName: 'AwoX', modelID: 'TLSR82xx', endpoints: [
                {ID: 1, profileID: 260, deviceID: 268, inputClusters: [0, 3, 4, 5, 6, 8, 768, 4096, 64599], outputClusters: [6]},
                {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]},
                {ID: 3, profileID: 4751, deviceID: 268, inputClusters: [65360, 65361], outputClusters: [65360, 65361]},
            ],
        },
        {
            type: 'Router', manufacturerName: 'AwoX', modelID: 'TLSR82xx', endpoints: [
                {ID: 1, profileID: 260, deviceID: 268, inputClusters: [0, 3, 4, 5, 6, 8, 768, 4096, 64599, 10], outputClusters: [6]},
                {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]},
                {ID: 3, profileID: 4751, deviceID: 268, inputClusters: [65360, 65361], outputClusters: [65360, 65361]},
            ],
        },
        {
            type: 'Router', manufacturerName: 'AwoX', modelID: 'TLSR82xx', endpoints: [
                {ID: 1, profileID: 260, deviceID: 268, inputClusters: [0, 3, 4, 5, 6, 8, 768, 4096, 64599, 10], outputClusters: [6]},
                {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]},
                {ID: 3, profileID: 4751, deviceID: 268, inputClusters: [65360, 65361, 4], outputClusters: [65360, 65361]},
            ],
        },
    ],
    model: '33957',
    vendor: 'AwoX',
    description: 'LED light with color temperature',
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}),
    whiteLabel: [{vendor: 'EGLO', model: '12239'}],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

capturetheventure, 2023-01-13:
Thank you so much for the fast response!

I added the external converter and now the bulbs are tagged as AwoX 33957 again, color temp control is working fine as well.

But there is one more issue that I still have after the firmware update, even with the added external converter - the bulbs don't turn fully turn off as soon as I add the `transition` option to the `light.turn_off` command within Home Assistant. Instead, they dim down to 1% brightness and then stay ON, even though the state is reported as 'OFF' at the HA entity as well as in the Zigbee2MQTT frontend.
It doesn't matter which time is set for the `transition` option, even 0 seconds result in the same behaviour. As long as I don't use the `transition` option at all when calling the `light.turn_off` command, the bulbs do turn off completely.

I believe this is some kind of issue with the new firmware as the transition option was working flawlessly before.

Below you can see a log where I send the `light.turn_off` command to one of the lights with 4s transition. The light reports state OFF afterwards. When I press `get state` in the frontend, it shows as ON again, and when I press `get brightness`, it also updates to the 1% which it actually has.

The funny thing is: when I send the `light.turn_off` command + transition option to a lamp that's already off, it actually turns on the light to 1% brightness as well, while the light's state remains OFF in the frontend status.

Any idea what could cause this behaviour?

[image]

capturetheventure, 2023-01-13:
At first I thought it might be an issue with the 'Circadian Light' addon I'm using, but this is definitely not the problem.

Koenkk, 2023-01-14:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/16132#issuecomment-1382064283, can you check if it is fixed?

capturetheventure, 2023-01-14:
Updated the external converter and restarted z2m, however the issue still persists.
`light.turn_off` with `transition` stops at brightness 1 and the light stays on, even though status is shown as off.
When exectued while off, the light still turns on to brightness 1.
`light.turn_on` with `transition` when the light is off also still jumps to target brightness straight away.

Koenkk, 2023-01-14:
@capturetheventure if this worked with the old firmware, I think the new firmware has a bug causing this.

capturetheventure, 2023-01-14:
@Koenkk Thanks, I will send a bug report in the AwoX app. Maybe they'll fix it in an upcoming firmware version.
I'll just remove all `light.turn_on/off` transitions for now until it's eventually fixed, not a big deal.

Regarding the wrong identification, I can remove your external converter when z2m 1.29.3 is released, right?

Koenkk, 2023-01-15:
> Regarding the wrong identification, I can remove your external converter when z2m 1.29.3 is released, right?

Yes, with the 1 February release (probably 1.29.3 or 1.30.0)

closed 2023-01-15 by Koenkk

capturetheventure, 2023-01-16:
@Koenkk I recieved feedback from EGLO / AwoX support today. Apparently, these lights were never supposed to support ON / OFF transitions. I thought this was a standard function for Zigbee lights.
Guess I'll just have to live without the transitions from now on.
[image]

bnowakow, 2023-07-24:
@capturetheventure did you get it running?
I'm on fw 20230507 and I have AwoX 33957 and when I run toggle on it - it goes to brightness 1 instead to turn off. I've same behaviour with ext_converter.js and without it.