Koenkk/zigbee2mqtt #16133: Xiaomi Aqara wireless remote switch E1 (single rocker) (WXKG16LM) paired but no actions/data

Dramenia, 2023-01-12:
### What happened?

succesfully paired the deivce:
`./log.txt:1251:info  2023-01-12 21:32:21: Successfully interviewed '0x54ef4410005d271b', device has successfully been paired
./log.txt:1252:info  2023-01-12 21:32:21: Device '0x54ef4410005d271b' is supported, identified as: Xiaomi Aqara wireless remote switch E1 (single rocker) (WXKG16LM)
./log.txt:1253:info  2023-01-12 21:32:21: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara wireless remote switch E1 (single rocker)","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Voltage of the battery in millivolts","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["single","double","hold"]},{"access":7,"description":"Click mode, fast: only supports single click which will be send immediately after clicking.multi: supports more events like double and hold","name":"click_mode","property":"click_mode","type":"enum","values":["fast","multi"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"WXKG16LM","options":[],"supports_ota":false,"vendor":"Xiaomi"},"friendly_name":"0x54ef4410005d271b","ieee_address":"0x54ef4410005d271b","status":"successful","supported":true},"type":"device_interview"}'
./log.txt:1254:info  2023-01-12 21:32:21: Configuring '0x54ef4410005d271b'
./log.txt:1255:info  2023-01-12 21:32:22: Successfully configured '0x54ef4410005d271b'
`
After pairing, i sometimes get this response when pressing the button:
`./log.txt:1280:debug 2023-01-12 21:36:05: Received Zigbee message from '0x54ef4410005d271b', type 'attributeReport', cluster 'genBasic', data '{"appVersion":7,"modelId":"lumi.remote.acn003"}' from endpoint 1 with groupID 0
./log.txt:1281:debug 2023-01-12 21:36:20: Received Zigbee message from '0x54ef4410005d271b', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.remote.acn003"}' from endpoint 1 with groupID 0
./log.txt:1282:debug 2023-01-12 21:36:28: Received Zigbee message from '0x54ef4410005d271b', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.remote.acn003"}' from endpoint 1 with groupID 0
./log.txt:1283:debug 2023-01-12 21:36:31: Received Zigbee message from '0x54ef4410005d271b', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.remote.acn003"}' from endpoint 1 with groupID 0
./log.txt:1284:debug 2023-01-12 21:36:35: Received Zigbee message from '0x54ef4410005d271b', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.remote.acn003"}' from endpoint 1 with groupID 0
./log.txt:1288:debug 2023-01-12 21:36:40: Received Zigbee message from '0x54ef4410005d271b', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.remote.acn003"}' from endpoint 1 with groupID 0
`
But i do not receive any proper data. am i missing something?

### What did you expect to happen?

when pressing the button, seeing the action command appear on MQTT. in the debug i see that the device does give a response (see last bit of the log)

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

20210120

### Adapter

zStack3x0

Dramenia, 2023-01-12:
[image]

Koenkk, 2023-01-13:
Your coordinator firmware is too old (see [docs](https://www.zigbee2mqtt.io/devices/WXKG16LM.html#adapter-firmware) for the minimum required version). After updating re-pair your device.

closed 2023-01-13 by Koenkk

cgarciaq, 2023-01-23:
I am having the same problem: succesful interview, but then no data is received. The exposed sensors and inputs all show N/A.
And, I would say, my coordinator firmware is the last one:

Zigbee2MQTT version
    1.29.2-dev commit: [80c464c](https://github.com/Koenkk/zigbee2mqtt/commit/80c464c)

Coordinador type
    zStack12

Coordinador version
    **20211116**

Coordinator IEEE Address
    0x00124b001949cac2

Interface version
    0.6.123

I have tried several times to remove the device and repair it with no success.

My log:
`MQTT publish: topic 'homeassistant/sensor/0x54ef441000687030/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef441000687030"],"manufacturer":"Xiaomi","model":"Aqara E1 wireless mini switch (WXKG20LM)","name":"0x54ef441000687030"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x54ef441000687030","name":"0x54ef441000687030 voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef441000687030","unique_id":"0x54ef441000687030_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
info 2023-01-23 00:52:35MQTT publish: topic 'homeassistant/sensor/0x54ef441000687030/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef441000687030"],"manufacturer":"Xiaomi","model":"Aqara E1 wireless mini switch (WXKG20LM)","name":"0x54ef441000687030"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x54ef441000687030","name":"0x54ef441000687030 action","state_topic":"zigbee2mqtt/0x54ef441000687030","unique_id":"0x54ef441000687030_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
info 2023-01-23 00:52:35MQTT publish: topic 'homeassistant/sensor/0x54ef441000687030/device_temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef441000687030"],"manufacturer":"Xiaomi","model":"Aqara E1 wireless mini switch (WXKG20LM)","name":"0x54ef441000687030"},"device_class":"temperature","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x54ef441000687030","name":"0x54ef441000687030 device temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef441000687030","unique_id":"0x54ef441000687030_device_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.device_temperature }}"}'
info 2023-01-23 00:52:35MQTT publish: topic 'homeassistant/sensor/0x54ef441000687030/power_outage_count/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef441000687030"],"manufacturer":"Xiaomi","model":"Aqara E1 wireless mini switch (WXKG20LM)","name":"0x54ef441000687030"},"enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x54ef441000687030","name":"0x54ef441000687030 power outage count","state_topic":"zigbee2mqtt/0x54ef441000687030","unique_id":"0x54ef441000687030_power_outage_count_zigbee2mqtt","value_template":"{{ value_json.power_outage_count }}"}'
info 2023-01-23 00:52:35MQTT publish: topic 'homeassistant/sensor/0x54ef441000687030/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef441000687030"],"manufacturer":"Xiaomi","model":"Aqara E1 wireless mini switch (WXKG20LM)","name":"0x54ef441000687030"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x54ef441000687030","name":"0x54ef441000687030 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef441000687030","unique_id":"0x54ef441000687030_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info 2023-01-23 00:52:35Starting interview of '0x54ef441000687030'
info 2023-01-23 00:52:35MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000687030","ieee_address":"0x54ef441000687030","status":"started"},"type":"device_interview"}'
info 2023-01-23 00:52:35MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x54ef441000687030"},"type":"pairing"}'
info 2023-01-23 00:52:36MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000687030","ieee_address":"0x54ef441000687030"},"type":"device_announce"}'
info 2023-01-23 00:52:36MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x54ef441000687030"},"type":"device_announced"}'
info 2023-01-23 00:52:38MQTT publish: topic 'zigbee2mqtt/Enchufe 1', payload '{"child_lock":"UNLOCK","current":0,"device":{"applicationVersion":68,"dateCode":"","friendlyName":"Enchufe 1","hardwareVersion":1,"ieeeAddr":"0xa4c138f640ea962f","manufacturerID":4417,"manufacturerName":"_TZ3000_gjnozsaz","model":"TS011F_plug_3","networkAddress":37991,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":0.2,"indicator_mode":"off/on","linkquality":92,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":230}'
[... cut ...] '
info 2023-01-23 00:53:28MQTT publish: topic 'zigbee2mqtt/Termómetro del baño', payload '{"battery":67,"device":{"applicationVersion":70,"dateCode":"","friendlyName":"Termómetro del baño","hardwareVersion":1,"ieeeAddr":"0xa4c13822cd774c50","manufacturerID":4417,"manufacturerName":"_TZE200_bjawzodf","model":"TS0601_temperature_humidity_sensor","networkAddress":63167,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"humidity":34.4,"linkquality":86,"temperature":17.3}'
`

str3g, 2023-02-07:
Good day! same problem!

cgarciaq, 2023-02-08:
I changed my coordinator from a CC2531 to a CC2652P (sonoff) and the problem was gone.