Koenkk/zigbee2mqtt #16134: Support for Tuya USB Zigbee Repeater - Device wrongly detected

MonacledMonkey, 2023-01-13:
### What happened?

Similar to https://github.com/Koenkk/zigbee2mqtt/issues/13465

Device recognised as water leak detector,

### What did you expect to happen?

Device should be recognised as TS0207_repeater.

### How to reproduce it (minimal and precise)

Pair Zigbee device
Zigbee Model: TS0207
Zigbee Manufacturer: TZ3000_misw04hq
IEEE address: 0x1c34f1fffeb76556
Network Address: 0xDC2F

### Zigbee2MQTT version

1.29.1-dev commit: 75533c2

### Adapter firmware version

0x26580700

### Adapter

Conbee II

Koenkk, 2023-01-14:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-14 by Koenkk

ggocsei, 2023-11-07:
I'm still facing with this issue. I'm using HA Edge version and there is no update, so I'm thinking I'm using tha latest version.

MartinNeurol, 2024-07-31:
Hi, any update please I just bought _TZ3000_kxlmv9ag and guess what ... water leak sensor :/ could you please add this EMOS IP-2213Z repeater to the tuya.js? Many thanks...