Koenkk/zigbee2mqtt #16136: Xiaomi ZNCLBL01LM Aqara Curtain OTA firmware 6168 update wonky results

kris-anderson, 2023-01-13:
### What happened?

I have a Xiaomi ZNCLBL01LM Curtain Driver here:
https://www.zigbee2mqtt.io/devices/ZNCLBL01LM.html

When I check for an update via the GUI, it says an update is available. I apply the update, wait for it to finish, and it looks like everything is up to date. When I check on the status of this device I see this:

```
    "update": {
        "installed_version": 6168,
        "latest_version": 6168,
        "state": "idle"
    },
```

However, when I check for an update again, it says the device has an update available for it. When I then check the status of this device I now see this:

```
    "update": {
        "installed_version": 24,
        "latest_version": 6168,
        "state": "available"
    },
```

If I apply the OTA update again, the same process repeats itself.

### What did you expect to happen?

I expected that after updating the device, when I check for a firmware update, to be told the device is up to date.

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10407622/log.txt)

Aljutor, 2023-02-03:
I have same issue on 1.30.0

GewoonRoy, 2023-02-09:
Same problems in issue #15745

Aljutor, 2023-03-12:
1.30.2 Still same issue

lundyfpv, 2023-03-13:
So I'm not super versed in this but I think the issue could be related to Aqara/Xiaomi region locking. I purchased/rented a Aqara hub to update my firmware to fix this annoying notification and it simply wouldn't let me add the device in their software using the China server. Just an FYI for my experience.

I also seem to have one stuck on the 24 version while the other was updated to the 27 version. 24 version always proceeds like it will update but fails in the end. 27 version always acts like it has an update but it always fails when. Trying to even start the update.

marcgarciamarti, 2024-01-14:
> So I'm not super versed in this but I think the issue could be related to Aqara/Xiaomi region locking. I purchased/rented a Aqara hub to update my firmware to fix this annoying notification and it [...]

There is a similar issue here: https://github.com/Koenkk/zigbee2mqtt/issues/20662

closed 2024-01-31 by Koenkk

Otnow, 2024-01-31:
The [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6983) fixing current problem has been merged.

Note for PR: when pairing, device will immediately display correct firmware version, otherwise you need to wait until device sends next heartbeat message (~ 6-6,5 minutes) or perform its manual reconfiguration and then restart zigbee2mqtt (so that correct version is displayed in web interface).