Koenkk/zigbee2mqtt #16137: HS3HT-EFA-3.0 support

sbouchex, 2023-01-13:
### Link

https://www.heimantech.com/

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0x84fd27fffe8395ed","nwkAddr":49442,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"HS3HT-EFA-3.0","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,1026,2821],"outClusterList":[25],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}},"msTemperatureMeasurement":{"attributes":{"61440":1500,"61441":3500,"61442":1,"61443":200,"measuredValue":1947}},"genBasic":{"attributes":{"modelId":"HS3HT-EFA-3.0","manufacturerName":"HEIMAN","powerSource":3,"zclVersion":3,"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2021.4.11"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":770,"inClusterList":[0,3,1029],"outClusterList":[],"clusters":{"msRelativeHumidity":{"attributes":{"61440":1500,"61441":9000,"61442":1000,"measuredValue":6091}},"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2021.4.11","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1673592624525,"defaultSendRequestWhen":"immediate"}

### Comments

The device is reported as "Not supported"

threnard, 2023-01-28:
Hi. I also have this device and was able to make it work with the following converter.

`zigbeeModel: ['HS3HT-EFA-3.0'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'HS3HT', // Vendor model number, look on the device for a model number
    vendor: 'HEIMAN', // Vendor of the device (only used for documentation and startup logging)
    description: 'Heiman temperature & humidity sensor with display', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.temperature, fz.humidity, fz.battery], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
        await reporting.temperature(endpoint1);
        await reporting.batteryPercentageRemaining(endpoint1);
        await endpoint1.read('genPowerCfg', ['batteryPercentageRemaining']);

        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['msRelativeHumidity']);
        await reporting.humidity(endpoint2);
    },
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};`

Hope this will help.

Koenkk, 2023-01-29:
Thanks @threnard, added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-29 by Koenkk