Koenkk/zigbee2mqtt #16138: Failed to interview Develco AQSZB-110

scontomio, 2023-01-13:
### What happened?

Through Zigbee2MQTT I was able to connect https://www.zigbee2mqtt.io/devices/E2007.html without problems, now I tried the connection of https://www.zigbee2mqtt.io/devices/AQSZB-110.html for 3 hours.

I always get this error:
Unable to query '0x0015bc0036000fa4', the device was not paired correctly.

I've tried changing the batteries, changing channels, pairing near the antenna.

I tried to buy another AQSZB-110 but I still get the same error.

### Zigbee2MQTT version

1.29.1 commit: 7d67ffc

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8 - Sonoff ZBDongle-E

### Debug log

Using '/app/data' as data directory
[WARN  tini (7)] Tini is not running as PID 1 and isn't registered as a child subreaper.
Zombie processes will not be re-parented to Tini, so zombie reaping won't work.
To fix the problem, use the -s option or set the environment variable TINI_SUBREAPER to register Tini as a child subreaper, or run Tini as PID 1.
Zigbee2MQTT:debug 2023-01-13 08:24:59: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2023-01-13 08:24:59: Logging to console and directory: '/tmp' filename: log.txt
Zigbee2MQTT:info  2023-01-13 08:24:59: Starting Zigbee2MQTT version 1.29.1 (commit #7d67ffc)
Zigbee2MQTT:info  2023-01-13 08:24:59: Starting zigbee-herdsman (0.14.83)
Zigbee2MQTT:debug 2023-01-13 08:24:59: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[15],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  2023-01-13 08:25:02: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-01-13 08:25:02: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
Zigbee2MQTT:debug 2023-01-13 08:25:02: Zigbee network parameters: {"channel":15,"extendedPanID":221,"panID":6754}
Zigbee2MQTT:info  2023-01-13 08:25:02: Currently 0 devices are joined:
Zigbee2MQTT:info  2023-01-13 08:25:02: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2023-01-13 08:25:02: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:debug 2023-01-13 08:25:02: Using MQTT login with username: manuel
Zigbee2MQTT:info  2023-01-13 08:25:02: Connected to MQTT server
Zigbee2MQTT:info  2023-01-13 08:25:02: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-01-13 08:25:02: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2023-01-13 08:25:02: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-01-13 08:25:02: Zigbee2MQTT started!
Zigbee2MQTT:debug 2023-01-13 08:25:24: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"h2jxl-1","value":true}'
Zigbee2MQTT:info  2023-01-13 08:25:24: Zigbee: allowing new devices to join.
Zigbee2MQTT:debug 2023-01-13 08:25:24: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,3,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
Zigbee2MQTT:info  2023-01-13 08:25:24: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"h2jxl-1"}'
Zigbee2MQTT:info  2023-01-13 08:25:59: Device '0x0015bc00360010ce' joined
Zigbee2MQTT:info  2023-01-13 08:25:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00360010ce","ieee_address":"0x0015bc00360010ce"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-01-13 08:25:59: Starting interview of '0x0015bc00360010ce'
Zigbee2MQTT:info  2023-01-13 08:25:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00360010ce","ieee_address":"0x0015bc00360010ce","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:error 2023-01-13 08:26:59: Failed to interview '0x0015bc00360010ce', device has not successfully been paired
Zigbee2MQTT:info  2023-01-13 08:26:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00360010ce","ieee_address":"0x0015bc00360010ce","status":"failed"},"type":"device_interview"}'
Zigbee2MQTT:info  2023-01-13 08:27:04: Starting interview of '0x0015bc00360010ce'
Zigbee2MQTT:info  2023-01-13 08:27:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00360010ce","ieee_address":"0x0015bc00360010ce","status":"started"},"type":"device_interview"}'

scontomio, 2023-01-13:
zigbee-herdsman log file 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10411338/log.txt)

cvicari, 2023-01-30:
I tried another thing. I took a cc2531, with the firmware, with firmware zStack12, and in zigbee2mqtt the aqszb-110 paired successfully after a couple of attempts.
So it definitely seems an issue with aqszb-110 + Sonoff ZBDongle-E

scontomio, 2023-02-03:
I've same problem with 1.30.0 :(

cvicari, 2023-03-06:
I tried again with version 1.30.2, still no success with ezsp and AQSZB-110

closed 2023-04-15 by github-actions[bot]

cvicari, 2024-04-08:
I tried again and found a workaround, with version 1.36.1.
Pairing the device against the EZSP v8 - Sonoff ZBDongle-E coordinator fails.
I have a CC2531 router. Pairing the device against the router succeeded after a few attempts.

This has been stable for more than one week now. Based on the map, the AQSZB-110 is currently attached to another router and not the CC2531 with which I began, so it seems like the solution is stable